facebookresearch/habitat-sim #824: What is the unit of time in step_physics?

qureshinomaan, 2020-10-03:
I wanted to know the unit of dt in step_physics function. I am working on a controller and want to update the velocity at every second, but I am confused about the unit of time. Is it milliseconds, seconds or something else?
Thanks!

Skylion007, 2020-10-03:
@qureshinomaan Physics run at 60FPS by default (1/60 seconds), but it is adjustable in as kwarg in `Simulator.step` https://github.com/facebookresearch/habitat-sim/blob/a0605bc6809bf0e35ed1540817ab2ceb99b1a5f8/habitat_sim/simulator.py#L271

See our RigidObjects ColabTutorial for more information: https://colab.research.google.com/github/facebookresearch/habitat-sim/blob/master/examples/tutorials/colabs/rigid_object_tutorial.ipynb#scrollTo=7jURIkMODqCi&line=2&uniqifier=1

qureshinomaan, 2020-10-04:
Thanks!
@Skylion007 One more thing that I wanted to know. 
```
agents[0].action_space[some_action].actuation.amount = 1
```
Here what is the unit of action. For example if some_action='move_forward', then it would be in meters or centimetres or something else. Similarly if some_action='look_up', what is the action here?

Skylion007, 2020-10-04:
It's in SI units. AKA meters. That action would make it go forward 1 meter.

qureshinomaan, 2020-10-05:
@Skylion007 
I think the angles are in degrees, which should have been in radians(following the SI convention).

Skylion007, 2020-10-05:
@qureshinomaan Yeah, actuation for turn is in degree, actuation for movement is in meters.

closed 2020-10-06 by Skylion007