facebookresearch/habitat-sim #827: phys_scene_config.json file not found

Mingxiao-Li, 2020-10-05:
## ❓ Questions and Help
Hi, I got the below error, and have found this phys file in this github. Does anyone know where can I download this file ?
 File (./data/default.phys_scene_config.json) not found so new, default physics manager attributes created and registered.

jturner65, 2020-10-05:
Can you share more information about your habitat-sim installation?

Mingxiao-Li, 2020-10-05:
Hi. 
I follow the instruction on the github webpage and use conda package to install habitat-sim. The system is Ubuntu. 
I got this error when I try to evaluate the baseline model of VLN-CE model. But then I just found that it is the below error abort the program. 
E1005 14:26:40.204874  5873 StageAttributesManager.cpp:121] StageAttributesManager::registerAttributesTemplateFinalize : Render asset template handle : data/scene_datasets/mp3d/v1/task/mp3d/x8F5xyUWy9e/x8F5xyUWy9e.glb specified in stage template with handle : data/scene_datasets/mp3d/v1/task/mp3d/x8F5xyUWy9e/x8F5xyUWy9e.glb does not correspond to any existing file or primitive render asset.  Aborting. 
Could you kindly inform me how should I solve this error ?

Mingxiao-Li, 2020-10-05:
Problem solved. Thanks

closed 2020-10-05 by Mingxiao-Li

yuanhui-code, 2020-11-17:
> Problem solved. Thanks

Hello! I also encountered the same problem, how did you solve it? Thank you!

wwtx9, 2021-01-24:
Hello, I have encountered the same problem, how did you solve it?

> > Problem solved. Thanks
> 
> Hello! I also encountered the same problem, how did you solve it? Thank you!

abstinentcode, 2022-08-23:
> 

Hello! I also encountered the same problem, how did you solve it? Thank you!

abstinentcode, 2022-08-23:
Hello, I also encountered the same problem, how did you solve it? Thank you! @wwtx9 @yuanhui-code @Mingxiao-Li

aclegg3, 2022-08-23:
@abstinentcode likely you are getting this message because the file [data/default.physics_config.json](https://github.com/facebookresearch/habitat-sim/blob/main/data/default.physics_config.json) needs to be in your data folder or provided during initialization with [SimulatorConfiguration.physics_config_file](https://aihabitat.org/docs/habitat-sim/habitat_sim.sim.SimulatorConfiguration.html#physics_config_file).

If you clone the habitat-sim repo (recommended) and run code from within that directory, this file will be in the correct place. Otherwise you should create this file and manually set the path to it on your system.