facebookresearch/habitat-sim #828: How to keep track of inserted objects in addition to object_ids, order of object_ids ?

ronjamoller, 2020-10-05:
## ❓ Questions and Help
Hi,
I am using a simulation environment that contains inserted objects which are regularly moved, removed and reinserted. I try to keep track of them via the object_id return values from `env._sim._sim.add_object_by_handle()` but I am observing that the correspondence object to object id changes after a removal. For example: 
I insert 10 objects that are added properly and get the ids 0 to 9, I remove all of them, I reinsert all of them and they receive the object_ids 9 to 0 - so the exact opposite order.
If i insert 10 items, then insert one additional item, then remove all, then reinsert the initial 10 items again the ids that are returned for the added objects are 10 to 1 - so opposite order and the offset caused by the one additional object. 
This way of keeping track of the objects is rather stressful for me, is there a way of assigning an individual name to an inserted object as well or is there some additional parameter or setting that forces the insertion to always start at id = 0 ?
thank you !

jturner65, 2020-10-05:
Hello
Thank you for sharing this situation, and I'm sorry that this is a cause of frustration for you.  While there are many mechanisms in place to manage the templates used to create objects, there currently is no way other than via the ID to tag/manage the individual object instances.  

When an item is removed it does not persist in any meaningful way, so there is currently no way to reassign it to the same ID it once had.   The ID it once had is returned to a stack of available IDs that are available to be assigned, which is queried for available IDs when a new object is created before using a new ID value.  

One possible work around is to remove the objects in the opposite order of how they were added.  Then, the IDs will be available in the same order that they were originally consumed.

jturner65, 2020-10-05:
In an ideal world, how would you like to access individual objects?  For example, would it help to be able to give them names upon creation?

ronjamoller, 2020-10-05:
Hi, thank you for answering so quickly - i think the workaround of removing the items in the reverse order might be a start for me. In terms of ideal world - it would of course be nice to call objects by a string, but I am not sure how many people are actually changing the environments to this extent. I'm assuming that my problem is a bit too specific to justify development time. I will check out the removal order, thank you for the tip !

aclegg3, 2022-03-01:
Please see the [managed-rigid-object-tutorial](https://aihabitat.org/docs/habitat-sim/managed-rigid-object-tutorial.html) for details on Habitat 2.0 object management and wrapper system.

closed 2022-03-01 by aclegg3