facebookresearch/habitat-sim #851: Habitat-sim: Platform::WindowlessEglApplication::tryCreateContext(): unable to find EGL device for CUDA device 0

albert-yue, 2020-10-10:
## ❓ Questions and Help
Hi there,

I'm trying to set up habitat-sim on a remote Ubuntu 18.04 machine and running into the following error when trying to run simulations. Any help with installation would be appreciated:

```
I1010 18:31:04.590993 11476 SceneGraph.h:93] Created DrawableGroup:
Platform::WindowlessEglApplication::tryCreateContext(): unable to find EGL device for CUDA device 0
WindowlessContext: Unable to create windowless context
```

## Specs
Ubuntu 18.04.4 with 2 NVIDIA GPUs. SSH via Ubuntu for Windows 10

## Steps to recreate
* Install habitat-sim:
```
conda env create -n habitat python=3.8 cmake=3.14.0
conda activate habitat
conda install -n habitat habitat-sim -c conda-forge -c aihabitat
```
* Download and run example: Cloned the repo for `examples/example.py` and downloaded test data. Ran
```
python examples/example.py --scene  ../test_data/scene_datasets/habitat-test-scenes/skokloster-castle.glb
```
and got the error
```
sim_cfg.physics_config_file = ./data/default.phys_scene_config.json
==== Initialized Sensor Spec: =====
Sensor uuid:  color_sensor
Sensor type:  SensorType.COLOR
Sensor position:  [0.  1.5 0. ]
===================================
WARNING: Logging before InitGoogleLogging() is written to STDERR
I1010 18:31:04.589915 11476 AssetAttributesManager.cpp:122] Asset attributes (capsule3DSolid : capsule3DSolid_hemiRings_4_cylRings_1_segments_12_halfLen_0.75_useTexCoords_false_useTangents_false) created and registered.
I1010 18:31:04.589975 11476 AssetAttributesManager.cpp:122] Asset attributes (capsule3DWireframe : capsule3DWireframe_hemiRings_8_cylRings_1_segments_16_halfLen_1) created and registered.
I1010 18:31:04.590009 11476 AssetAttributesManager.cpp:122] Asset attributes (coneSolid : coneSolid_segments_12_halfLen_1.25_rings_1_useTexCoords_false_useTangents_false_capEnd_true) created and registered.
I1010 18:31:04.590031 11476 AssetAttributesManager.cpp:122] Asset attributes (coneWireframe : coneWireframe_segments_32_halfLen_1.25) created and registered.
I1010 18:31:04.590044 11476 AssetAttributesManager.cpp:122] Asset attributes (cubeSolid : cubeSolid) created and registered.
I1010 18:31:04.590054 11476 AssetAttributesManager.cpp:122] Asset attributes (cubeWireframe : cubeWireframe) created and registered.
I1010 18:31:04.590085 11476 AssetAttributesManager.cpp:122] Asset attributes (cylinderSolid : cylinderSolid_rings_1_segments_12_halfLen_1_useTexCoords_false_useTangents_false_capEnds_true) created and registered.
I1010 18:31:04.590111 11476 AssetAttributesManager.cpp:122] Asset attributes (cylinderWireframe : cylinderWireframe_rings_1_segments_32_halfLen_1) created and registered.
I1010 18:31:04.590124 11476 AssetAttributesManager.cpp:122] Asset attributes (icosphereSolid : icosphereSolid_subdivs_1) created and registered.
I1010 18:31:04.590137 11476 AssetAttributesManager.cpp:122] Asset attributes (icosphereWireframe : icosphereWireframe_subdivs_1) created and registered.
I1010 18:31:04.590155 11476 AssetAttributesManager.cpp:122] Asset attributes (uvSphereSolid : uvSphereSolid_rings_8_segments_16_useTexCoords_false_useTangents_false) created and registered.
I1010 18:31:04.590189 11476 AssetAttributesManager.cpp:122] Asset attributes (uvSphereWireframe : uvSphereWireframe_rings_16_segments_32) created and registered.
I1010 18:31:04.590196 11476 AssetAttributesManager.cpp:108] AssetAttributesManager::buildCtorFuncPtrMaps : Built default primitive asset templates : 12
W1010 18:31:04.590773 11476 ObjectAttributesManager.cpp:326] Cannot find ./data/objects or ./data/objects.phys_properties.json. Aborting parse.
I1010 18:31:04.590783 11476 PhysicsAttributesManager.cpp:39] File (./data/default.phys_scene_config.json) Based physics manager attributes created and registered.
I1010 18:31:04.590869 11476 StageAttributesManager.cpp:79] File (../habitat-sim_test-data/scene_datasets/habitat-test-scenes/skokloster-castle.glb) Based stage attributes created and registered.
I1010 18:31:04.590881 11476 Simulator.cpp:145] Loading navmesh from ../habitat-sim_test-data/scene_datasets/habitat-test-scenes/skokloster-castle.navmesh
I1010 18:31:04.590975 11476 Simulator.cpp:147] Loaded.
I1010 18:31:04.590993 11476 SceneGraph.h:93] Created DrawableGroup:
Platform::WindowlessEglApplication::tryCreateContext(): unable to find EGL device for CUDA device 0
WindowlessContext: Unable to create windowless context
```

erikwijmans, 2020-10-10:
You can set the environment variables 
```
export MAGNUM_LOG=verbose 
export MAGNUM_GPU_VALIDATION=ON
```
for additional logging about why it couldn't find a device.

Please look at https://github.com/facebookresearch/habitat-sim/issues/288 as it likely contains the answer. (We changed the headless GPU context code since that issue to error logging will look slightly different, but the fix is likely the same)

albert-yue, 2020-10-11:
Ah turns out that the remote machine hadn't had the nvidia-drivers installed. Thanks!

closed 2020-10-11 by albert-yue