facebookresearch/habitat-sim #856: Add clean Python API to add sensors without restarting the simulator

Skylion007, 2020-10-13:
## 🚀 Feature

According to @erikwijmans , we can actually add sensors after the simulator has been instationated by we currently don't have clean bindings to do so. 

` You need to add the sensor spec to the agent's _sensors SensorSuite via .add(sensor_object) and then you add the actual rendering object to the simulators _sensors dictionary (the habitat_sim.simulator.Sensor object)`

We should add a nice Python function to do so.
## Motivation
Removes another reason to have to restart the simulator and is a useful feature. It would be useful to doing things like creating 360 sensors on the fly in Habitat_Lab. 

## Pitch
You should be able to add and remove sensors without restarting the environment. (A fun use case would be a robot turning on an arm mounted camera for example).

erikwijmans, 2020-10-13:
> clean bindings

Just to clarify, no cpp/python bindings are needed.  There just needs to be a little bit of plumbing done on the python side (all the requisite things already have python bindings as sensors are instantiated in/by python)

yacth, 2020-10-13:
Hi,
I am interested in working on this PR as a new contributor in open source projects.

Could you give me some insights to start working on that.

(do I need a good GPU for this)

erikwijmans, 2020-10-14:
Hi,
The way I see this working is adding a
```
add_sensor(self, sensor_config, agent_id = 0):
```
method to the simulator class.

That would in-turn an add_sensor method on the agent so that it can be added to the agent's config and create the sensor in its SensorSuite.  Then that new sensor would also be added to the `_sensors` dictionary that is part of the simulator (this is what actually makes the rendering calls and transfers the results into memory owned by python).

You don't need a GPU for development.  There are some tests that require one, but the CI will run those.

Relevant files: https://github.com/facebookresearch/habitat-sim/blob/master/habitat_sim/simulator.py#L231 https://github.com/facebookresearch/habitat-sim/blob/master/habitat_sim/agent/agent.py#L145

yacth, 2020-10-15:
Hi @erikwijmans,

I started looking at it, just the time to be more familiar with the code and I'll try to do it.

Skylion007, 2020-11-05:
This might be a (minor) blocker @aclegg3 , @jturner65 to our viz_utils idea.

PR #946 (2020-12-07): Refactor Multi-Agent Sensors API and allow dynamic adding of Sensors
## Motivation and Context

Closes #856 

This also allows other static sensors to be added to the scene (like security cameras / static ortho cameras / bird eye views etc...)

## How Has This Been Tested
Locally. Will add a test case shortly, just want to finalize the API. 

Also expanded the _sensor_dict to be multi-agent compatible, let me know if I should be doing it in a different way, like by making _sensors a ```dict[int, dict]``` instead. 
With this, the only thing unblocking multi-agent support on the habitat-sim part [... cut]
closed 2020-12-07 by Skylion007