facebookresearch/habitat-sim #858: Conda installation issues

unnat, 2020-10-14:
## ❓ Questions and Help

1. Even after adding the conda-forge channel the nightly build doesn't seem to be available:

```
$ conda install habitat-sim -c conda-forge -c aihabitat-nightly
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.

PackagesNotFoundError: The following packages are not available from current channels:

  - habitat-sim

Current channels:

  - https://conda.anaconda.org/conda-forge/osx-64
  - https://conda.anaconda.org/conda-forge/noarch
  - https://conda.anaconda.org/aihabitat-nightly/osx-64
  - https://conda.anaconda.org/aihabitat-nightly/noarch
  - https://repo.anaconda.com/pkgs/main/osx-64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/r/osx-64
  - https://repo.anaconda.com/pkgs/r/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page
```

2. After installing the stable build, the `python examples/example.py --scene /path/to/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb` works fine:
```
 ========================= Performance ======================== 
 640 x 480, total time 2.38 s, frame time 2.380 ms (420.1 FPS)
 ============================================================== 
```
 but the `python examples/example.py` in `habitat-lab` gives the following error:
```
  File "/Users/unnatj/habitat-lab/habitat/sims/habitat_simulator/habitat_simulator.py", line 315, in reconfigure
    is_same_scene = habitat_config.SCENE == self._current_scene
AttributeError: 'Configuration' object has no attribute 'SCENE'
Segmentation fault: 11
``` 

Recommendations to fix this?

erikwijmans, 2020-10-14:
I don't think the macOS binaries are up to date and I don't think there are any in the nightly channel as nightly is managed by the CI.  CC @mathfac

Skylion007, 2020-10-20:
@mathfac we are getting multiple issues on this now. See the most recent issues ie: #872

Lucaweihs, 2020-10-20:
👍 This is also causing problems for me. The main problem is that the most recent version of `habitat-lab` (with the `gym` version update) only works with recent versions of `habitat-sim` (because of the way that `habitat_sim.Simulator` was updated to expect a new config type).

erikwijmans, 2020-10-20:
I started a build a MacOS binaries for 0.1.6

erikwijmans, 2020-10-21:
Uploaded 0.1.6 binaries for `{3.6, 3.7, 3.8} x {bullet, no-bullet}`.  Should be solved.

dhruvbatra, 2020-10-21:
@unnat and @Lucaweihs -- this 0.1.6 build should resolve this for now. We are looking into adding macOS support for nightly builds.

unnat, 2020-10-21:
Thanks for the conda fix @erikwijmans @Skylion007 @dhruvbatra!

closed 2020-10-21 by unnat