facebookresearch/habitat-sim #859: How to add one or more green circles to the topdown map？

yyf17, 2020-10-14:
How to add one or more green circles to the topdown map, the geographic coordinates of these circles are given, for example (1.390355, -1.5434817, 3.183166)，The drawing effect is shown below
[image]

My version of habitat_sim is v0.1.5, and the version of habitat is v0.1.5.
Thanks a lot

erikwijmans, 2020-10-14:
This function will convert to grid coordinates from world coordinates: https://github.com/facebookresearch/habitat-lab/blob/master/habitat/utils/visualizations/maps.py#L184

This is slightly confusing, but given a point `pt = (x, y, z)` in world coordinates, that function needs to be called as `to_grid(pt[2], pt[0]....)`

yyf17, 2020-10-14:
Thank you for pointing out the key point. How to achieve it? Is this achieved by modifying the habitat_simulator class or the maps class?
Can you give a reference example of implementation?
My version of habitat_sim is v0.1.5, and the version of habitat is v0.1.5.
Thanks a lot

yyf17, 2020-10-15:
@erikwijmans Thank you for your suggestions. Can you give me more details about the above question?

erikwijmans, 2020-10-15:
That file/method exists in 1.5 also.  It'll be a little bit different there as in 1.5 there was an additional crop on the top down map.  You can see how the red and blue boxes are added in the `TopDownMap` class https://github.com/facebookresearch/habitat-lab/blob/master/habitat/tasks/nav/nav.py#L676 (in the same file but likely different line number in 1.5) for reference.

closed 2020-10-21 by dhruvbatra

kaushikb258, 2021-02-21:
I am unable to get the points of the robot within the unoccupied/freespace region in the top down map. I tried this:  

top_down_map = get_topdown_map_from_sim(env, map_resolution=MAP_RESOLUTION, draw_border=False, agent_id=0)
this_agent = env.get_agent(0)
pt = this_agent.get_state().position
row, col = to_grid(pt[2], pt[0], grid_resolution=(MAP_RESOLUTION, MAP_RESOLUTION), sim=env, pathfinder=env.pathfinder)

When I place a marker at the (row,col) location in top_down_map, it appears in the occupied region (i.e., where top_down_map = 0). How to solve this?