facebookresearch/habitat-sim #865: Error in rendering high quality images from Replica dataset

Ainaz99, 2020-10-16:
## ❓ Questions and Help
Hi, I tried rendering high-quality RGB images from Replica dataset. I used `sorted faces` and pointed to the `mesh.ply` file in the scene. Looks like textures are loading correctly, but there's an error in loading mesh.ply : `ResourceManager::loadStage : Stage mesh.ply Collision mesh load failed. Aborting scene initialization.` Thank you so much for your help.

Here's also the full error.

```
I1016 19:13:42.846678 11286 PTexMeshData.cpp:471] Splitting mesh... 
I1016 19:13:43.435354 11286 PTexMeshData.cpp:381] The number of quads: 1908569, which equals to 3817138 triangles.
I1016 19:13:43.435405 11286 PTexMeshData.cpp:812] Loading mesh 1/12... 
I1016 19:13:43.441056 11286 PTexMeshData.cpp:812] Loading mesh 2/12... 
I1016 19:13:43.444914 11286 PTexMeshData.cpp:812] Loading mesh 3/12... 
I1016 19:13:43.449556 11286 PTexMeshData.cpp:812] Loading mesh 4/12... 
I1016 19:13:43.454962 11286 PTexMeshData.cpp:812] Loading mesh 5/12... 
I1016 19:13:43.456931 11286 PTexMeshData.cpp:812] Loading mesh 6/12... 
I1016 19:13:43.457381 11286 PTexMeshData.cpp:812] Loading mesh 7/12... 
I1016 19:13:43.462954 11286 PTexMeshData.cpp:812] Loading mesh 8/12... 
I1016 19:13:43.465827 11286 PTexMeshData.cpp:812] Loading mesh 9/12... 
I1016 19:13:43.468580 11286 PTexMeshData.cpp:812] Loading mesh 10/12... 
I1016 19:13:43.471900 11286 PTexMeshData.cpp:812] Loading mesh 11/12... 
I1016 19:13:43.476325 11286 PTexMeshData.cpp:812] Loading mesh 12/12... 
I1016 19:13:43.479084 11286 PTexMeshData.cpp:838] Calculating mesh adjacency... 
I1016 19:13:44.575686 11286 PTexMeshData.cpp:885] loading atlas textures: 
I1016 19:13:44.576220 11286 PTexMeshData.cpp:894] Loading atlas 1/12 from textures/0-color-ptex.hdr. 
I1016 19:13:44.966266 11286 PTexMeshData.cpp:894] Loading atlas 2/12 from textures/1-color-ptex.hdr. 
I1016 19:13:45.298418 11286 PTexMeshData.cpp:894] Loading atlas 3/12 from textures/2-color-ptex.hdr. 
I1016 19:13:45.627838 11286 PTexMeshData.cpp:894] Loading atlas 4/12 from textures/3-color-ptex.hdr. 
I1016 19:13:46.026206 11286 PTexMeshData.cpp:894] Loading atlas 5/12 from textures/4-color-ptex.hdr. 
I1016 19:13:46.171928 11286 PTexMeshData.cpp:894] Loading atlas 6/12 from textures/5-color-ptex.hdr. 
I1016 19:13:46.220913 11286 PTexMeshData.cpp:894] Loading atlas 7/12 from textures/6-color-ptex.hdr. 
I1016 19:13:46.682073 11286 PTexMeshData.cpp:894] Loading atlas 8/12 from textures/7-color-ptex.hdr. 
I1016 19:13:46.898110 11286 PTexMeshData.cpp:894] Loading atlas 9/12 from textures/8-color-ptex.hdr. 
I1016 19:13:47.123327 11286 PTexMeshData.cpp:894] Loading atlas 10/12 from textures/9-color-ptex.hdr. 
I1016 19:13:47.383754 11286 PTexMeshData.cpp:894] Loading atlas 11/12 from textures/10-color-ptex.hdr. 
I1016 19:13:47.633412 11286 PTexMeshData.cpp:894] Loading atlas 12/12 from textures/11-color-ptex.hdr. 
E1016 19:13:47.841044 11286 ResourceManager.cpp:331] ResourceManager::loadStage : Stage mesh.ply Collision mesh load failed. Aborting scene initialization.
E1016 19:13:47.841078 11286 Simulator.cpp:199] Cannot load mesh.ply
I1016 19:13:47.841223 11286 PhysicsManager.cpp:33] Deconstructing PhysicsManager
I1016 19:13:47.841229 11286 SceneManager.h:25] Deconstructing SceneManager
I1016 19:13:47.841235 11286 SceneGraph.h:26] Deconstructing SceneGraph
I1016 19:13:48.071816 11286 Renderer.cpp:34] Deconstructing Renderer
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/root/miniconda3/lib/python3.8/site-packages/habitat_sim/utils/data/data_extractor.py", line 93, in __init__
    sim = habitat_sim.Simulator(self.cfg)
  File "<attrs generated init habitat_sim.simulator.Simulator>", line 9, in __init__
  File "/root/miniconda3/lib/python3.8/site-packages/habitat_sim/simulator.py", line 100, in __attrs_post_init__
    self.__set_from_config(self.config)
  File "/root/miniconda3/lib/python3.8/site-packages/habitat_sim/simulator.py", line 212, in __set_from_config
    self._config_backend(config)
  File "/root/miniconda3/lib/python3.8/site-packages/habitat_sim/simulator.py", line 149, in _config_backend
    super().__init__(config.sim_cfg)
ValueError: Cannot load: mesh.ply
```

Skylion007, 2020-10-16:
This issue is known and tracked in #613 . Marking as duplicate.

erikwijmans, 2020-10-16:
This is different than #613 as I can reproduce without bullet so re-opening.

@jturner65 @aclegg3 It seems like `buildMeshGroups` isn't valid for PTex meshes, there is a "// TODO : PTEX collision support" in the function.  In the case of no bullet, do we need a collision mesh/meshGroup?

aclegg3, 2020-10-16:
Thanks, @erikwijmans, completely right. We shouldn't need to query the MeshGroups at all when physics is disabled. 

@Ainaz99 thanks for bringing this bug to our attention! Try the patch (#866) for now, it should unblock you.

PR #867 (2020-10-19): Add ptex collision mesh data
## Motivation and Context

Add collision mesh data to ptex meshes.  Slightly messy as there are some complications of ownership, but, at some point, we will want to use a decimated mesh for collisions so the containers that only are used when the mesh is split will be used all the time.

Closes #866 
Fixes #865 

## How Has This Been Tested

Locally.  Viewer loads the ptex mesh and navmesh recomputation works as expected with it.

## Types of changes

Bug fix (non-breaking change which fixes an issue)
New feature (non-breaking change which adds functionality)
closed 2020-10-19 by erikwijmans