facebookresearch/habitat-sim #871: How to change camera parameters?

Ainaz99, 2020-10-20:
## ❓ Questions and Help
Hi,
How can I change the camera parameters such as focal length and field of view? Thanks!

Skylion007, 2020-10-20:
Changing the HFOV and VFOV of any of sensor can be done by modifying the sensor config. See the Python bindings for how to set them or utilities to generate the configs in the examples folder.

We use a pinhole camera model so the focal length should be able to be calculate from the other camera parameters like FOV and resolution if I am not mistaken.

@erikwijmans Maybe we should add a helper function to return the intrinsic matrix since this seems to be a common ask?

Ainaz99, 2020-10-20:
Is this how to change the HFOV? Using `examples/settings.py` ?
[image]

Skylion007, 2020-10-20:
Yes

Ainaz99, 2020-10-21:
Got it! Thanks!

closed 2020-10-21 by Ainaz99

AgentEXPL, 2021-07-28:
> Changing the HFOV and VFOV of any of sensor can be done by modifying the sensor config. See the Python bindings for how to set them or utilities to generate the configs in the examples folder.

Which attributes of a sensor can be set to be a specific value? As far as I know, the "position (including height), resolution, uuid, and sensor_type" of the sensor can be specified. Is there any notes or files on explaining how many kinds of attributes of a RGB sensor and a depth sensor can be specified?

For now, there are two attributes of the camera needing to be specified by me, but I'm not sure whether they can be specified and how are they specified in habitat_sim and habitat_lab respectively.
1. The "VFOV" of the camera. For habitat_sim, is it OK to use "sensor_spec.VFOV = 110" in the python file? For habitat_lab, is it OK to use "VFOV: 110" in the configuration file (.yaml)?

2. The "tilt" of the camera. I have no idea on how to adjust the tilt of the camera (i.e. looking more toward the ground or the ceiling instead of straight ahead). Is it possible to set the value of tilt_angle of the camera (not the agent) in habitat_api? If yes, what should I capture in the configuration file (.yaml)?

@Skylion007 I'll be very appreciated if some guidance could be provided.