facebookresearch/habitat-sim #873: ImportError: libCorradePluginManager.so.2: cannot open shared object file: No such file or directory

Sunnyzhr, 2020-10-20:
## 🐛 Bug
I am running tutorial codes on https://github.com/facebookresearch/habitat-lab/tree/master/habitat_baselines#additional-utilities
## Command
I succeed to run "train" code:

> (habitat) root@c:~/Desktop/habitat-lab# python -u habitat_baselines/run.py --exp-config habitat_baselines/config/pointnav/ppo_pointnav_example.yaml --run-type train

I fail to run "eval" code:

> (habitat) root@c:~/Desktop/habitat-lab# python -u habitat_baselines/run.py --exp-config habitat_baselines/config/pointnav/ppo_pointnav_example.yaml --run-type eval
`ImportError: libCorradePluginManager.so.2: cannot open shared object file: No such file or directory`

Then I try to run the "trian" agian, but fail, and get the same ImportError.
`ImportError: libCorradePluginManager.so.2: cannot open shared object file: No such file or directory`

## Info
- I have activated the "habitat" env of annaconda.

- Run the command below
>(habitat) root@c:~/Desktop/habitat-lab# locate libCorradePluginManager.so.2

- The results is

> /home/u/Desktop/original-habitat-sim/build/lib.linux-x86_64-3.6/habitat_sim/_ext/libCorradePluginManager.so.2
/home/u/Desktop/original-habitat-sim/build/lib.linux-x86_64-3.6/habitat_sim/_ext/libCorradePluginManager.so.2.4
/home/u/anaconda3/envs/habitat/lib/python3.6/site-packages/habitat_sim-0.1.6-py3.6-linux-x86_64.egg/habitat_sim/_ext/libCorradePluginManager.so.2
/home/u/anaconda3/envs/habitat/lib/python3.6/site-packages/habitat_sim-0.1.6-py3.6-linux-x86_64.egg/habitat_sim/_ext/libCorradePluginManager.so.2.4
/home/u/anaconda3/lib/python3.8/site-packages/habitat_sim/_ext/libCorradePluginManager.so.2
/home/u/anaconda3/lib/python3.8/site-packages/habitat_sim/_ext/libCorradePluginManager.so.2.4
/home/u/anaconda3/pkgs/habitat-sim-0.1.6-py3.8_linux_781f787714c7f8e6da7b62c69390ada7109167ae/lib/python3.8/site-packages/habitat_sim/_ext/libCorradePluginManager.so.2
/home/u/anaconda3/pkgs/habitat-sim-0.1.6-py3.8_linux_781f787714c7f8e6da7b62c69390ada7109167ae/lib/python3.8/site-packages/habitat_sim/_ext/libCorradePluginManager.so.2.4

- Why there is ImportError, since libCorradePluginManager.so.2 exits in ''/home/u/anaconda3/envs/habitat/lib/python3.6/site-packages/habitat_sim-0.1.6-py3.6-linux-x86_64.egg/habitat_sim/_ext/libCorradePluginManager.so.2''?

## Expected behavior

Successfully run the code

erikwijmans, 2020-10-27:
What does 
```
ldd /home/u/anaconda3/envs/habitat/lib/python3.6/site-packages/habitat_sim-0.1.6-py3.6-linux-x86_64.egg/habitat_sim/_ext/habitat_sim_bindings.cpython-36m-x86_64-linux-gnu.so
```

show and what does 

```
readelf -d /home/u/anaconda3/envs/habitat/lib/python3.6/site-packages/habitat_sim-0.1.6-py3.6-linux-x86_64.egg/habitat_sim/_ext/habitat_sim_bindings.cpython-36m-x86_64-linux-gnu.so
```

Something is wrong with your install but I am not sure what.

Sunnyzhr, 2020-11-30:
Miraculously, after reinstalling the whole Ubuntu system, the bug disappears.
But I am still appreciate your help! Thanks for your attention! @erikwijmans

closed 2021-08-16 by dhruvbatra