facebookresearch/habitat-sim #875: Mapping camera rotations from blender to habitat

Ainaz99, 2020-10-20:
## ❓ Questions and Help
Hi,
I'm trying to map some camera rotations from blender to habitat on replica dataset. I know the Blender coordinate system is Z up, Y forward, and Habitat is  Y up, -Z forward, but I can't figure out how to correctly map the camera euler angles from blender to habitat. I really appreciate it if you can help me with this. Thanks a lot.

aclegg3, 2020-10-20:
Hey @Ainaz99. Assuming your Blender asset is oriented Z up and Y forward with the scene, you can simply apply the Blender->Habitat rotation to the camera position and orientation.

My Blender camera state (note I am choosing quaternion representation for rotation, you can toggle this):
[image]

Example C++ code (e.g. Habitat Viewer application) to set the agent transform from this Blender camera state (in this example the agent has no height, and is coincident with the camera):
```
  Mn::Vector3 blenderPos(0.408654, 0.307441, 2.63599);
  Mn::Quaternion blenderRotation({0.223108,-0.528059,-0.818912},0.027591);
  blenderRotation = blenderRotation.normalized();
  //coordinate frame transform from Blender -> Habitat
  Mn::Quaternion blendToHabitat = Mn::Quaternion::rotation(Mn::Rad(-1.56),{1,0,0});
  Mn::Matrix4 R(blendToHabitat.toMatrix());
  auto habitatPos = R.transformPoint(blenderPos);
  auto habitatRotation = blendToHabitat*blenderRotation;
 //set the transform to the agent
  agentBodyNode_->setTranslation(habitatPos);
  agentBodyNode_->setRotation(habitatRotation);
```

If you want to do the same for a camera attached with some offset to the agent (the default setup), you can create an empty frame in Blender to parent your camera with some offset.

Ainaz99, 2020-10-20:
Thanks a lot! It worked!

closed 2020-10-20 by Ainaz99