facebookresearch/habitat-sim #891: Got this error ""

Mingxiao-Li, 2020-10-22:
## 🐛 Bug

Hi, I got this error "Process finished with exit code 139" when I call Simulator.pathfinder.getbounds(). Is there anyone knows how to solve it ?

## Command
`# command that you run`

## To Reproduce

Steps to reproduce the behavior:

1.
2.
3.

## Expected behavior

## Additional context

## System Info

Please copy and paste the output from the
[environment collection script](https://github.com/facebookresearch/habitat-sim/tree/master/habitat_sim/utils/collect_env.py)
(or fill out the checklist below manually).

You can run the script with:
```
# For security purposes, please check the contents of collect_env.py before running it.
python habitat_sim/utils/collect_env.py
```

 - OS (e.g., Linux):
 - How you installed PyTorch (`conda`, `pip`, source):
 - Build command you used (if compiling from source):
 - Python version:
 - GPU models and configuration:
 - CUDA version:
 - PIP/conda dependencies packages versions
 - Any other relevant information:

Skylion007, 2020-10-22:
Pathfinder requires a navmesh, if you are using a custom scene, did you recompute the NavMesh? Without more info, I can reproduce.

Skylion007, 2020-11-01:
Closing until more info is provided.

closed 2020-11-01 by Skylion007