facebookresearch/habitat-sim #903: Rotation about an object

ayushjain1144, 2020-10-28:
## ❓ Questions and Help
Hi,

I want my agent to rotate about an object (let say a couch) and also maintain the camera fixation on it so that it captures the object semantics from different viewpoints. I am new to habitat, could you please let me know how can I go about doing that?

Thanks

ronjamoller, 2020-10-28:
Hi, i am not a developer so take this advice with caution - I would:
-  get the 3d positions of both the agent and the object : position of the agent: `agent_translation = env._sim.get_agent(0).scene_node.translation` or `env._sim.get_agent(0).get_state()`, which has a position and translation entry, this will depend on what env/sim you have access to and you can access inserted objects with an object id you get when you insert them, for example `env._sim.get_translation(object_id)`, same for rotation
- for simplicity's sake, just implement the rotation matrix by yourself - rotation around y axis
- multiply your position coordinates with the rotation matrix to get the desired position
- set the agent translation

- as for the rotation - looking at the object, i got this snipped from a habitat doc i think (??) i cant find it right now but it will do something like what you want, it also includes an example of setting the agent position (this might put your agent somewhere weird though, depending on your 3d environment)

```
 # turn agent toward the object
    print("turning agent toward the physics!")
    agent_state = env._sim.get_agent(0).get_state()
    agent_state.position = start_agent_position
    agent_state.position = [3.304, 0.209926, 3.86515]
    agent_to_obj = object_position - agent_state.position
    agent_local_forward = np.array([0, 0, -1.0])
    flat_to_obj = np.array([agent_to_obj[0], 0.0, agent_to_obj[2]])
    flat_dist_to_obj = np.linalg.norm(flat_to_obj)
    flat_to_obj /= flat_dist_to_obj

    # unit y normal plane for rotation
    det = (flat_to_obj[0] * agent_local_forward[2]- agent_local_forward[0] * flat_to_obj[2])
    turn_angle = math.atan2(det, np.dot(agent_local_forward, flat_to_obj))
    agent_state.rotation = quat_from_angle_axis(turn_angle, np.array([0, 1.0, 0]))

    # need to move the sensors too
    for sensor in agent_state.sensor_states:
        agent_state.sensor_states[sensor].rotation = agent_state.rotation
        agent_state.sensor_states[sensor].position = agent_state.position + np.array([0, 1.5, 0])

    env._sim.get_agent(0).set_state(agent_state)
```
I hope this is a starting point for what you want to do :)

Skylion007, 2020-10-28:
Even easier than that, you can use this utility get the rotation from origin or the forward vector fo the agent to the look at vector: 
https://github.com/facebookresearch/habitat-lab/blob/d3282b821ec1df3f598a4c1a72236e70762e61f4/habitat/utils/geometry_utils.py#L25

ayushjain1144, 2020-10-28:
Hi @Skylion007, Thank you so much for your reply. I could not completely understand your response and would highly appreciate if you could expand upon it a bit more. 

I think that I can use this utility to get the quaternion and ultimately the rotation matrix. But I am not clear how should I move the agent once I have the rotation matrix. Is it similar to what @ronjamoller  described in his answer? Or is there some another way to go about it?

Skylion007, 2020-10-28:
Yeah similar to what @ronjamoller just a nice utility function to do so.

Skylion007, 2020-10-28:
You can also do get_observations_at() with the proper args.

ayushjain1144, 2020-10-29:
Thank you @ronjamoller and @Skylion007 !!

closed 2020-10-29 by ayushjain1144