facebookresearch/habitat-sim #909: Trying to load mesh_semantic_semantic.ply instead of mesh_semantic.ply

krantiparida, 2020-10-30:
## ❓ Questions and Help
I am trying to get the data from the semantic sensor but it seems that the simulator is loading a mesh file with a different name. Hence, the loading is failing. I am correctly passing the file name in the source file. Can you please point out where I might be going wrong? The error log is mentioned below.
W1030 21:54:12.197676 15167 ResourceManager.cpp:310] ResourceManager::loadScene : Not loading semantic mesh - File Name : data/scene_datasets/replica/apartment_0/habitat/mesh_semantic_semantic.ply does not exist.

erikwijmans, 2020-10-30:
For replica, this is fine/expected.  The semantic annotations are baked into the render mesh (when not using PTex).

aclegg3, 2020-10-30:
A [stage configuration file](https://aihabitat.org/docs/habitat-sim/attributesJSON.html#stage-attributes) can also be used to explicitly specify the assets and prevent this automated path searching.

krantiparida, 2020-11-01:
Thanks, erikwijmans and aclegg3. The loading issue doesn't make any difference and the output looks fine. But the problem is while I am trying to get the semantic annotations using semantic_annotations(), the semantiLevel returns an empty list. So, there are no annotations available in this.
I am using the print_scene_recur function from [](https://aihabitat.org/docs/habitat-api/habitat-api-demo.html#scene-semantic-annotations) to display the values. Could you please let me know what might be the issue? I am doing this for Replica dataset.

erikwijmans, 2020-11-01:
There aren't level annotations for the replica dataset, just object annotations.

krantiparida, 2020-11-01:
Thanks, erikwijmans for the reply. Could you please provide the exact syntax to get the object annotations? I am trying to adapt the function available [here](https://aihabitat.org/docs/habitat-api/habitat-api-demo.html#scene-semantic-annotations) but I am unable to do.

erikwijmans, 2020-11-01:
`scene.objects` will give you the list of all the objects in the scene.

krantiparida, 2020-11-01:
I am using the following code to get the objects.
`for obj in scene.objects:
        print(
            f"Object id:{obj.id}, category:{obj.category.name()},"
            f" center:{obj.aabb.center}, dims:{obj.aabb.sizes}"
            )`
But it seems the `obj` are NoneType as I am getting the following error.
AttributeError: 'NoneType' object has no attribute 'id'

erikwijmans, 2020-11-02:
Just skip those.  The object IDs in Gibson aren't contiguous (like MP3D and replica), but it is convenient if list index == object id, so we padded Gibson with None

krantiparida, 2020-11-02:
Thanks, erikwijmans for the reply. I think there is confusion. I am using Replica, not Gibson. But still, I am getting None.

krantiparida, 2020-11-02:
It seems the objects ids aren't contiguous for Replica as well. I skipped the None type id and still, there was some None type in the semantic category as well. After skipping both of them, I could get the category name. Here is the workaround which worked for me.
`for obj in scene_annot.objects:
                    if obj != None:
                        if obj.category != None:
                            print(f"Object id:{obj.id}, category:{obj.category.name()},"
                            f" center:{obj.aabb.center}, dims:{obj.aabb.sizes}"
                            )`
Thanks, erikwijmans for the help.

krantiparida, 2020-11-02:
I a confused if I am doing everything correctly. The semantic ouptut and the labels I am getting doesn't seem to match. I am getting the semantic map, rgb image and depth as displayed below. But I am getting a long list of semantic labels for the output which seems to not appear in the image. The labels are:
{'faucet', 'rug', 'basket', 'cushion', 'refrigerator', 'indoor-plant', 'blinds', 'pillow', 'mat', 'sink', 'door', 'window', 'lamp', 'comforter', 'wall', 'toilet', 'switch', 'bed', 'wall-plug', 'bathtub', 'book', 'shower-stall', 'picture', 'floor', 'cooktop', 'panel', 'bottle', 'desk', 'vase', 'cabinet', 'major-appliance', 'pot', 'nightstand', 'cup', 'countertop', 'wall-cabinet', 'handrail', 'stool', 'rack', 'ceiling', 'table', 'plate', 'chair', 'vent', 'base-cabinet', 'stair', 'sofa', 'plant-stand', 'curtain'}
[image]
Please guide me if I am interpreting something incorrectly.

erikwijmans, 2020-11-02:
How are you getting the labels?  `scene.objects` is not updated based on current agent position.

krantiparida, 2020-11-02:
I see. I might be getting the wrong labels. By the way, I am getting the labels by accumulating all the object's `category.name()` from the `scene.objects`. What are these labels signify then?

If I want to get the labels of the object only that the agent can see depending upon the current position, what should I do?

erikwijmans, 2020-11-02:
1. Those are all the labels of all the objects in the entire environment
2. `object_ids_in_view = np.unique(semantic_sensor_output)`

krantiparida, 2020-11-02:
Many Thanks, Erik!
I think I can get the required labels now.

closed 2020-11-02 by krantiparida