facebookresearch/habitat-sim #910: Instance Id to Semantic labels mapping in Gibson

Singh-sid930, 2020-10-31:
I am trying to do the same thing as here. ( https://github.com/facebookresearch/habitat-sim/issues/760) With the gibson data set. 
Trying to create the semantic map using the semantic information from scene. 
I am following this tutorial https://aihabitat.org/docs/habitat-api/habitat-sim-demo.html#scene-semantic-annotations
But somehow the `semantic_scene.objects()` comes out to be a none value. 

Is the mapping available somewhere else through the code?

erikwijmans, 2020-11-01:
That shouldn't be empty.  Can you turn on logging? i.e.
```
unset GLOG_minloglevel MAGNUM_LOG
export GLOG_v=1
```

KenaHemnani, 2021-03-22:
> I am trying to do the same thing as here. ( #760) With the gibson data set.
> Trying to create the semantic map using the semantic information from scene.
> I am following this tutorial [...]

Hii, was that possible to create semantic map using the semantic information from scene. How did you do it, can you tell me in brief . I am also attempting similar task but via. egocentric RGBD images. So, just want to know. Thank you.

Singh-sid930, 2021-03-29:
We made use of the .glb, .scn and .id files  and stored the relevant information in npz files. 
Something like this: 
You might have to take care of the paths to the data set files. 
```
    import habitat_sim
    from plyfile import PlyData, PlyElement
    import json
    sim_cfg = habitat_sim.SimulatorConfiguration()
    agent_cfg = habitat_sim.AgentConfiguration()

    scenes_dir = "/<path>/habitat-api/data/scene_datasets/mp3d/"
    preprocessed_scenes_dir = "/<path>/data/scene_datasets/mp3d_scene_pclouds/"
   
    scene_list = os.listdir(scenes_dir)
  

    for i in range(len(scene_list)):
        scene_id = scene_list[i]
        print('Scene:', i, scene_id)

        if os.path.exists(preprocessed_scenes_dir+scene_id+'_pcloud.npz'):
            continue

        glb_path = scenes_dir + scene_id + '/' + scene_id + '.glb' # mp3d case
        scn_path = scenes_dir + scene_id + ".scn"
        ids_path = scenes_dir + scene_id + ".ids"

        sim_cfg.scene.id = glb_path
        sim = habitat_sim.Simulator(habitat_sim.Configuration(sim_cfg, [agent_cfg]))

        scene = sim.semantic_scene
        # Load .scn and .ids files
        #scn = np.load(scn_path)
        ids_data = np.fromfile(ids_path, dtype = np.int16)

        scn_data = json.load(open(scn_path, "r"))
        ply_path = scenes_dir + scene_id + '_semantic.ply' # gibson case
        data = PlyData.read(ply_path)

        instance_id_to_label_id = {int(obj.id.split("_")[-1]): obj.category.index() for obj in scene.objects}

        # store the pcloud geometry and semantic labels
        x, y, z = data['vertex']['x'], data['vertex']['y'], data['vertex']['z']
        object_id = data['face']['object_id']
        object_id = np.asarray(object_id)
        object_id = np.repeat(object_id,3)
        vertex_indices = data['face']['vertex_indices']
        vertex_indices = np.concatenate(vertex_indices).ravel()
        label_id = np.vectorize(instance_id_to_label_id.get)(object_id)
        label_seq = np.zeros(len(x))
        label_seq[vertex_indices] = label_id
        label_seq = np.asarray(label_seq)
        label_seq = label_seq.reshape((len(label_seq),1))
        np.savez_compressed(preprocessed_scenes_dir+scene_id+'_pcloud.npz', x=x, y=y, z=z, label_seq=label_seq)

        # store the pcloud color info for visualization
        r, g, b = data['vertex']['red'], data['vertex']['green'], data['vertex']['blue']
        r_seq = np.ones(len(x))*255
        g_seq = np.ones(len(x))*255
        b_seq = np.ones(len(x))*255
        col_map = utils.color_mapping
        for color_label in col_map:
            color = col_map[color_label]
            r_seq[vertex_indices[np.where(label_id == color_label)]]= color[0]
            g_seq[vertex_indices[np.where(label_id == color_label)]]= color[1]
            b_seq[vertex_indices[np.where(label_id == color_label)]]= color[2]
        np.savez_compressed(preprocessed_scenes_dir+scene_id+'_color.npz', r=r, g=g, b=b, r_seq=r_seq, g_seq=g_seq, b_seq=b_seq)
        data=None
        sim.close()
        
        ```

closed 2021-05-26 by dhruvbatra