facebookresearch/habitat-sim #915: Getting 2D box in pixel coordinates from obb box

ayushjain1144, 2020-11-04:
## ❓ Questions and Help
Hi,

I want to obtain a 2D bounding box of the objects captured by habitat sensors. I couldn't find a direct way to save this, so I saved the obb box center, obb box size, and obb local_to_world. If there is a more direct/cleaner way, please let me know. I am specifically using replica dataset.

Here are the steps I followed to get the 2D box:

1. Generate an (8, 3) xyz corners of the box in the following way:
```
lx, ly, lz = torch.unbind(bbox_size, axis=2)
xs = torch.stack([bbox_center[0]-lx/2., bbox_center[0]-lx/2., bbox_center[0]-lx/2., bbox_center[0]-lx/2., bbox_center[0]+lx/2., 
                                    bbox_center[0]+lx/2., bbox_center[0]+lx/2., bbox_center[0]+lx/2.], axis=2)
ys = torch.stack([bbox_center[1]-ly/2., bbox_center[1]-ly/2., bbox_center[1]+ly/2., bbox_center[1]+ly/2., bbox_center[1]-ly/2., 
                                     bbox_center[1]-ly/2., bbox_center[1]+ly/2., bbox_center[1]+ly/2.], axis=2)
zs = torch.stack([bbox_center[2]-lz/2., bbox_center[2]+lz/2., bbox_center[2]-lz/2., bbox_center[2]+lz/2., bbox_center[2]-lz/2., 
                                     bbox_center[2]+lz/2., bbox_center[2]-lz/2., bbox_center[2]+lz/2.], axis=2)
```
2. Transforming it to world coordinate by applying matrix multiplying local_to_world to the above corners.
3. Using intrinsics to project the box to 2D

I am getting the following bounding box which is incorrect. It should have surrounded the tv (the red blob)
[image]

Could you please let me know how can I fix this or get the 2D box in another way? Thanks a lot!

Skylion007, 2020-11-04:
You don't even need to do anything that complicated.

Just used the SemanticSensor, find the instanceId of the object you want to draw a bounding box around and you can do it in a few lines of numpy

```(sensor_obs == object_id).nonzero()``` then just use np.min and np.max to find the min/max of the Y and X coordinates.

ayushjain1144, 2020-11-06:
Thanks, @Skylion007, that solved my issue. But I am still struggling a bit to get tight 3D bounding box around the objects. Apart from the actual 3D box, I would like to project it to pixel space too to get a 3D box in 2D as well. 

I capture the OBB box from the sensors. I then build a 3D box using the centre and obb box sizes to get (8, 3) coordinates. I then multiply it by local_to_world coordinates. I assume this should have given me the 3D box right? Or I am misunderstanding something?

erikwijmans, 2020-11-06:
The `local_to_world` rotation needs to be applied before the center offset is added -- i.e. the rotation assumes the center of the box is at (0, 0, 0).

closed 2020-11-18 by ayushjain1144