facebookresearch/habitat-sim #920: Tilt the sensor up/down without moving the agent position

zfang399, 2020-11-06:
## ❓ Questions and Help
Hi! Thank you so much for the great simulator.

I'm trying to set the agent so that the center of its view (the center of the 2D image it captures) is aligned with the center of a 3D object, without moving the agent's location, but just make the sensors look up or down. Is there a function that could do this? 

Thanks a lot!!

aclegg3, 2020-11-06:
Hey @zfang399, sounds like you may want to directly modify the rotation of the SceneNode for the sensor attached to the agent. If you just care about getting the image you could use a lookAt function as in [this example](https://colab.research.google.com/github/facebookresearch/habitat-sim/blob/master/examples/tutorials/colabs/ECCV_2020_Advanced_Features.ipynb#scrollTo=3XoaXaXNnqpU) to track the object with the camera.

If you want the agent itself to be facing the object (not just the sensor), you could first compute the necessary rotation for the agent as in [this example](https://github.com/facebookresearch/habitat-sim/blob/706b23f823cd83b368438fe7f475a89b6ee179ef/examples/demo_runner.py#L130) and then apply the lookAt (or compute another similar angle axis rotation) to tilt the camera.

zfang399, 2020-11-06:
@aclegg3 Thank you so much for your help! I think I get it now!

closed 2020-11-06 by zfang399