facebookresearch/habitat-sim #925: How can we render a normal map?

Cenbylin, 2020-11-10:
## ❓ Questions and Help

Thanks for the useful platform and your help.

I am looking for a way to obtain a normal map rendered by the camera. Is there an existing sensor? Or maybe we need to customize a shader.

Skylion007, 2020-11-11:
Can you clarify what you mean by a normal map? Are you talking about screen-space normals?

Cenbylin, 2020-11-11:
@Skylion007 Thank you so much. I think it would be better to have an example:

The following three figures are Colour, Depth (not standard here) and **Normal map**. 

In the Normal map, those RGB values are "x,y,z components" of the 3D normal vectors on the object surface.

[image]

I am a beginner in this field and hope my description is clear enough 😀.

erikwijmans, 2020-11-11:
We don't expose the normal map.  However, surface normals can be approximated as the first derivative of the depth map (like here: https://github.com/danielgordon10/dg_util/blob/master/dg_util/python_utils/pytorch_util.py#L872).  Depending on your use, approximate normals may be fine.  The ones from the renderer can be exposed, but this requires changes to the shaders.

Skylion007, 2020-11-11:
If you do a create a useful utility functions for getting approximate normals, feel free to open a PR.

closed 2021-03-01 by dhruvbatra