facebookresearch/habitat-sim #945: Instance Id in semanctic sensor

krantiparida, 2020-11-22:
## ❓ Questions and Help
Hi,
I got the semantic label using the code below.
instance_id_to_label_id = {int(obj.id.split("_")[-1]): obj.category.index() for obj in scene.objects}
However, I noticed the instance id for the object changes every scene. e.g. the semantic label 'wall' is 129 in one scene whereas it is '134' in another. Is the mapping not consistent?

erikwijmans, 2020-11-22:
Many different objects (where an object will specified by its instance ID) will have the same semantic category, so a given semantic label cannot map to a single instance ID.  The instance IDs are not consistent across scenes because different scenes have different objects.  So no, the mapping from semantic label to instance ID is not consistent because this is not possible.

Skylion007, 2020-11-23:
@krantiparida Yeah, you need to rebuilt this mapping for every scene that you load.

krantiparida, 2020-11-26:
Thanks for the clarification.

closed 2020-11-26 by krantiparida