facebookresearch/habitat-sim #947: Import error while importing habitat_sim

rutadesai, 2020-11-24:
## 🐛 Bug

After installing habitat-lab and habitat-sim based on the guidelines in the respective repos, I get the following error while importing habitat_sim from within habitat-lab folder in a conda environment:
`    from habitat_sim._ext.habitat_sim_bindings import (  # noqa: F401 isort:skip`
`ImportError: libBulletDynamics.so.2.88: cannot open shared object file: No such file or directory`

## Command
` import habitat_sim`

## Additional context

- Habitat-sim was installed using conda stable build: `conda install habitat-sim withbullet headless -c conda-forge -c aihabitat`
- Packages in conda environment:
````
# packages in environment at /home/rutadesai/anaconda3/envs/habitat:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                        main
absl-py                   0.11.0                   pypi_0    pypi
astor                     0.8.1                    pypi_0    pypi
attrs                     20.3.0             pyhd3deb0d_0    conda-forge
backcall                  0.2.0                      py_0    anaconda
bullet                    2.89             py36h7c3b610_1    conda-forge
bzip2                     1.0.8                h7b6447c_0
ca-certificates           2020.10.14                    0    anaconda
cached-property           1.5.2                    pypi_0    pypi
cachetools                4.1.1                    pypi_0    pypi
certifi                   2020.6.20                py36_0    anaconda
chardet                   3.0.4                    pypi_0    pypi
cmake                     3.14.0               h52cb24c_0
cycler                    0.10.0                     py_2    conda-forge
dataclasses               0.8                      pypi_0    pypi
dbus                      1.13.6               hfdff14a_1    conda-forge
decorator                 4.4.2                    pypi_0    pypi
expat                     2.2.10               he6710b0_2
ffmpeg                    4.3.1                h3215721_1    conda-forge
fontconfig                2.13.1            h86ecdb6_1001    conda-forge
freetype                  2.10.4               h7ca028e_0    conda-forge
future                    0.18.2                   pypi_0    pypi
gast                      0.4.0                    pypi_0    pypi
gettext                   0.19.8.1          hf34092f_1004    conda-forge
gitdb                     4.0.5                      py_0    conda-forge
gitpython                 3.1.11                     py_0    conda-forge
glib                      2.66.3               h58526e2_0    conda-forge
gmp                       6.2.1                h58526e2_0    conda-forge
gnutls                    3.6.13               h85f3911_1    conda-forge
google-auth               1.23.0                   pypi_0    pypi
google-auth-oauthlib      0.4.2                    pypi_0    pypi
grpcio                    1.34.0rc1                pypi_0    pypi
gst-plugins-base          1.14.5               h0935bb2_2    conda-forge
gstreamer                 1.14.5               h36ae1b5_2    conda-forge
gym                       0.10.9                   pypi_0    pypi
h5py                      3.1.0                    pypi_0    pypi
habitat                   0.1.6                     dev_0    <develop>
habitat-sim               0.1.6                    pypi_0    pypi
headless                  1.0                           0    aihabitat
icu                       64.2                 he1b5a44_1    conda-forge
idna                      2.10                     pypi_0    pypi
ifcfg                     0.21                     pypi_0    pypi
imageio                   2.9.0                      py_0    conda-forge
imageio-ffmpeg            0.4.2                      py_0    conda-forge
importlib-metadata        3.0.0                    pypi_0    pypi
ipykernel                 5.3.4            py36h5ca1d4c_0    anaconda
ipython                   7.16.1           py36h5ca1d4c_0    anaconda
ipython_genutils          0.2.0                    py36_0    anaconda
jedi                      0.17.2                   py36_0    anaconda
jpeg                      9d                   h36c2ea0_0    conda-forge
jupyter_client            6.1.7                      py_0    anaconda
jupyter_core              4.6.3                    py36_0    anaconda
keras-applications        1.0.8                    pypi_0    pypi
[... 159 lines cut ...]
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
|   3  Tesla K80           On   | 00000000:00:1A.0 Off |                    0 |
| N/A   30C    P8    29W / 149W |      3MiB / 11441MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
|   4  Tesla K80           On   | 00000000:00:1B.0 Off |                    0 |
| N/A   34C    P8    27W / 149W |      3MiB / 11441MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
|   5  Tesla K80           On   | 00000000:00:1C.0 Off |                    0 |
| N/A   28C    P8    29W / 149W |      3MiB / 11441MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
|   6  Tesla K80           On   | 00000000:00:1D.0 Off |                    0 |
| N/A   38C    P8    27W / 149W |      3MiB / 11441MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
|   7  Tesla K80           On   | 00000000:00:1E.0 Off |                    0 |
| N/A   33C    P8    31W / 149W |      3MiB / 11441MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
Pip packages versions:

Conda packages versions:

rutadesai, 2020-11-24:
I tried installing the nightly build of habitat-sim and that worked!

closed 2020-11-24 by rutadesai

shhuangcoder, 2024-04-07:
> I tried installing the nightly build of habitat-sim and that worked!

what's this?