facebookresearch/habitat-sim #948: Wrong category for semantic_scene.object

fraazor, 2020-11-25:
## ❓ Questions and Help
Hello again!

I am not sure whether I am dealing with an error of mine, an error in habitat sim or if the mp3 annotation itself is faulty.
The scene is 17DRP5sb8fy.
I am building a semantic map  on top of the gt maps of the occupancy anticipation paper.
For the semantic masks, I am using the position/dimension of objects from the semantic_scene. I built this dic that I use to map the objects.

scene = env.habitat_env.sim.semantic_scene
obj_pos = {}
for name in label_idx.keys():
    obj_pos[name] = {obj.id : [obj.aabb.center,obj.aabb.sizes]
                        for obj in scene.objects
                        if name == obj.category.name()}

For the key "tables" I receive four obj. 3 of them working out fine but one of them does not seem to exist in reality. That table is in the middle of the room where there is supposed to be nothing. The only thing the bb could possibly capture is the carpet. I have a similar situation with one "chair" object where once again, most chairs are fine, but again one shows possibly in place of a carpet.

This scene also only appears to have one level so I don't think it comes from a lower or upper floor.

[image]

[image]

erikwijmans, 2020-11-25:
For the first thing, that is due to a difference in reconstruction for the mesh used for RGB rendering and the mesh with semantic labels.  The mesh with semantic labels is generally a much worse reconstruction, but sometimes it has things that the RGB mesh doesn't.
[image]

For the second case, there is something there in that location:
[image]
I would probably call that a table tho.

In general, the annotations in the mp3d dataset are not perfect.  They are pretty good but not perfect.

fraazor, 2020-11-25:
Thank you again Erik! The second table is supossed to be a chair but I assume it categorized the couch as a chair. It seems all good then.

closed 2021-05-26 by dhruvbatra

vincent341, 2022-02-24:
> ## ❓ Questions and Help
> Hello again!
> 
> I am not sure whether I am dealing with an error of mine, an error in habitat sim or if the mp3 annotation itself is faulty. The scene is 17DRP5sb8fy. I am building a semantic map on top of the gt maps of the occupancy anticipation paper. For the semantic masks, I am using the position/dimension of objects from the semantic_scene. I built this dic that I use to map the objects.
> 
> scene = env.habitat_env.sim.semantic_scene obj_pos = {} for name in label_idx.keys(): obj_pos[name] = {obj.id : [obj.aabb.center,obj.aabb.sizes] for obj in scene.objects if name == obj.category.name()}
> 
> For the key "tables" I receive four obj. 3 of them working out fine but one of them does not seem to exist in reality. That table is in the middle of the room where there is supposed to be nothing. The only thing the bb could possibly capture is the carpet. I have a similar situation with one "chair" object where once again, most chairs are fine, but again one shows possibly in place of a carpet.
> 
> The table object in doubt has the id "0_7_132".
> 
> This scene also only appears to have one level so I don't think it comes from a lower or upper floor.
> 
> [image]
> 
> [image]

Hi. I'm attempting to create 2D top-down semantic maps for MP3D. I know this link was some time ago. Would you mind giving more details on how you created the 2D top-down semantic maps? Great thanks!

vincentcartillier, 2022-02-24:
Hi @vincent341 , you can check the following project: [Semantic-MapNet](https://github.com/vincentcartillier/Semantic-MapNet) in which we built such top-down semantic maps for MP3D.

Specifically you should check the following scripts:
```
data/build_point_cloud_from_mesh_h5.py
compute_GT_topdown_semantic_maps/build_semmap_from_obj_point_cloud.py
```

vincent341, 2022-02-25:
> Hi @vincent341 , you can check the following project: [Semantic-MapNet](https://github.com/vincentcartillier/Semantic-MapNet) in which we built such top-down semantic maps for MP3D.
>
> Specifically [...]

@vincentcartillier Thanks so much for the link. You saved my day!