facebookresearch/habitat-sim #950: Build error

Mingxiao-Li, 2020-11-27:
## ❓ Questions and Help
Hi, I came across below error when I tried to  build the hiabitat-sim branch:  ff62b6dddcee04a6f5a065e4ae2d093b37f95744 :
subprocess.CalledProcessError: Command '['cmake', '-H/export/home1/NoCsBack/hci/mingxiao/R2R_ORIGIN/habitat-sim/src', '-Bbuild', '-DBUILD_PYTHON_BINDINGS=ON', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/export/home1/NoCsBack/hci/mingxiao/R2R_ORIGIN/habitat-sim/build/lib.linux-x86_64-3.6/habitat_sim/_ext', '-DPYTHON_EXECUTABLE=/export/home1/NoCsBack/hci/mingxiao/miniconda3/envs/vlnce_origin/bin/python', '-DCMAKE_EXPORT_COMPILE_COMMANDS=ON', '-DCMAKE_BUILD_TYPE=RelWithDebInfo', '-GNinja', '-DBUILD_GUI_VIEWERS=ON', '-DTARGET_HEADLESS=OFF', '-DBUILD_TEST=OFF', '-DBUILD_WITH_BULLET=OFF', '-DBUILD_DATATOOL=OFF', '-DBUILD_WITH_CUDA=OFF']' returned non-zero exit status 1.

The gcc version of my system is gcc 7.5.0 and the cmake version is 1.18.2. 
Do someone know how to solve this error  ? Thanks in advance.

erikwijmans, 2020-11-27:
Please include more of the error log.  This currently only shows that an error happened, not what caused it.

Mingxiao-Li, 2020-11-27:
OK。 Below are the details:
Requirement already satisfied: attrs>=19.1.0 in /export/home1/NoCsBack/hci/mingxiao/miniconda3/lib/python3.7/site-packages (from -r requirements.txt (line 1)) (19.3.0)
Requirement already satisfied: numba in /export/home1/NoCsBack/hci/mingxiao/miniconda3/lib/python3.7/site-packages/numba-0.51.2-py3.7-linux-x86_64.egg (from -r requirements.txt (line 2)) (0.51.2)
Requirement already satisfied: numpy in /export/home1/NoCsBack/hci/mingxiao/miniconda3/lib/python3.7/site-packages (from -r requirements.txt (line 3)) (1.18.1)
Requirement already satisfied: numpy-quaternion in /export/home1/NoCsBack/hci/mingxiao/miniconda3/lib/python3.7/site-packages/numpy_quaternion-2021.0.0a0-py3.7-linux-x86_64.egg (from -r requirements.txt (line 4)) (2021.0.0a0)
Requirement already satisfied: pillow in /export/home1/NoCsBack/hci/mingxiao/miniconda3/lib/python3.7/site-packages (from -r requirements.txt (line 5)) (6.2.0)
Requirement already satisfied: scipy>=1.3.0 in /export/home1/NoCsBack/hci/mingxiao/miniconda3/lib/python3.7/site-packages (from -r requirements.txt (line 6)) (1.4.1)
Requirement already satisfied: tqdm in /home/mingxiao/.local/lib/python3.7/site-packages (from -r requirements.txt (line 7)) (4.46.1)
Requirement already satisfied: llvmlite<0.35,>=0.34.0.dev0 in /export/home1/NoCsBack/hci/mingxiao/miniconda3/lib/python3.7/site-packages/llvmlite-0.34.0-py3.7-linux-x86_64.egg (from numba->-r requirements.txt (line 2)) (0.34.0)
Requirement already satisfied: setuptools in /export/home1/NoCsBack/hci/mingxiao/miniconda3/lib/python3.7/site-packages (from numba->-r requirements.txt (line 2)) (47.3.1)
mingxiao@rose:/export/home1/NoCsBack/hci/mingxiao/R2R_ORIGIN/habitat-sim$ python setup.py install
running install
running bdist_egg
running egg_info
writing habitat_sim.egg-info/PKG-INFO
writing dependency_links to habitat_sim.egg-info/dependency_links.txt
writing requirements to habitat_sim.egg-info/requires.txt
writing top-level names to habitat_sim.egg-info/top_level.txt
reading manifest file 'habitat_sim.egg-info/SOURCES.txt'
writing manifest file 'habitat_sim.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
Traceback (most recent call last):
  File "setup.py", line 194, in run
    subprocess.check_output(["cmake", "--version"])
  File "/export/home1/NoCsBack/hci/mingxiao/miniconda3/lib/python3.7/subprocess.py", line 395, in check_output
    **kwargs).stdout
  File "/export/home1/NoCsBack/hci/mingxiao/miniconda3/lib/python3.7/subprocess.py", line 472, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/export/home1/NoCsBack/hci/mingxiao/miniconda3/lib/python3.7/subprocess.py", line 775, in __init__
    restore_signals, start_new_session)
  File "/export/home1/NoCsBack/hci/mingxiao/miniconda3/lib/python3.7/subprocess.py", line 1522, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'cmake': 'cmake'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "setup.py", line 379, in <module>
    zip_safe=False,
  File "/export/home1/NoCsBack/hci/mingxiao/miniconda3/lib/python3.7/site-packages/setuptools/__init__.py", line 161, in setup
    return distutils.core.setup(**attrs)
  File "/export/home1/NoCsBack/hci/mingxiao/miniconda3/lib/python3.7/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/export/home1/NoCsBack/hci/mingxiao/miniconda3/lib/python3.7/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/export/home1/NoCsBack/hci/mingxiao/miniconda3/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/export/home1/NoCsBack/hci/mingxiao/miniconda3/lib/python3.7/site-packages/setuptools/command/install.py", line 67, in run
    self.do_egg_install()
  File "/export/home1/NoCsBack/hci/mingxiao/miniconda3/lib/python3.7/site-packages/setuptools/command/install.py", line 109, in do_egg_install
    self.run_command('bdist_egg')
  File "/export/home1/NoCsBack/hci/mingxiao/miniconda3/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/export/home1/NoCsBack/hci/mingxiao/miniconda3/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/export/home1/NoCsBack/hci/mingxiao/miniconda3/lib/python3.7/site-packages/setuptools/command/bdist_egg.py", line 173, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/export/home1/NoCsBack/hci/mingxiao/miniconda3/lib/python3.7/site-packages/setuptools/command/bdist_egg.py", line 159, in call_command
    self.run_command(cmdname)
  File "/export/home1/NoCsBack/hci/mingxiao/miniconda3/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/export/home1/NoCsBack/hci/mingxiao/miniconda3/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/export/home1/NoCsBack/hci/mingxiao/miniconda3/lib/python3.7/site-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/export/home1/NoCsBack/hci/mingxiao/miniconda3/lib/python3.7/distutils/command/install_lib.py", line 107, in build
    self.run_command('build_ext')
  File "/export/home1/NoCsBack/hci/mingxiao/miniconda3/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/export/home1/NoCsBack/hci/mingxiao/miniconda3/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "setup.py", line 198, in run
    + ", ".join(e.name for e in self.extensions)
RuntimeError: CMake must be installed to build the following extensions: habitat_sim._ext.habitat_sim_bindings

Mingxiao-Li, 2020-11-27:
Hi. Below are the details:
running install
running bdist_egg
running egg_info
writing habitat_sim.egg-info/PKG-INFO
writing dependency_links to habitat_sim.egg-info/dependency_links.txt
writing requirements to habitat_sim.egg-info/requires.txt
writing top-level names to habitat_sim.egg-info/top_level.txt
reading manifest file 'habitat_sim.egg-info/SOURCES.txt'
writing manifest file 'habitat_sim.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
error: [Errno 2] No such file or directory: 'build/compile_commands.json'
Thanks.

erikwijmans, 2020-11-27:
Please add `--cmake` to your build command, seems like the error happened in cmake so it needs to be forced to re-run

Mingxiao-Li, 2020-11-27:
Thanks. But when I added --cmake to build command, I got another error shown below :
running install
running bdist_egg
running egg_info
writing habitat_sim.egg-info/PKG-INFO
writing dependency_links to habitat_sim.egg-info/dependency_links.txt
writing requirements to habitat_sim.egg-info/requires.txt
writing top-level names to habitat_sim.egg-info/top_level.txt
reading manifest file 'habitat_sim.egg-info/SOURCES.txt'
writing manifest file 'habitat_sim.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
-- Shared libraries disabled
-- Looking for ZLIB...
-- Checking for module 'zzip-zlib-config'
--   No package 'zzip-zlib-config' found
-- Found ZLIB: optimized;/usr/lib/x86_64-linux-gnu/libz.so;debug;/usr/lib/x86_64-linux-gnu/libz.so
-- Checking for module 'minizip'
--   No package 'minizip' found
CMake Warning (dev) at /export/home1/NoCsBack/hci/mingxiao/miniconda3/envs/vlnce_origin/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:273 (message):
  The package name passed to `find_package_handle_standard_args` (rt) does
  not match the name of the calling package (RT).  This can lead to problems
  in calling code that expects `find_package` result variables (e.g.,
  `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  deps/assimp/cmake-modules/FindRT.cmake:19 (find_package_handle_standard_args)
  deps/assimp/code/CMakeLists.txt:842 (FIND_PACKAGE)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Enabled formats: AMF 3DS AC ASE ASSBIN ASSXML B3D BVH COLLADA DXF CSM HMP IRRMESH IRR LWO LWS MD2 MD3 MD5 MDC MDL NFF NDO OFF OBJ OGRE OPENGEX PLY MS3D COB BLEND IFC XGL FBX Q3D Q3BSP RAW SIB SMD STL TERRAGEN 3D X X3D GLTF 3MF MMD
-- Disabled formats:
-- Found PythonInterp: /export/home1/NoCsBack/hci/mingxiao/miniconda3/envs/vlnce_origin/bin/python (found suitable version "3.6.12", minimum required is "3.6")
-- Bindings being generated for python at /export/home1/NoCsBack/hci/mingxiao/miniconda3/envs/vlnce_origin/bin/python
-- Found PythonInterp: /export/home1/NoCsBack/hci/mingxiao/miniconda3/envs/vlnce_origin/bin/python (found version "3.6.12")
-- pybind11 v2.3.0
-- Could NOT find Vulkan (missing: VULKAN_LIBRARY VULKAN_INCLUDE_DIR)
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
-- Using X11 for window creation
CMake Error at /export/home1/NoCsBack/hci/mingxiao/miniconda3/envs/vlnce_origin/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:165 (message):
  Could NOT find X11 (missing: X11_X11_INCLUDE_PATH X11_X11_LIB)
Call Stack (most recent call first):
  /export/home1/NoCsBack/hci/mingxiao/miniconda3/envs/vlnce_origin/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:458 (_FPHSA_FAILURE_MESSAGE)
  /export/home1/NoCsBack/hci/mingxiao/miniconda3/envs/vlnce_origin/share/cmake-3.18/Modules/FindX11.cmake:414 (find_package_handle_standard_args)
  deps/glfw/CMakeLists.txt:218 (find_package)

-- Configuring incomplete, errors occurred!
See also "/export/home1/NoCsBack/hci/mingxiao/R2R_ORIGIN/habitat-sim/build/CMakeFiles/CMakeOutput.log".
See also "/export/home1/NoCsBack/hci/mingxiao/R2R_ORIGIN/habitat-sim/build/CMakeFiles/CMakeError.log".
Traceback (most recent call last):
  File "setup.py", line 379, in <module>
    zip_safe=False,
  File "/export/home1/NoCsBack/hci/mingxiao/miniconda3/envs/vlnce_origin/lib/python3.6/site-packages/setuptools/__init__.py", line 153, in setup
    return distutils.core.setup(**attrs)
  File "/export/home1/NoCsBack/hci/mingxiao/miniconda3/envs/vlnce_origin/lib/python3.6/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/export/home1/NoCsBack/hci/mingxiao/miniconda3/envs/vlnce_origin/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/export/home1/NoCsBack/hci/mingxiao/miniconda3/envs/vlnce_origin/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/export/home1/NoCsBack/hci/mingxiao/miniconda3/envs/vlnce_origin/lib/python3.6/site-packages/setuptools/command/install.py", line 67, in run
    self.do_egg_install()
  File "/export/home1/NoCsBack/hci/mingxiao/miniconda3/envs/vlnce_origin/lib/python3.6/site-packages/setuptools/command/install.py", line 109, in do_egg_install
    self.run_command('bdist_egg')
  File "/export/home1/NoCsBack/hci/mingxiao/miniconda3/envs/vlnce_origin/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/export/home1/NoCsBack/hci/mingxiao/miniconda3/envs/vlnce_origin/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/export/home1/NoCsBack/hci/mingxiao/miniconda3/envs/vlnce_origin/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 167, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/export/home1/NoCsBack/hci/mingxiao/miniconda3/envs/vlnce_origin/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 153, in call_command
    self.run_command(cmdname)
  File "/export/home1/NoCsBack/hci/mingxiao/miniconda3/envs/vlnce_origin/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/export/home1/NoCsBack/hci/mingxiao/miniconda3/envs/vlnce_origin/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
[... cut ...]
subprocess.CalledProcessError: Command '['cmake', '-H/export/home1/NoCsBack/hci/mingxiao/R2R_ORIGIN/habitat-sim/src', '-Bbuild', '-DBUILD_PYTHON_BINDINGS=ON', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/export/home1/NoCsBack/hci/mingxiao/R2R_ORIGIN/habitat-sim/build/lib.linux-x86_64-3.6/habitat_sim/_ext', '-DPYTHON_EXECUTABLE=/export/home1/NoCsBack/hci/mingxiao/miniconda3/envs/vlnce_origin/bin/python', '-DCMAKE_EXPORT_COMPILE_COMMANDS=ON', '-DCMAKE_BUILD_TYPE=RelWithDebInfo', '-GNinja', '-DBUILD_GUI_VIEWERS=ON', '-DTARGET_HEADLESS=OFF', '-DBUILD_TEST=OFF', '-DBUILD_WITH_BULLET=OFF', '-DBUILD_DATATOOL=OFF', '-DBUILD_WITH_CUDA=OFF']' returned non-zero exit status 1.

erikwijmans, 2020-11-27:
Looks like you are missing x11 headers.  You can get those via `sudo apt install libx11-dev` or possibly this conda package: https://anaconda.org/conda-forge/libx11-devel-cos7-x86_64 (I haven't personally tried that conda package, but it is just headers so should be fine).

As a side node, x11 headers missing typically indicates that this is a remote server, not a machine with an attached display.  If that is the case, you will need to build with `--headless` (and then x11 headers won't be needed anymore).

Mingxiao-Li, 2020-11-28:
I installed x11 and also tried the command with --headless. But then I got another error:
running install
running bdist_egg
running egg_info
writing habitat_sim.egg-info/PKG-INFO
writing dependency_links to habitat_sim.egg-info/dependency_links.txt
writing requirements to habitat_sim.egg-info/requires.txt
writing top-level names to habitat_sim.egg-info/top_level.txt
reading manifest file 'habitat_sim.egg-info/SOURCES.txt'
writing manifest file 'habitat_sim.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
error: [Errno 2] No such file or directory: 'build/compile_commands.json'

erikwijmans, 2020-11-29:
Looks like cmake is still not cooperating.  Please clear the build folder with `rm -rf build` and try again.

closed 2021-03-01 by dhruvbatra