facebookresearch/habitat-sim #951: Rigid Body tutorial example throws error

abhishek-tcs-rni, 2020-11-30:
## ❓ Questions and Help

I have installed `habitat-sim` as per given [instructions](https://github.com/facebookresearch/habitat-sim#installation), and also successfully performed [testing](https://github.com/facebookresearch/habitat-sim#testing), including extraction of pre-processed object dataset to `habitat-sim/data/objects/`.

Corresponding to [Interactive Rigid Objects](https://aihabitat.org/docs/habitat-sim/rigid-object-tutorial.html) tutorial, on running given example script in installed `conda` environment in `habitat-sim` directory
```
python examples/tutorials/nb_python/rigid_object_tutorial.py
```
the following error is displayed
```
Simulating 1.5 world seconds.
Encoding the video: examples/tutorials/rigid_object_tutorial_output/sim_basics.mp4 
VLC media player 3.0.8 Vetinari (revision 3.0.8-0-gf350b6b5a7)
Traceback (most recent call last):
  File "examples/tutorials/nb_python/rigid_object_tutorial.py", line 196, in <module>
    )[0]
IndexError: list index out of range
```

It appears the [statement](https://github.com/facebookresearch/habitat-sim/blob/0aea76a27eb70ba5cf80be9b847557f5857f09de/examples/tutorials/nb_python/rigid_object_tutorial.py#L193) loading the template for object, `cheezit` using the instruction, [`obj_templates_mgr.get_template_handles`](https://aihabitat.org/docs/habitat-sim/habitat_sim.attributes_managers.StageAttributesManager.html#get_template_handles) is throwing the above error.

Please help in resolving above error, to run the complete [example script](https://github.com/facebookresearch/habitat-sim/blob/master/examples/tutorials/nb_python/rigid_object_tutorial.py) corresponding to [Rigid Body tutorial](https://aihabitat.org/docs/habitat-sim/rigid-object-tutorial.html) without errors.

aclegg3, 2020-11-30:
Hey @abhishek-tcs-rni, 

This tutorial is working on my system and [the Collab](https://colab.research.google.com/github/facebookresearch/habitat-sim/blob/master/examples/tutorials/colabs/rigid_object_tutorial.ipynb) for current master branch. 

Looks like you may not have a properly configured data folder if the object cannot be found. You should have the object assets and configs in the `data/objects/` directory like so:
[image]

Is this the case?

abhishek-tcs-rni, 2020-12-01:
Hey @aclegg3,

The data sub-folder does contain the same contents. As instructed, I had extracted to `habitat-sim/data/objects/` directory from [here](http://dl.fbaipublicfiles.com/habitat/objects_v0.2.zip). A screenshot of contents of directories given below:
`habitat-sim/data/objects/`
[image] 
`habitat-sim`
[image]

Is anything missing here ?

aclegg3, 2020-12-01:
Hm, that looks fine. I'll need to see more log details before the error to know what is happening here. Likely that some path configuration is resulting in the object configs not being loaded into the ObjectAttributesManager properly.

abhishek-tcs-rni, 2020-12-14:
@aclegg3 I have attached the output on running the command below
```
python examples/tutorials/nb_python/rigid_object_tutorial.py
```
[rigid_object_tutorial.log](https://github.com/facebookresearch/habitat-sim/files/5690450/rigid_object_tutorial.log)

Please let me know if any other logs are required

wouterreijgers, 2020-12-17:
I am experiencing the same problem, after debugging I noticed that the command on line 159:
 `obj_templates_mgr = sim.get_object_template_manager()`
 returns nothing and that  the `obj_templates_mgr` is empty. 
 I was not yet able to find a solution yet.

aclegg3, 2020-12-17:
Thanks for the logs and details. @jturner65 any thoughts on this?

jturner65, 2020-12-17:
One thing i notice : this line "I1214 22:46:37.283061 16096 ObjectAttributesManager.cpp:294] Loading file-based object template: data/test_assets/objects/sphere.phys_properties.json"

This should instead attempt to load an ".object_config.json" file.  Something may have the inappropriate extension hardcoded.

wouterreijgers, 2020-12-18:
One thing I already discovered is that the code in the [tutorial about rigid objects](https://aihabitat.org/docs/habitat-sim/rigid-object-tutorial.html#simulation-quickstart) and the code that gets installed by default are slightly different. In the code that is already in the folder `examples/tutorials/nb_python/rigid_object_tutorial.py ` the command `obj_templates_mgr.load_configs(str(os.path.join(data_path, "objects")))` is missing, i believe this command is supposed to load the objects that are in the `data/objects` folder?

Although after adding this line the problem was not solved... 

@jturner65  the sphere actually works, the problem occurs when trying to load other objects. I think the problem is located in the following lines:

```
# [dynamic_control]
observations = []
obj_templates_mgr.load_configs(str(os.path.join(data_path, "objects")))
# search for an object template by key sub-string
cheezit_template_handle = obj_templates_mgr.get_template_handles("data/objects/cheezit")[0]
```
When executing these lines the log shows the following: 
```
Traceback (most recent call last):
  File "/home/wouter/PycharmProjects/habitat_sim_official_2/habitat-sim/examples/tutorials/nb_python/rigid_object_tutorial.py", line 194, in <module>
    obj_templates_mgr.load_configs(str(os.path.join(data_path, "objects")))
AttributeError: 'habitat_sim._ext.habitat_sim_bindings.ObjectAttrib' object has no attribute 'load_configs'
```
After the debugging I found that the object `obj_templates_mgr` seems to be empty. It is created with the following command: `obj_templates_mgr = sim.get_object_template_manager()` but after looking in the `simulator.py` file I noticed that the function `get_object_template_manager` is not there although it is mentioned in [the documentation about the simulator](https://aihabitat.org/docs/habitat-sim/habitat_sim.simulator.Simulator.html#get_object_template_manager).

Also, when testing the simulator as mentioned [here](https://github.com/facebookresearch/habitat-sim#testing) (The thirth test). I noticed that although I loaded all the objects inside the correct folder I can not press the 'o', it keeps telling me that no objects are loaded.

aclegg3, 2021-04-15:
Closing as the updated issue (1014) was resolved.

closed 2021-04-15 by aclegg3