facebookresearch/habitat-sim #953: API for manipulating Locobot configuration

abhishek-tcs-rni, 2020-12-01:
## ❓ Questions and Help

For Locobot agent, at a given location and time, which API can be used to modify configuration/state of the associated manipulator e.g. changing joint angles, opening/closing end-effector gripper ?

It appears that [Locobot](http://locobot.org) is [mentioned](https://aihabitat.org/docs/habitat-sim/rigid-object-tutorial.html#embodied-agents) in tutorials, but I couldn't find any documentation, in both [`habitat-sim`](https://aihabitat.org/docs/habitat-sim/index.html) and [`habitat-lab`](https://aihabitat.org/docs/habitat-lab/index.html), for interfacing with it.

aclegg3, 2020-12-01:
Habitat-sim master branch only supports rigid body dynamics currently. If you'd like to use the articulated Locobot model and manipulate joints see the experimental [articulated-objects-prototype](https://github.com/facebookresearch/habitat-sim/tree/articulated-objects-prototype) branch.

closed 2020-12-01 by aclegg3

abhishek-tcs-rni, 2020-12-14:
@aclegg3 Thanks for pointing to the related branch !