facebookresearch/habitat-sim #955: Is there LIDAR support? how can i create my own datasets supported simulator?

Lovemyse1f, 2020-12-03:
## ❓ Questions and Help
Hi, i want to use lidar as  the experiment sensor. Besides,  i want to create a simple scene around my lab and how can i make the dataset.

erikwijmans, 2020-12-03:
There isn't a LIDAR sensor currently.  For making your scenes, we have used the Matterport Pro2 3D camera to scan lab spaces.

krishnachaitanya7, 2021-07-04:
Hi @erikwijmans! I, too, had the same question. For my project, I need a 2D LIDAR; that's when I learned about this issue.  In the first [Habitat ArXiv paper](https://arxiv.org/pdf/1904.01201.pdf) you guys published in section 3 Habitat Platform -> Design Overview, you mentioned that LIDAR could be implemented as plugins. Would you please give more information regarding that? Thank you so much!

closed 2021-08-16 by dhruvbatra

AgentEXPL, 2021-08-24:
> There isn't a LIDAR sensor currently. For making your scenes, we have used the Matterport Pro2 3D camera to scan lab spaces.

Hi @erikwijmans . I'm wondering whether to buy a Matterport Pro2 3D camera for generating high-precision and high-resolution datasets of lab scenes for habitat. Would you mind to provide more explanations on "using the Matterport Pro2 3D camera to scan lab spaces". After the Matterport Pro2 3D camera scanning a lab space, will it directly generate a .glb file which can be used by habitat? If no, which format of data is generated by the camera and how to convert it to a format (e.g., glb or other formats) used by habitat?

sparsh-umd, 2022-01-01:
Hi @erikwijmans !

Could you point me to an example showing how to implement a custom task-level sensor in habitat-lab?

Thanks!

neepacmu, 2023-02-16:
Any update on this? @sparsh-umd  @krishnachaitanya7  were you able to create a custom Lidar sensor?

keroroxzz, 2023-07-03:
Not sure if this helps but I got a ROS package providing simulated LiDAR and Laser with depth cameras.
https://github.com/keroroxzz/habitat_ros