facebookresearch/habitat-sim #956: ImportError on runing examples/example.py

jsdd25, 2020-12-03:
## ❓ Questions and Help
Hi guys!

I have installed habitat-sim using conda with the following command:

`conda install habitat-sim -c conda-forge -c aihabita`,

and then I have built Habitat-Sim within habitat conda environment with the following command:

`python setup.py install`,

and then I tried to run the `examples/example.py` within habitat conda environment with the following command:

`python examples/example.py --scene /home/jin/habitat-test-scenes/data/scene_datasets/habitat-test-scenes/skokloster-
castle.glb`.

Here is the error I got:

`Traceback (most recent call last):
  File "examples/example.py", line 10, in <module>
    import demo_runner as dr
  File "/home/jin/habitat-sim2/examples/demo_runner.py", line 15, in <module>
    from settings import default_sim_settings, make_cfg
  File "/home/jin/habitat-sim2/examples/settings.py", line 5, in <module>
    import habitat_sim
  File "/home/jin/anaconda3/envs/habitat/lib/python3.7/site-packages/habitat_sim/__init__.py", line 13, in <module>
    from habitat_sim import (  # noqa: F401
  File "/home/jin/anaconda3/envs/habitat/lib/python3.7/site-packages/habitat_sim/agent/__init__.py", line 7, in <module>
    from .agent import *  # noqa: F401,F403
  File "/home/jin/anaconda3/envs/habitat/lib/python3.7/site-packages/habitat_sim/agent/agent.py", line 14, in <module>
    from habitat_sim import bindings as hsim
  File "/home/jin/anaconda3/envs/habitat/lib/python3.7/site-packages/habitat_sim/bindings/__init__.py", line 29, in <module>
    from habitat_sim._ext.habitat_sim_bindings import (  # noqa: F401 isort:skip
ImportError: /home/jin/anaconda3/envs/habitat/lib/python3.7/site-packages/habitat_sim/_ext/habitat_sim_bindings.cpython-37m-x86_64-linux-gnu.so: undefined symbol: _ZN7Corrade7Utility6String5rtrimESs`

Could anyone help me with this? Thank you very much!

erikwijmans, 2020-12-03:
Don't run `python setup.py install` if you are installing through conda.  That will lead to version mismatches between Corrade/Magnum and habitat which is likely what happened here.

jsdd25, 2020-12-05:
Thank you very much!

closed 2020-12-05 by jsdd25

hyang23333, 2020-12-06:
Hello！I have installed habitat-sim using conda and what should I do next? I couldn't find ./build/viewer.

yunsujeon, 2021-02-02:
> Hello！I have installed habitat-sim using conda and what should I do next? I couldn't find ./build/viewer.

./build.sh