facebookresearch/habitat-sim #965: problem in install habitat-sim-0.1.5

Pixie412, 2020-12-09:
## ❓ Questions and Help
Hi,
I download the v0.1.5 source code from https://github.com/facebookresearch/habitat-sim/releases/tag/v0.1.5. I install it according to https://github.com/facebookresearch/habitat-sim/tree/v0.1.5. When I run "python setup.py install", I encounter the following problem:  
[image]
What should I do? (Since I want to install soundspace, and soundspace is dependent on habitat-sim v0.1.5, I have to install this version.)

Pixie412, 2020-12-09:
After install habitat-lab v0.1.5, I tried to install habitat-sim v0.1.5 again, here is the error:

running install
running bdist_egg
running egg_info
writing habitat_sim.egg-info/PKG-INFO
writing dependency_links to habitat_sim.egg-info/dependency_links.txt
writing requirements to habitat_sim.egg-info/requires.txt
writing top-level names to habitat_sim.egg-info/top_level.txt
reading manifest file 'habitat_sim.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'habitat_sim.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
error: [Errno 2] No such file or directory: 'build/compile_commands.json'

In the build folder:
[image]

What should I do? Any suggestion?

erikwijmans, 2020-12-09:
> RandR headers not found; install libxrandr development package

Looks like you are missing development headers, `sudo apt-get install xorg-dev libglu1-mesa-dev` should fix that.  You will either need to add `--cmake` to force a cmake or do `rm -rf build` to pruge the build folder after the install.

AgentEXPL, 2021-07-08:
Hi, @erikwijmans 
The following problem occurs when I install habitat-sim. What should I do to fix this problem?
--------------------------------
-- Found Git: /usr/bin/git (found version "2.25.1") 
CMake Warning at deps/corrade/src/Corrade/CMakeLists.txt:53 (message):
  Can't match Git version from v0.1.4-204-g2689aee1

-- Found Corrade: /home/zhuhao/OccupancyAnticipation/environments/habitat/habitat-sim/src/deps/corrade/src  found components: Containers rc Utility 
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Looking for dlfcn.h
-- Looking for dlfcn.h - found
-- Looking for execinfo.h
-- Looking for execinfo.h - found
-- Looking for glob.h
-- Looking for glob.h - found
-- Looking for inttypes.h
-- Looking for inttypes.h - found
-- Looking for libunwind.h
-- Looking for libunwind.h - not found
-- Looking for memory.h
-- Looking for memory.h - found
-- Looking for pwd.h
-- Looking for pwd.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stdlib.h
-- Looking for stdlib.h - found
-- Looking for string.h
-- Looking for string.h - found
-- Looking for strings.h
-- Looking for strings.h - found
-- Looking for sys/stat.h
-- Looking for sys/stat.h - found
-- Looking for sys/syscall.h
-- Looking for sys/syscall.h - found
-- Looking for sys/time.h
-- Looking for sys/time.h - found
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for sys/utsname.h
-- Looking for sys/utsname.h - found
-- Looking for syscall.h
-- Looking for syscall.h - found
-- Looking for syslog.h
-- Looking for syslog.h - found
-- Looking for ucontext.h
-- Looking for ucontext.h - found
-- Looking for unistd.h
-- Looking for unistd.h - found
-- Looking for unwind.h
-- Looking for unwind.h - found
-- Looking for C++ include ext/hash_map
-- Looking for C++ include ext/hash_map - found
-- Looking for C++ include ext/hash_set
-- Looking for C++ include ext/hash_set - found
-- Looking for C++ include ext/slist
-- Looking for C++ include ext/slist - found
-- Looking for C++ include tr1/unordered_map
-- Looking for C++ include tr1/unordered_map - found
[... 122 lines cut ...]
  File "/home/zhuhao/anaconda3/envs/oc_an/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/zhuhao/anaconda3/envs/oc_an/lib/python3.7/site-packages/setuptools/command/bdist_egg.py", line 164, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/home/zhuhao/anaconda3/envs/oc_an/lib/python3.7/site-packages/setuptools/command/bdist_egg.py", line 150, in call_command
    self.run_command(cmdname)
  File "/home/zhuhao/anaconda3/envs/oc_an/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/zhuhao/anaconda3/envs/oc_an/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/zhuhao/anaconda3/envs/oc_an/lib/python3.7/site-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/home/zhuhao/anaconda3/envs/oc_an/lib/python3.7/distutils/command/install_lib.py", line 107, in build
    self.run_command('build_ext')
  File "/home/zhuhao/anaconda3/envs/oc_an/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/zhuhao/anaconda3/envs/oc_an/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "setup.py", line 202, in run
    self.build_extension(ext)
  File "setup.py", line 271, in build_extension
    env=env,
  File "/home/zhuhao/anaconda3/envs/oc_an/lib/python3.7/subprocess.py", line 363, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '-H/home/zhuhao/OccupancyAnticipation/environments/habitat/habitat-sim/src', '-Bbuild', '-DBUILD_PYTHON_BINDINGS=ON', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/home/zhuhao/OccupancyAnticipation/environments/habitat/habitat-sim/build/lib.linux-x86_64-3.7/habitat_sim/_ext', '-DPYTHON_EXECUTABLE=/home/zhuhao/anaconda3/envs/oc_an/bin/python', '-DCMAKE_EXPORT_COMPILE_COMMANDS=ON', '-DCMAKE_BUILD_TYPE=RelWithDebInfo', '-DBUILD_GUI_VIEWERS=OFF', '-DTARGET_HEADLESS=ON', '-DBUILD_TEST=OFF', '-DBUILD_WITH_BULLET=OFF', '-DBUILD_DATATOOL=OFF', '-DBUILD_WITH_CUDA=ON']' returned non-zero exit status 1.

erikwijmans, 2021-07-08:
To build from source you need development headers for things, running `sudo apt install libgl1-mesa-dev` (or your distro's equivalent) will get the OpenGL ones.

AgentEXPL, 2021-07-09:
> To build from source you need development headers for things, running `sudo apt install libgl1-mesa-dev` (or your distro's equivalent) will get the OpenGL ones.

Thanks a lot! After installing libgl1-mesa-dev, the install (python setup.py install --headless --with-cuda) works as follows. 
--------------------------------------------
Using /home/zhuhao/anaconda3/envs/oc_an/lib/python3.7/site-packages
Finished processing dependencies for habitat-sim==0.1.5
Processing ./build/deps/magnum-bindings/src/python
  DEPRECATION: A future pip version will change local packages to be built in-place without first copying to a temporary directory. We recommend you use --use-feature=in-tree-build to test your packages with this new behavior before it becomes the default.
   pip 21.3 will remove support for this functionality. You can find discussion regarding this at https://github.com/pypa/pip/issues/7555.
Building wheels for collected packages: magnum
  Building wheel for magnum (setup.py) ... done
  Created wheel for magnum: filename=magnum-0.0.0-cp37-cp37m-linux_x86_64.whl size=2123022 sha256=410982c17fa6abbd9e0718fe548d9f5df134b78040f7435adcca9c71c538b112
  Stored in directory: /tmp/pip-ephem-wheel-cache-4bxniui7/wheels/08/7e/b8/f366cc92f55daa836fe0e74a149f3989d08a51f096678023c9
Successfully built magnum
Installing collected packages: magnum
Successfully installed magnum-0.0.0
--------------------------------------------

However, import error occurs as follows. Maybe the reason is that I am trying to install habitat-sim==0.1.5 in a conda environment, as discussed in [#956 ](https://github.com/facebookresearch/habitat-sim/issues/956). What should I do to solve this issue?

erikwijmans, 2021-07-09:
Seems like this is the same as in https://github.com/facebookresearch/habitat-sim/issues/971

AgentEXPL, 2021-07-12:
Since I install the habitat-sim in a server as a customer, I do not have the right to re-install CUDA which needs to uninstall the old version. Previously, I thought the problem is caused by installing habitat-sim in a conda env with unmatched command "python setup.py install --headless --with-cuda". Then, I tried to install habitat-sim in a conda env in the server with "conda install habitat-sim headless -c conda-forge -c aihabitat", but the same issue was occurred. Now I have no idea on what to do.
The problem is that when I tried to import habitat_sim in a python file after the install, the following error was shown.

from _magnum import *
ImportError: /lib/x86_64-linux-gnu/libOpenGL.so.0: undefined symbol: _glapi_tls_Current

erikwijmans, 2021-07-12:
You'll need to talk with the admins for that server.  This isn't something that can be fixed at the habitat-sim level as it's caused a broken OpenGL install

AgentEXPL, 2021-07-13:
Thanks for your suggestion and patient guidance.

closed 2022-01-26 by dhruvbatra