facebookresearch/habitat-sim #970: Error Building from source

elientumba2019, 2020-12-12:
Hello. I am trying to build from source and get the following error. (See log file)

**Steps to reproduce** 
conda create -n habitat python=3.6 cmake=3.14.0
conda activate habitat
git clone https://github.com/facebookresearch/habitat-sim.git
cd habitat-sim
pip install -r requirements.txt
python setup.py install --with-cuda --cmake

**System**
ubuntu1~18.04
gcc 7.5.0

[log.txt](https://github.com/facebookresearch/habitat-sim/files/5682541/log.txt)

elientumba2019, 2020-12-12:
what could be the cause and how can I fix it? Thanks

erikwijmans, 2020-12-12:
```
fatal: not a git repository (or any parent up to mount point /media/lf216)
```

It seems like you downloaded a zip of the repo instead of cloning.  Please clone the repo instead.

closed 2020-12-12 by erikwijmans