facebookresearch/habitat-sim #971: ImportError: /lib/x86_64-linux-gnu/libOpenGL.so.0: undefined symbol: _glapi_tls_Current

KenaHemnani, 2020-12-12:
## ❓ Questions and Help
Hello all;
I was getting following error while running : **python setup.py install --headless**

-- Could NOT find EGL (missing: EGL_INCLUDE_DIR)
CMake Error at deps/magnum/src/Magnum/Platform/CMakeLists.txt:796 (message):
EGL library, required by some contexts, was not found. Set
WITH_*APPLICATION and/or WITH_EGLCONTEXT to OFF to skip building them.
-- Configuring incomplete, errors occurred!

Then I tried solution given in: https://github.com/facebookresearch/habitat-sim/issues/256

Now, I got following error while running **python examples/example.py**:

(habitat1) hemnani1@boss8:~/DIR1/habitat-lab$ python examples/example.py
2020-12-12 18:49:48,396 Initializing dataset PointNav-v1
Traceback (most recent call last):
  File "examples/example.py", line 28, in <module>
    example()
  File "examples/example.py", line 14, in example
    config=habitat.get_config("configs/tasks/pointnav.yaml")
  File "/home/hemnani1/DIR1/habitat-lab/habitat/core/env.py", line 78, in __init_                                                          _
    id_dataset=config.DATASET.TYPE, config=config.DATASET
  File "/home/hemnani1/DIR1/habitat-lab/habitat/datasets/registration.py", line 2                                                          0, in make_dataset
    return _dataset(**kwargs)
  File "/home/hemnani1/DIR1/habitat-lab/habitat/datasets/pointnav/__init__.py", l                                                          ine 30, in __init__
    raise pointnav_import_error
  File "/home/hemnani1/DIR1/habitat-lab/habitat/datasets/pointnav/__init__.py", l                                                          ine 20, in _try_register_pointnavdatasetv1
    from habitat.datasets.pointnav.pointnav_dataset import (  # noqa: F401
  File "/home/hemnani1/DIR1/habitat-lab/habitat/datasets/pointnav/pointnav_datase                                                          t.py", line 15, in <module>
    from habitat.tasks.nav.nav import (
  File "/home/hemnani1/DIR1/habitat-lab/habitat/tasks/nav/nav.py", line 31, in <m                                                          odule>
    from habitat.sims.habitat_simulator.actions import HabitatSimActions
  File "/home/hemnani1/DIR1/habitat-lab/habitat/sims/habitat_simulator/actions.py                                                          ", line 12, in <module>
    import habitat_sim
  File "/home/hemnani1/anaconda3/envs/habitat1/lib/python3.6/site-packages/habita                                                          t_sim-0.1.6-py3.6-linux-x86_64.egg/habitat_sim/__init__.py", line 13, in <module>
    from habitat_sim import (  # noqa: F401
  File "/home/hemnani1/anaconda3/envs/habitat1/lib/python3.6/site-packages/habita                                                          t_sim-0.1.6-py3.6-linux-x86_64.egg/habitat_sim/agent/__init__.py", line 7, in <mo                                                          dule>
    from .agent import *  # noqa: F401,F403
  File "/home/hemnani1/anaconda3/envs/habitat1/lib/python3.6/site-packages/habita                                                          t_sim-0.1.6-py3.6-linux-x86_64.egg/habitat_sim/agent/agent.py", line 10, in <modu                                                          le>
    import magnum as mn
  File "/home/hemnani1/anaconda3/envs/habitat1/lib/python3.6/site-packages/magnum                                                          /__init__.py", line 28, in <module>
    from _magnum import *
ImportError: /lib/x86_64-linux-gnu/libOpenGL.so.0: undefined symbol: _glapi_tls_C                                                          urrent

Can anyone please help me resolve this error.

rpartsey, 2021-04-13:
Hi @KenaHemnani 
I had the same problem. In my case, I was configuring the server from scratch (to run the _habitat_).

**STEPS TO REPRODUCE:**
I installed the _nvidia-driver_ and _cuda_ first.
Then I installed the _habitat_ dependencies listed [here](https://github.com/facebookresearch/habitat-sim/blob/fc7baba4f208a7589a7af752ab686e39cccf28c4/BUILD_FROM_SOURCE.md) by running:

```
sudo apt-get update || true
# These are fairly ubiquitous packages and your system likely has them already,
# but if not, let's get the essentials for EGL support:
sudo apt-get install -y --no-install-recommends \
     libjpeg-dev libglm-dev libgl1-mesa-glx libegl1-mesa-dev mesa-utils xorg-dev freeglut3-dev
```
And got the same error message as you attached above.

The **root cause** is: 

> Note: Installing Mesa may overwrite the /usr/lib/libGL.so that was previously installed by the NVIDIA driver, so a reinstallation of the NVIDIA driver might be required after installing these libraries. 

Source: [cuda installation guide](https://docs.nvidia.com/cuda/cuda-installation-guide-linux/index.html).

**SOLUTION:** reinstallation of the NVIDIA driver solved the issue for me.

fyhMer, 2022-04-23:
Hi there,
I came across the same problem, and I found a more convenient solution by reinstalling `libglvnd`.
I solved the problem by running (on CentOS7): 
```
sudo yum reinstall libglvnd
```
For your reference: https://github.com/openai/mujoco-py/issues/513

aclegg3, 2022-08-31:
Looks like this is resolved.

closed 2022-08-31 by aclegg3