facebookresearch/habitat-sim #974: Examples for Locobot

abhishek-tcs-rni, 2020-12-14:
## ❓ Questions and Help

In [articulated objects prototype](https://github.com/facebookresearch/habitat-sim/tree/articulated-objects-prototype), the given [tutorial](https://github.com/facebookresearch/habitat-sim/blob/articulated-objects-prototype/examples/tutorials/URDF_robotics_tutorial.py) provides examples related to URDF options `iiwa` and `aliengo` given [here](https://github.com/facebookresearch/habitat-sim/blob/81d2730832f4c0a65354edf7f2cfebdc876b7f14/examples/tutorials/URDF_robotics_tutorial.py#L112). However, for `locobot` and `locobot_light`, currently it appears there are no examples.

Can examples related to using `locobot` be found somewhere ? I am trying to understand ways to control the state of base and manipulator of Locobot, and would much appreciate being provided related examples

aclegg3, 2020-12-15:
At this time we don't have any stable examples prepared for the locobot URDF. The API will be the same as the other robots. 

There is some [`LocobotController`](https://github.com/facebookresearch/habitat-sim/blob/81d2730832f4c0a65354edf7f2cfebdc876b7f14/src/utils/viewer/viewer.cpp#L101) code in the Viewer application on that branch which you could play with or migrate to python as a starting point. This just actuates the wheels of a simplified locobot model as shown below.

[image]

abhishek-tcs-rni, 2020-12-19:
@aclegg3 Thanks for clarifying that and pointing to [`LocobotController`](https://github.com/facebookresearch/habitat-sim/blob/81d2730832f4c0a65354edf7f2cfebdc876b7f14/src/utils/viewer/viewer.cpp#L101), along with gif. Actually, more than movement, I was looking for ways to use its manipulator.

In order to attempt migration to python, if I am able to, could you please point to some migration examples here, or corresponding learning resources.

aclegg3, 2021-01-12:
Hey @abhishek-tcs-rni,

We don't have any examples for using the Locobot manipulator. However, if you are interested in some exploration I can suggest an approach for finding the answers. You can still use the position control motors as the python tutorial shows for the AlienGo and Iiwa. You'll just need to discover the correct DOF indices for the manipulator joints you want to control. I suggest using the "DOF" mouse mode in the viewer application or writing a script in python to manually manipulate each DoF and discover which joint it corresponds to.

closed 2021-05-26 by dhruvbatra

hpf9017, 2021-11-26:
@aclegg3 hi, I checkout to the branch, and add a Locobot with OPEN/CLOSE mode. I hit the robot and press the left button with mouse, but the locobot didn't move forward or backward. what could be the reason for the problem? Thank you.