facebookresearch/habitat-sim #976: phys_properties.json not provided

gianscarpe, 2020-12-15:
## ❓ Questions and Help
Hi and thanks for the great effort you are putting into this project! I wanted to ask a question related to `phys_properties.json`. I notice that these files are necessary for calling `get_template_handles` API. If the file is not provided, the get_template_handles cannot load the related object.
```
chefcan_template_handle = obj_templates_mgr.get_template_handles(                                                                                                                                      
        "data/objects/chefcan"                                                                                                                                                                             
    )[0]  
```
However, many of the objects and agents aren't shipped with these files. I overcome the issue by creating a simple json as:
```
 {                                                                                                                                                                                                         
     "render mesh": "locobot_merged.glb",                                                                                                                                                                  
     "join collision meshes": true                                                                                                                                                                         
 }                                                                                                                                                                                                         
```
I was wondering if this is the best solution, or if there are other ways to define physic properties for objects.
Thanks!!

aclegg3, 2020-12-15:
It is true that you can create these manually on your file-system for new objects as you've done here. 

You can also create the [`ObjectAttributes`](https://aihabitat.org/docs/habitat-sim/habitat_sim.attributes.ObjectAttributes.html) programmatically for your asset with [`ObjectAttributeManager.create_new_template`](https://aihabitat.org/docs/habitat-sim/habitat_sim.attributes_managers.ObjectAttributesManager.html#create_new_template) and then fill out the fields as you see fit. Once you are done you can [register the template](https://aihabitat.org/docs/habitat-sim/habitat_sim.attributes_managers.ObjectAttributesManager.html#register_template) with any handle you'd like. 

For more info on this process and system, see the Habitat-Sim Advanced Topics segment of our [ECCV 2020 tutorial series](https://aihabitat.org/tutorial/2020/). There is a video and [accompanying interactive Colab](https://colab.research.google.com/github/facebookresearch/habitat-sim/blob/master/examples/tutorials/colabs/ECCV_2020_Advanced_Features.ipynb) with an example/sandbox.

aclegg3, 2020-12-15:
Also note that the `phys_properties.json` file type has been replaced by `object_config.json` as of [this commit](https://github.com/facebookresearch/habitat-sim/commit/c4c6051644a36491748881a26d844bc2151da0a0).

closed 2020-12-27 by gianscarpe