facebookresearch/habitat-sim #977: nvcc fatal   : Unknown option 'flto' while installing habitat-sim @ 2689aee

ruddradev, 2020-12-16:
## 🐛 Bug
When I try to set up the habitat sim commit 2689aee which is required for the Occupancy Anticipation research work, I find that installing with CUDA give the error - 

nvcc fatal   : Unknown option 'flto' at the 681st executable while building. This does not occur when building without --with-cuda flag.

## Command
python setup.py install --headless --with-cuda

## To Reproduce

Steps to reproduce the behavior:

python setup.py install --headless --with-cuda

## Expected behavior

## Additional context

## System Info

Please copy and paste the output from the
[environment collection script](https://github.com/facebookresearch/habitat-sim/tree/master/habitat_sim/utils/collect_env.py)
(or fill out the checklist below manually).

You can run the script with:
```
# For security purposes, please check the contents of collect_env.py before running it.
python habitat_sim/utils/collect_env.py
```

 - OS (e.g., Linux): Ubuntu 18.04
 - How you installed PyTorch (`conda`, `pip`, source): conda
 - Build command you used (if compiling from source): python setup.py install --headless --with-cuda
 - Python version: 3.7
 - GPU models and configuration: Nvidia P5000
 - CUDA version: 10.1
 - PIP/conda dependencies packages versions
 - Any other relevant information:

Skylion007, 2020-12-16:
What's your CMake version? This is usually caused by CMake being too old.

ruddradev, 2020-12-22:
Hello @Skylion007, 

nvcc --version gives output 

nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2018 NVIDIA Corporation
Built on Sat_Aug_25_21:08:01_CDT_2018
Cuda compilation tools, release 10.0, V10.0.130

cmake --version gives output as

cmake version 3.12.0

Could you tell me if I need to upgrade any of these?

Skylion007, 2021-01-05:
If I recall, the cmake version may ned to be slightly newer. What's the gcc/clang version?

ruddradev, 2021-01-06:
Hi @Skylion007 : Thanks for the response. I tried the below command - 

g++ --version
I gives - 
g++ (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0

Skylion007, 2021-01-06:
I am pretty sure this error was resolved previously by updating the cmake to 3.14 or higher. See if that solves the issue.

ruddradev, 2021-01-19:
Thank you. Upgrading Cmake to 3.19 solved the issue.

closed 2021-01-19 by ruddradev