facebookresearch/habitat-sim #978: [WSL2 Ubuntu18.04 Cuda11.1] unable to fine EGL device for CUDA device 0

moyutianque, 2020-12-16:
## ❓ Questions and Help
May I ask a simple question, I have tried many times but the following error still exists, for both the normal version and headless version with conda install. In wsl2, I can use my Nvidia 1060 GPU normally for pytorch.  The test sample in cuda-toolkits can also pass.  

(base) root@DESKTOP:~/ExternalLib/habitat-sim-stable# python examples/example.py
sim_cfg.physics_config_file = ./data/default.phys_scene_config.json
==== Initialized Sensor Spec: =====
Sensor uuid:  color_sensor
Sensor type:  SensorType.COLOR
Sensor position:  [0.  1.5 0. ]
===================================
Test scenes not downloaded locally, downloading and extracting now...
Downloaded and extracted test scenes data.
WARNING: Logging before InitGoogleLogging() is written to STDERR
I1216 14:17:32.518131 19962 AssetAttributesManager.cpp:122] Asset attributes (capsule3DSolid : capsule3DSolid_hemiRings_4_cylRings_1_segments_12_halfLen_0.75_useTexCoords_false_useTangents_false) created and registered.
I1216 14:17:32.518218 19962 AssetAttributesManager.cpp:122] Asset attributes (capsule3DWireframe : capsule3DWireframe_hemiRings_8_cylRings_1_segments_16_halfLen_1) created and registered.
I1216 14:17:32.518257 19962 AssetAttributesManager.cpp:122] Asset attributes (coneSolid : coneSolid_segments_12_halfLen_1.25_rings_1_useTexCoords_false_useTangents_false_capEnd_true) created and registered.
I1216 14:17:32.518306 19962 AssetAttributesManager.cpp:122] Asset attributes (coneWireframe : coneWireframe_segments_32_halfLen_1.25) created and registered.
I1216 14:17:32.518321 19962 AssetAttributesManager.cpp:122] Asset attributes (cubeSolid : cubeSolid) created and registered.
I1216 14:17:32.518332 19962 AssetAttributesManager.cpp:122] Asset attributes (cubeWireframe : cubeWireframe) created and registered.
I1216 14:17:32.518368 19962 AssetAttributesManager.cpp:122] Asset attributes (cylinderSolid : cylinderSolid_rings_1_segments_12_halfLen_1_useTexCoords_false_useTangents_false_capEnds_true) created and registered.
I1216 14:17:32.518424 19962 AssetAttributesManager.cpp:122] Asset attributes (cylinderWireframe : cylinderWireframe_rings_1_segments_32_halfLen_1) created and registered.
I1216 14:17:32.518462 19962 AssetAttributesManager.cpp:122] Asset attributes (icosphereSolid : icosphereSolid_subdivs_1) created and registered.
I1216 14:17:32.518476 19962 AssetAttributesManager.cpp:122] Asset attributes (icosphereWireframe : icosphereWireframe_subdivs_1) created and registered.
I1216 14:17:32.518496 19962 AssetAttributesManager.cpp:122] Asset attributes (uvSphereSolid : uvSphereSolid_rings_8_segments_16_useTexCoords_false_useTangents_false) created and registered.
I1216 14:17:32.518565 19962 AssetAttributesManager.cpp:122] Asset attributes (uvSphereWireframe : uvSphereWireframe_rings_16_segments_32) created and registered.
I1216 14:17:32.518600 19962 AssetAttributesManager.cpp:108] AssetAttributesManager::buildCtorFuncPtrMaps : Built default primitive asset templates : 12
W1216 14:17:32.519341 19962 ObjectAttributesManager.cpp:326] Cannot find ./data/objects or ./data/objects.phys_properties.json. Aborting parse.
I1216 14:17:32.519376 19962 PhysicsAttributesManager.cpp:39] File (./data/default.phys_scene_config.json) Based physics manager attributes created and registered.
I1216 14:17:32.519459 19962 StageAttributesManager.cpp:79] File (data/scene_datasets/habitat-test-scenes/skokloster-castle.glb) Based stage attributes created and registered.
I1216 14:17:32.519491 19962 Simulator.cpp:145] Loading navmesh from data/scene_datasets/habitat-test-scenes/skokloster-castle.navmesh
I1216 14:17:32.519577 19962 Simulator.cpp:147] Loaded.
I1216 14:17:32.519611 19962 SceneGraph.h:93] Created DrawableGroup:
Platform::WindowlessEglApplication::tryCreateContext(): unable to find EGL device for CUDA device 0
WindowlessContext: Unable to create windowless context

erikwijmans, 2020-12-16:
Please do `export MAGNUM_LOG=verbose MAGNUM_GPU_VALIDATION=ON` and run again for additional logging.

moyutianque, 2020-12-16:
> export MAGNUM_LOG=verbose

Thanks for your quick reply. I follow your suggestion. Here is the message. In addition, I have seen someone ask a similar question, and he solved it by installing cuda driver. I am not very sure whether the cuda installed in WSL2 following this [link](https://docs.nvidia.com/cuda/wsl-user-guide/index.html) has additional problems. I can run the examples in cuda-toolkit, and pytorch can load the correct GPU.

```
(base) root@DESKTOP-TJIUGS8:~/ExternalLib/habitat-sim-stable# python examples/example.py
sim_cfg.physics_config_file = ./data/default.phys_scene_config.json
==== Initialized Sensor Spec: =====
Sensor uuid:  color_sensor
Sensor type:  SensorType.COLOR
Sensor position:  [0.  1.5 0. ]
===================================
WARNING: Logging before InitGoogleLogging() is written to STDERR
I1216 19:30:58.756547    65 AssetAttributesManager.cpp:122] Asset attributes (capsule3DSolid : capsule3DSolid_hemiRings_4_cylRings_1_segments_12_halfLen_0.75_useTexCoords_false_useTangents_false) created and registered.
I1216 19:30:58.756641    65 AssetAttributesManager.cpp:122] Asset attributes (capsule3DWireframe : capsule3DWireframe_hemiRings_8_cylRings_1_segments_16_halfLen_1) created and registered.
I1216 19:30:58.756683    65 AssetAttributesManager.cpp:122] Asset attributes (coneSolid : coneSolid_segments_12_halfLen_1.25_rings_1_useTexCoords_false_useTangents_false_capEnd_true) created and registered.
I1216 19:30:58.756739    65 AssetAttributesManager.cpp:122] Asset attributes (coneWireframe : coneWireframe_segments_32_halfLen_1.25) created and registered.
I1216 19:30:58.756805    65 AssetAttributesManager.cpp:122] Asset attributes (cubeSolid : cubeSolid) created and registered.
I1216 19:30:58.756873    65 AssetAttributesManager.cpp:122] Asset attributes (cubeWireframe : cubeWireframe) created and registered.
I1216 19:30:58.757001    65 AssetAttributesManager.cpp:122] Asset attributes (cylinderSolid : cylinderSolid_rings_1_segments_12_halfLen_1_useTexCoords_false_useTangents_false_capEnds_true) created and registered.
I1216 19:30:58.757055    65 AssetAttributesManager.cpp:122] Asset attributes (cylinderWireframe : cylinderWireframe_rings_1_segments_32_halfLen_1) created and registered.
I1216 19:30:58.757072    65 AssetAttributesManager.cpp:122] Asset attributes (icosphereSolid : icosphereSolid_subdivs_1) created and registered.
I1216 19:30:58.757086    65 AssetAttributesManager.cpp:122] Asset attributes (icosphereWireframe : icosphereWireframe_subdivs_1) created and registered.
I1216 19:30:58.757115    65 AssetAttributesManager.cpp:122] Asset attributes (uvSphereSolid : uvSphereSolid_rings_8_segments_16_useTexCoords_false_useTangents_false) created and registered.
I1216 19:30:58.757198    65 AssetAttributesManager.cpp:122] Asset attributes (uvSphereWireframe : uvSphereWireframe_rings_16_segments_32) created and registered.
I1216 19:30:58.757251    65 AssetAttributesManager.cpp:108] AssetAttributesManager::buildCtorFuncPtrMaps : Built default primitive asset templates : 12
W1216 19:30:58.767678    65 ObjectAttributesManager.cpp:326] Cannot find ./data/objects or ./data/objects.phys_properties.json. Aborting parse.
I1216 19:30:58.767729    65 PhysicsAttributesManager.cpp:39] File (./data/default.phys_scene_config.json) Based physics manager attributes created and registered.
I1216 19:30:58.767854    65 StageAttributesManager.cpp:79] File (data/scene_datasets/habitat-test-scenes/skokloster-castle.glb) Based stage attributes created and registered.
I1216 19:30:58.779930    65 Simulator.cpp:145] Loading navmesh from data/scene_datasets/habitat-test-scenes/skokloster-castle.navmesh
I1216 19:30:58.844296    65 Simulator.cpp:147] Loaded.
I1216 19:30:58.861404    65 SceneGraph.h:93] Created DrawableGroup:
Platform::WindowlessEglApplication: eglQueryDeviceStringEXT(EGLDevice=0): EGL_MESA_device_software
eglQueryDeviceAttribEXT(): eglQueryDeviceStringEXT
Platform::WindowlessEglApplication::tryCreateContext(): unable to find EGL device for CUDA device 0
WindowlessContext: Unable to create windowless context
```

erikwijmans, 2020-12-17:
You are likely missing Nvidia's OpenGL libs, please take a look at the issue here: https://github.com/facebookresearch/habitat-sim/issues/288

moyutianque, 2020-12-17:
> You are likely missing Nvidia's OpenGL libs, please take a look at the issue here: #288

There is no libEGL_nvidia.so.0 if follow the standard way to install cuda-toolkit-11-1 on WSL2. 

The official document from Nvidia says:

> Do not choose the cuda, cuda-11-0, or cuda-drivers meta-packages under WSL 2 since these packages will result in an attempt to install the Linux NVIDIA driver under WSL 2.
>$ apt-get install -y cuda-toolkit-11-0

I guess that is why this feature is not supported on WSL2. 

Anyway, thanks for your suggestion.

erikwijmans, 2020-12-17:
If this is a machine with an attached display, the non-headless build may work.  That piggybacks off the display server instead of going through Nvidia's custom EGL libs.  I am not sure if WSL2 exposes the display server properly or not, but it is worth a shot.

vincent341, 2021-01-08:
Hi @erikwijmans . I run into the error "Platform::WindowlessEglApplication::tryCreateContext(): unable to find EGL device for CUDA device 0" too when I was working on habitat challenge. I didn't find "libEGL_nvidia.so.0 " in the computer. Would you mind letting me know any hints on its installation or which apt-get install package contains "libEGL_nvidia.so.0 "? Thanks very much.

erikwijmans, 2021-01-08:
Those are in the normal cuda-drivers package

vincent341, 2021-01-09:
> Those are in the normal cuda-drivers package

Hi @erikwijmans ,
Thanks for your quick reply. Does it mean that libEGL_nvidia.so.0 should be in the computer if the cuda driver is installed? The "nvidia-smi" reported that my CUDA version is 11.0 and Driver Version is 450.51.05. And "torch.cuda.is_available()" and "torch.cuda.device_count()" returns True and 4 respectively. It seems that my cuda works. But I didn't find libEGL_nvidia.so.0 in my computer. Would you mind giving me some hints to solve it? I truly appreciate your help.

erikwijmans, 2021-01-09:
It isn't always called `libEGL_nvidia.so.0`, look for `libEGL_nvidia.so*`

This has also been discussed extensively in https://github.com/facebookresearch/habitat-sim/issues/288

vincent341, 2021-01-11:
@erikwijmans Thanks very much for your help.

vincent341, 2021-01-13:
Hi @erikwijmans ,

I'm still struggling with this problem. Previously, an "unable to fine EGL device for CUDA device 0" occured. At that time, I checked the computer. libEGL_nvidia.so* cannot be found in the computer. Then I installed "libnvidia-gl" package.  Consequently, libEGL_nvidia.so can be found in 

> /usr/lib/x86_64-linux-gnu/libEGL_nvidia.so.0
> /usr/lib/x86_64-linux-gnu/libEGL_nvidia.so.450.102.04
> /usr/lib/i386-linux-gnu/libEGL_nvidia.so.0
> /usr/lib/i386-linux-gnu/libEGL_nvidia.so.450.102.04

"ldconfig -p | grep libEGL" returns 

> libEGL_nvidia.so.0 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libEGL_nvidia.so.0
> libEGL_nvidia.so.0 (libc6) => /usr/lib/i386-linux-gnu/libEGL_nvidia.so.0
> libEGL_mesa.so.0 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libEGL_mesa.so.0
> libEGL_mesa.so.0 (libc6) => /usr/lib/i386-linux-gnu/libEGL_mesa.so.0
> libEGL.so.1 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libEGL.so.1
> libEGL.so.1 (libc6) => /usr/lib/i386-linux-gnu/libEGL.so.1
> libEGL.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libEGL.so
> libEGL.so (libc6) => /usr/lib/i386-linux-gnu/libEGL.so

I also export LD_LIBRARY=/usr/lib/x86_64-linux-gnu:$LD_LIBRARY. Then run the program again. The "unable to fine EGL device for CUDA device 0" error disappeared but got another error "Platform::WindowlessEglApplication::tryCreateContext(): no EGL devices found, likely a driver issue;". "nvidia-smi" works as expected.

Any hints would be highly appreciated!

erikwijmans, 2021-01-13:
`export MAGNUM_LOG=verbose MAGNUM_GPU_VALIDATION=ON` will provide additional logging

ZeroExistence, 2021-03-10:
@vincent341 you might need to align your NVIDIA driver version with your nvidia-gl library.
```
$ nvidia-smi
Wed Mar 10 10:17:14 2021
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 418.87.00    Driver Version: 418.87.00    CUDA Version: 11.0     |
|-------------------------------+----------------------+----------------------+

$ dpkg -l | grep -i nvidia
iU  libnvidia-common-418                   418.87.00-0ubuntu1                  all          Shared files used by the NVIDIA libraries
iU  libnvidia-gl-418:amd64                 418.87.00-0ubuntu1                  amd64        NVIDIA OpenGL/GLX/EGL/GLES GLVND libraries and Vulkan ICD
```

The output above might not be the same with you, because I'm dealing with a container.

I had the same issue when I used the libnvidia-gl package provided by Ubuntu repository. I strongly advice to use the package from NVIDIA repository, and specify the exact version.

For my case, I use this command to fix my issue.
`sudo apt-get install libnvidia-gl-418=418.87.00-0ubuntu1 libnvidia-compute-418=418.87.00-0ubuntu1 libnvidia-common-418=418.87.00-0ubuntu1`

Use this command to check the available packages on your repository.
`apt-cache search libnvidia-gl; apt-cache madison libnvidia-gl-418`

I hope this helps.

alpc72, 2022-01-08:
@vincent341 Hi, I meet the same problem, have you solved this problem?

vincent341, 2022-01-08:
> @vincent341 Hi, I meet the same issue, have you solved this problem?

Hi, it has been some time. I'm not sure if I remember correctly. I solved it by using docker. The image/container contains the environments for configuring habitat.

closed 2022-08-31 by aclegg3

wbhwbh, 2023-11-28:
> @vincent341 you might need to align your NVIDIA driver version with your nvidia-gl library.
>
> ```
> $ nvidia-smi
> Wed Mar 10 10:17:14 2021 [...]

Hi, I meet the same question as you and really want to try your solutions. But I really get some problems. I know the time has passed for a long time and I will be appreciate if you can give me any details about this issue.

My ubuntu version is 20.04.6, nvidia driver version is 535.113.01. I search matched version libnvidia-gl from website https://pkgs.org/download/libnvidia-gl-535, but only find 535.113.01 version for ubuntu23.10. On this website, I can't even find the 418.87.00 version that you provided. I guess that if the website has missed some old version? 

Can you tell me where to find the libnvidia-gl download address for ubuntu version<20 and how to find the matched libnvidia-gl with nvidia driver version (such as 525.105.17 on ubuntu 16.04.5)?

ZeroExistence, 2023-11-28:
> Hi, I meet the same question as you and really want to try your solutions. But I really get some problems. I know the time has passed for a long time and I will be appreciate if you can give me any details about this issue.
> 
> My ubuntu version is 20.04.6, nvidia driver version is 535.113.01. I search matched version libnvidia-gl from website https://pkgs.org/download/libnvidia-gl-535, but only find 535.113.01 version for ubuntu23.10. On this website, I can't even find the 418.87.00 version that you provided (I can install it from the commands you offered on ubuntu 18.04.6 but can't install on ubuntu 20.04.6). I guess that if the website has missed some old version?
> 
> Can you tell me where to find the libnvidia-gl download address for ubuntu version<20 and how to find the matched libnvidia-gl with nvidia driver version (such as 525.105.17 on ubuntu 16.04.5)?

If you don't have the NVIDIA Developer repo added into your Ubuntu yet, you can run these commands.
```
apt-get update && apt-get install -y --no-install-recommends \
    gnupg2 curl ca-certificates && \
    curl -fsSL https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64/3bf863cc.pub | apt-key add - && \
    echo "deb https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64 /" > /etc/apt/sources.list.d/cuda.list && \
    apt-get update && \
    apt-cache madison libnvidia-gl
```

The related `libnvidia-gl` and package dependencies should be here.

But I checked your NVIDIA driver version, and it looks like the `535.113.xx` specific version was not included in their repositories for Ubuntu20 and Ubuntu22 stable branches.

If you have the ability to upgrade your NVIDIA driver and CUDA driver, it might be a good idea to upgrade to `535.129.03` so that there will be matching `libnvidia-gl` version.

If you want more references on managing NVIDIA driver packages, you can use the NVIDIA cuda container dockerfile.
[https://gitlab.com/nvidia/container-images/cuda/-/blob/master/dist/12.0.1/ubuntu2004/base/Dockerfile?ref_type=heads](https://gitlab.com/nvidia/container-images/cuda/-/blob/master/dist/12.0.1/ubuntu2004/base/Dockerfile?ref_type=heads)

wbhwbh, 2023-11-28:
> If you don't have the NVIDIA Developer repo added into your Ubuntu yet, you can run these commands.
>
> ```
> apt-get update && apt-get install -y --no-install-recommends \
>     gnupg2 curl [...]

Thank you for your so fast reply!

Unfortunately, I have limits of authority to upgrade NVIDIA driver and CUDA driver. This also prevents me from using the methods suggested by most people (reinstall the nvidia driver) to solve this problem. Luckily, I have solved this BUG following #1511 by creating some simlinks.

Thank you very much again for your active reply. The [website](https://developer.download.nvidia.com/compute/cuda/repos/) you offered is also useful to find desired packages.

shipshipliucixin1, 2025-04-18:
> > [@vincent341](https://github.com/vincent341) Hi, I meet the same issue, have you solved this problem?
> 
> Hi, it has been some time. I'm not sure if I remember correctly. I solved it by using docker. The image/container contains the environments for configuring habitat.

Could you please share the base image of the docker?Thanks a lot