facebookresearch/habitat-sim #979: PDB breakpoint casuses segmentation fault

Jbwasse2, 2020-12-16:
## 🐛 Bug
When I put a breakpoint in the code, a segmentation fault occurs. This occurs when I use PuDB or pdb.

## Command

## To Reproduce

Steps to reproduce the behavior:

1. Install habitat-sim and habitat-lab as directed by README.md
2. Run following code

### Code
```
import faulthandler
import pdb

import habitat

faulthandler.enable()

# Load embodied AI task (PointNav) and a pre-specified virtual robot
env = habitat.Env(config=habitat.get_config("configs/tasks/pointnav.yaml"))

observations = env.reset()

# Step through environment with random actions
while not env.episode_over:
    pdb.set_trace()
    observations = env.step(env.action_space.sample())
```
### Output
```
Renderer: llvmpipe (LLVM 10.0.0, 128 bits) by VMware, Inc.
OpenGL version: 3.3 (Core Profile) Mesa 20.0.8
Using optional features:
    GL_ARB_ES2_compatibility
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_invalidate_subdata
    GL_ARB_multi_bind
    GL_ARB_robustness
    GL_ARB_separate_shader_objects
    GL_ARB_texture_storage
    GL_ARB_texture_storage_multisample
    GL_ARB_vertex_array_object
    GL_KHR_debug
Using driver workarounds:
    no-layout-qualifiers-on-old-glsl
    mesa-dsa-createquery-except-pipeline-stats
    mesa-forward-compatible-line-width-range
479 AssetAttributesManager.cpp:122] Asset attributes (coneSolid : coneSolid_segments_12_halfLen_1.25_rings_1_useTexCoords_false_useTangents_false_capEnd_true) created and registered.
I1216 13:13:14.824842 46479 AssetAttributesManager.cpp:122] Asset attributes (coneWireframe : coneWireframe_segments_32_halfLen_1.25) created and registered.
I1216 13:13:14.824856 46479 AssetAttributesManager.cpp:122] Asset attributes (cubeSolid : cubeSolid) created and registered.
I1216 13:13:14.824867 46479 AssetAttributesManager.cpp:122] Asset attributes (cubeWireframe : cubeWireframe) created and registered.
I1216 13:13:14.824904 46479 AssetAttributesManager.cpp:122] Asset attributes (cylinderSolid : cylinderSolid_rings_1_segments_12_halfLen_1_useTexCoords_false_useTangents_false_capEnds_true) created and registered.
I1216 13:13:14.824936 46479 AssetAttributesManager.cpp:122] Asset attributes (cylinderWireframe : cylinderWireframe_rings_1_segments_32_halfLen_1) created and registered.
I1216 13:13:14.824951 46479 AssetAttributesManager.cpp:122] Asset attributes (icosphereSolid : icosphereSolid_subdivs_1) created and registered.
I1216 13:13:14.824965 46479 AssetAttributesManager.cpp:122] Asset attributes (icosphereWireframe : icosphereWireframe_subdivs_1) created and registered.
I1216 13:13:14.824987 46479 AssetAttributesManager.cpp:122] Asset attributes (uvSphereSolid : uvSphereSolid_rings_8_segments_16_useTexCoords_false_useTangents_false) created and registered.
I1216 13:13:14.825008 46479 AssetAttributesManager.cpp:122] Asset attributes (uvSphereWireframe : uvSphereWireframe_rings_16_segments_32) created and registered.
I1216 13:13:14.825016 46479 AssetAttributesManager.cpp:108] AssetAttributesManager::buildCtorFuncPtrMaps : Built default primitive asset templates : 12
I1216 13:13:14.825753 46479 PhysicsAttributesManager.cpp:39] File (./data/default.phys_scene_config.json) not found so new, default physics manager attributes created and registered.
I1216 13:13:14.825841 46479 StageAttributesManager.cpp:79] File (data/scene_datasets/habitat-test-scenes/skokloster-castle.glb) Based stage attributes created and registered.
I1216 13:13:14.825855 46479 Simulator.cpp:145] Loading navmesh from data/scene_datasets/habitat-test-scenes/skokloster-castle.navmesh
I1216 13:13:14.825937 46479 Simulator.cpp:147] Loaded.
I1216 13:13:14.825953 46479 SceneGraph.h:93] Created DrawableGroup:
I1216 13:13:15.701232 46479 ResourceManager.cpp:923] Importing Basis files as ASTC 4x4
I1216 13:13:16.869724 46479 ResourceManager.cpp:302] ResourceManager::loadStage : Not loading semantic mesh
W1216 13:13:16.869773 46479 Simulator.cpp:235] :
---
 The active scene does not contain semantic annotations.
---
I1216 13:13:16.870195 46479 simulator.py:181] Loaded navmesh data/scene_datasets/habitat-test-scenes/skokloster-castle.navmesh
2020-12-16 13:13:16,871 Initializing task Nav-v0
bind: Invalid command `enable-meta-key'.
Fatal Python error: Segmentation fault

Current thread 0x00007fe1ea3b0740 (most recent call first):
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 922 in create_module
  File "<frozen importlib._bootstrap>", line 571 in module_from_spec
  File "<frozen importlib._bootstrap>", line 658 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 955 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 971 in _find_and_load
  File "/home/justin/anaconda3/envs/habitat/lib/python3.6/pdb.py", line 154 in __init__
  File "/home/justin/anaconda3/envs/habitat/lib/python3.6/pdb.py", line 1585 in set_trace
  File "simple.py", line 16 in <module>
[1]    46479 segmentation fault (core dumped)
```

## Expected behavior

pdb should break in the code at line 16

## Additional context

This is running on a remote server, but it can display graphics on my local computer "ssh -X"

erikwijmans, 2020-12-16:
```
bind: Invalid command `enable-meta-key'.
Fatal Python error: Segmentation fault
```

That seems unrelated to habitat.  I tried the example script locally and it worked as expected.

erikwijmans, 2020-12-17:
>  but it can display graphics on my local computer "ssh -X".

Note that habitat does not work with X11 forwarding.  We have a headless install for remote servers.

Jbwasse2, 2020-12-17:
I will try this again on my local machine. Thank you for the warning also.

Jbwasse2, 2020-12-17:
I tried it again on my local machine and got the segmentation fault. I retried installing again on the remote computer, but with headless for habitat-sim and the debugger worked.

Feel free to close the issue if you'd like.

closed 2020-12-18 by Skylion007