facebookresearch/habitat-sim #981: 'habitat_sim._ext.habitat_sim_bindings.SimulatorConfiguration' object has no attribute 'scene_id'

Skylark0924, 2020-12-17:
## 🐛 Bug

## Command
Directly run the code in [ECCV 2020: Navigation example](https://colab.research.google.com/github/facebookresearch/habitat-sim/blob/master/examples/tutorials/colabs/ECCV_2020_Navigation.ipynb#scrollTo=g79noeyzn177) on my local PC as a python script.

## Additional context

## System Info

```
/home/lab/anaconda3/envs/habitat/bin/python /home/skylark/PycharmRemote/ReGReT/Test/eccv_2020_navigation.py
Traceback (most recent call last):
  File "/home/skylark/PycharmRemote/ReGReT/Test/eccv_2020_navigation.py", line 187, in <module>
    cfg = make_simple_cfg(sim_settings)
  File "/home/skylark/PycharmRemote/ReGReT/Test/eccv_2020_navigation.py", line 169, in make_simple_cfg
    sim_cfg.scene_id = settings["scene"]
AttributeError: 'habitat_sim._ext.habitat_sim_bindings.SimulatorConfiguration' object has no attribute 'scene_id'
```

 - OS (e.g., Linux): Ubuntu 18.04
 - How you installed PyTorch (`conda`, `pip`, source): pip
 - Build command you used (if compiling from source): 
 - Python version: 3.7
 - GPU models and configuration:
 - CUDA version: 10.2
 - PIP/conda dependencies packages versions
 - Any other relevant information:

**I wonder why I use the same version of habitat-sim (0.1.6) as the Colab, but got an error.**

Skylion007, 2020-12-18:
Your local habitat_sim is out of date. There was a breaking API change recently. Upgrading should fix it.

Skylark0924, 2020-12-19:
> Your local habitat_sim is out of date. There was a breaking API change recently. Upgrading should fix it. The collabs are running the nightly versions of habitat-sim.

Thanks for your reply! Does that mean I need to reinstall the habitat-sim as a nightly version? I will try again.

Skylark0924, 2020-12-19:
Well, after updating to a nightly version, I found another script that uses `habitat-lab` to create an environment has reported lots of unseen errors. Seem like the habitat-lab relies on a stable version of habitat-sim. 
```
2020-12-19 16:24:42,074 Initializing dataset ObjectNav-v1
2020-12-19 16:24:56,710 initializing sim Sim-v0
Traceback (most recent call last):
  File "/opt/pycharm-community-2019.3/plugins/python-ce/helpers/pydev/pydevd.py", line 1434, in _exec
    pydev_imports.execfile(file, globals, locals)  # execute the script
  File "/opt/pycharm-community-2019.3/plugins/python-ce/helpers/pydev/_pydev_imps/_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "/home/skylark/PycharmRemote/ReGReT/Env/habitat_env.py", line 675, in <module>
    example_Regret()
  File "/home/skylark/PycharmRemote/ReGReT/Env/habitat_env.py", line 600, in example_Regret
    env = HabitatEnv(env_config)
  File "/home/skylark/PycharmRemote/ReGReT/Env/habitat_env.py", line 46, in __init__
    self.eng = NavRLEnv(config=self.config)
  File "/home/skylark/PycharmRemote/ReGReT/Env/habitat_dataset/NavRL_env.py", line 44, in __init__
    super().__init__(self._core_env_config, dataset)
  File "/home/lab/Github/habitat-lab/habitat/core/env.py", line 331, in __init__
    self._env = Env(config, dataset)
  File "/home/lab/Github/habitat-lab/habitat/core/env.py", line 105, in __init__
    id_sim=self._config.SIMULATOR.TYPE, config=self._config.SIMULATOR
  File "/home/lab/Github/habitat-lab/habitat/sims/registration.py", line 19, in make_sim
    return _sim(**kwargs)
  File "/home/lab/Github/habitat-lab/habitat/sims/habitat_simulator/habitat_simulator.py", line 182, in __init__
    self.sim_config = self.create_sim_config(self._sensor_suite)
  File "/home/lab/Github/habitat-lab/habitat/sims/habitat_simulator/habitat_simulator.py", line 198, in create_sim_config
    sim_config.scene.id = self.habitat_config.SCENE
AttributeError: 'habitat_sim._ext.habitat_sim_bindings.SimulatorCon' object has no attribute 'scene'
Exception ignored in: <bound method Simulator.__del__ of HabitatSim(config=NOTHING, agents=NOTHING, _num_total_frames=NOTHING, _default_agent_id=NOTHING, _Simulator__sensors=NOTHING, _initialized=NOTHING, _previous_step_time=NOTHING, _Simulator__last_state=NOTHING)>
Traceback (most recent call last):
  File "/home/lab/anaconda3/envs/habitat/lib/python3.6/site-packages/habitat_sim/simulator.py", line 467, in __del__
    self.close()
  File "/home/lab/anaconda3/envs/habitat/lib/python3.6/site-packages/habitat_sim/simulator.py", line 118, in close
    for agent_sensorsuite in self.__sensors:
AttributeError: 'HabitatSim' object has no attribute '_Simulator__sensors'

Process finished with exit code 1
```

However, I just update my `habitat-lab` days ago.

Skylark0924, 2020-12-19:
Update to the nightly version is actually the solution to this issue! Thanks a lot!

However, other problems may cause by the update as I wrote in the previous comment.

erikwijmans, 2020-12-19:
The current stable tags for both habitat-sim and habitat-lab match with each other and are compatible, the current tip of habitat-sim and habitat-lab match, but you current can't mix and match those because of an BC breaking change between now and the last release.

Skylark0924, 2020-12-20:
@erikwijmans Then I will wait for your update about a stable version，cheer for you guys！(ง •̀_•́)ง！ Lol

closed 2020-12-20 by Skylark0924