facebookresearch/habitat-sim #988: Rendering equirectangular images from replica in HDR (.hdr or .exr)

elientumba2019, 2020-12-20:
## ❓ Questions and Help
 Hello. Thanks for this amazing platform.

I have the following questions :　
1) How do I render hdr (.hdr, .exr not rgba) images from the replica dataset using habitat-sim? 
2) How do I render equirectangular images? 

Thank you

Skylion007, 2020-12-20:
1. See the example from Habitat-Lab about how to render equirect images from a cubemap. 
2. Pretty sure Magnum supports reading/writing HDR files, but I don't know if we actually support HDR rendering in Habtiat-Sim. @bigbike @mosra

elientumba2019, 2020-12-23:
I have checked all the examples available in the repo and could not find the one about how to render equirectangular images from a cube map. could you please point me to that resource? Thank you 
@Skylion007

elientumba2019, 2020-12-23:
@bigbike , @mosra . Hello, I hope you are well. Does habitat support writing HDR files? exr, or hdr. Specifically for the replica dataset as it comes with high dynamic range textures. Thank you

waldenlakes, 2020-12-27:
@mosra @bigbike I also want to know how to render HDR images. Looking forward to your help. Thank you!

mosra, 2020-12-29:
Magnum does support [writing HDR and EXR files](https://doc.magnum.graphics/magnum/file-formats.html) but Habitat isn't ready for that right now -- it always renders to a 8-bit framebuffer. It would need to be updated ([here](https://github.com/facebookresearch/habitat-sim/blob/a3820c4ea7dad72026d8b2a6562349fd31b3ac8b/src/esp/gfx/RenderTarget.cpp#L64), [here](https://github.com/facebookresearch/habitat-sim/blob/d09b0bdbc6404b0f19c83d6ef61f178b27e48fe2/src/esp/sensor/CameraSensor.cpp#L193) and probably some other places as well) to add an option for 16-/32-bit float output next to 8-bit.

waldenlakes, 2021-01-05:
> Magnum does support [writing HDR and EXR files](https://doc.magnum.graphics/magnum/file-formats.html) but Habitat isn't ready for that right now -- it always renders to a 8-bit framebuffer. It would [...]

Thanks alot for your help and I have been able to render HDR images.

Skylion007, 2021-01-05:
@waldenlakes PR welcome.