facebookresearch/habitat-sim #996: Exporting the scene as a mesh after adding objects

Ainaz99, 2020-12-27:
## ❓ Questions and Help
Hi,
Is there a way to export the scene as a mesh after adding some objects? 
I am using one of the Replica meshes as the initial scene, and then add multiple objects to it. I want to know if there is a way I can save the scene as a mesh after adding the objects.
Thanks for your response.

jturner65, 2020-12-28:
We are currently [nearing completion on a PR that introduces Scene Instances](https://github.com/facebookresearch/habitat-sim/pull/954), which allows for loading a particular stage, objects, lighting and navmesh in a specific configuration described in a JSON config file.  Saving Scene Instances in the same format is next on the horizon.

Actually saving the stage and objects as a new mesh is not something we currently have on the radar.  If you would like to modify the actual mesh to include objects, you probably want to use blender or some other authoring tool for that.

Ainaz99, 2020-12-28:
Thanks for your response.

closed 2020-12-28 by Ainaz99