facebookresearch/habitat-sim #1004: Mapping object locations from habitat to blender

Ainaz99, 2021-01-03:
## ❓ Questions and Help
Hi. Happy new year :)
I'm trying to add some objects to a replica scene, both in habitat and blender. I know how to map the coordinate system between them. Object rotations seem to be matching, but there is some discrepancy between object locations in habitat and blender. I'm using `sim.set_translation(object_id)` in habitat and `obj.location` in blender which are perfectly matching after mapping the coordinate system. Is there any other way to set the object location in habitat? 
I'm attaching some photos from both habitat and blender (from left to right, the first image is from habitat, the second is from blender and they are merged in the third image).
Thanks for your help.

[image]
[image]
[image]

aclegg3, 2021-01-04:
Hey @Ainaz99,

Great debugging images. thanks. It looks like your Blender objects are not centered at their local origins. As in this image, the object origin and transformation pivot is at the edge of the box. 

[image]

When an object is loaded into Habitat, we move the origin to be coincident with the object's center of mass (COM) for physics simulation purposes (computed from the center of the object's bounding box if not provided in the config file). 

**You have a couple choices for handling this:**

You can reset the origin of your Blender reference objects with right click as shown below:
[image]

You can manually set the object's COM (e.g. via the object config file as described [here](https://aihabitat.org/docs/habitat-sim/attributesJSON.html#object-attributes)). Note that this approach will result in physics simulation artifacts and should be used only for kinematic objects.

You can re-export your objects from Blender centered at their COM using the Blender operation shown here and then use the edited object assets in Habitat:
[image]

Hope this helps. 👍

Ainaz99, 2021-01-04:
Changing the origin in Blender worked! Thanks a lot @aclegg3 .

closed 2021-01-04 by aclegg3

qingchu123, 2025-02-23:
now i try to map the object in blender,but i encounter some problem, can you @Ainaz99 give some suggestion on how to map the coordinate system between them?