facebookresearch/habitat-sim #1012: Progmesh utility for replica dataset

Ainaz99, 2021-01-07:
## ❓ Questions and Help
Hi.
As you know, most of the replica scenes crash with bullet physics. So I'm wondering if the progmesh utility works for reducing the complexity of replica meshes and using them with bullet physics. 
I used progmesh for one of the replica scenes and tried to use the output with bullet physics, but here's the error I get:

```
==== Initialized Sensor Spec: =====
Sensor uuid:  color_sensor
Sensor type:  SensorType.COLOR
Sensor position:  [0. 0. 0.]
Sensor resolution:  [512 512]
===================================
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0107 17:59:35.397192 91519 AssetAttributesManager.cpp:117] Asset attributes (capsule3DSolid : capsule3DSolid_hemiRings_4_cylRings_1_segments_12_halfLen_0.75_useTexCoords_false_useTangents_false) created and registered.
I0107 17:59:35.397265 91519 AssetAttributesManager.cpp:117] Asset attributes (capsule3DWireframe : capsule3DWireframe_hemiRings_8_cylRings_1_segments_16_halfLen_1) created and registered.
I0107 17:59:35.397302 91519 AssetAttributesManager.cpp:117] Asset attributes (coneSolid : coneSolid_segments_12_halfLen_1.25_rings_1_useTexCoords_false_useTangents_false_capEnd_true) created and registered.
I0107 17:59:35.397325 91519 AssetAttributesManager.cpp:117] Asset attributes (coneWireframe : coneWireframe_segments_32_halfLen_1.25) created and registered.
I0107 17:59:35.397338 91519 AssetAttributesManager.cpp:117] Asset attributes (cubeSolid : cubeSolid) created and registered.
I0107 17:59:35.397349 91519 AssetAttributesManager.cpp:117] Asset attributes (cubeWireframe : cubeWireframe) created and registered.
I0107 17:59:35.397382 91519 AssetAttributesManager.cpp:117] Asset attributes (cylinderSolid : cylinderSolid_rings_1_segments_12_halfLen_1_useTexCoords_false_useTangents_false_capEnds_true) created and registered.
I0107 17:59:35.397408 91519 AssetAttributesManager.cpp:117] Asset attributes (cylinderWireframe : cylinderWireframe_rings_1_segments_32_halfLen_1) created and registered.
I0107 17:59:35.397426 91519 AssetAttributesManager.cpp:117] Asset attributes (icosphereSolid : icosphereSolid_subdivs_1) created and registered.
I0107 17:59:35.397441 91519 AssetAttributesManager.cpp:117] Asset attributes (icosphereWireframe : icosphereWireframe_subdivs_1) created and registered.
I0107 17:59:35.397461 91519 AssetAttributesManager.cpp:117] Asset attributes (uvSphereSolid : uvSphereSolid_rings_8_segments_16_useTexCoords_false_useTangents_false) created and registered.
I0107 17:59:35.397480 91519 AssetAttributesManager.cpp:117] Asset attributes (uvSphereWireframe : uvSphereWireframe_rings_16_segments_32) created and registered.
I0107 17:59:35.397488 91519 AssetAttributesManager.cpp:105] AssetAttributesManager::buildCtorFuncPtrMaps : Built default primitive asset templates : 12
I0107 17:59:35.399307 91519 AttributesManagerBase.h:168] AttributesManager::loadAllFileBasedTemplates : Load Object template: /scratch-data/ainaz/habitat-sim/data/objects/cheezit.phys_properties.json
I0107 17:59:35.400321 91519 AttributesManagerBase.h:179] AttributesManager::loadAllFileBasedTemplates : Loaded file-based Object templates: 1
I0107 17:59:35.400673 91519 AttributesManagerBase.h:168] AttributesManager::loadAllFileBasedTemplates : Load Object template: /scratch-data/ainaz/habitat-sim/data/objects/chefcan.phys_properties.json
I0107 17:59:35.401590 91519 AttributesManagerBase.h:179] AttributesManager::loadAllFileBasedTemplates : Loaded file-based Object templates: 1
I0107 17:59:35.401612 91519 PhysicsAttributesManager.cpp:27] JSON File (/scratch-data/ainaz/habitat-sim/data/default.phys_scene_config.json) Based physics manager attributes created and registered.
I0107 17:59:35.404142 91519 AbstractObjectAttributesManagerBase.h:179] File (/scratch-data/ainaz/replica-google-objects/hotel_0/mesh2.ply) not parsed, so new default Stage attributes created and registered.
W0107 17:59:35.404155 91519 Simulator.cpp:148] Navmesh file not found, checked at 
I0107 17:59:35.404173 91519 SceneGraph.h:93] Created DrawableGroup: 
Renderer: Tesla V100-SXM2-32GB/PCIe/SSE2 by NVIDIA Corporation
OpenGL version: 4.6.0 NVIDIA 440.64
Using optional features:
    GL_ARB_ES2_compatibility
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_invalidate_subdata
    GL_ARB_multi_bind
    GL_ARB_robustness
    GL_ARB_separate_shader_objects
    GL_ARB_texture_filter_anisotropic
    GL_ARB_texture_storage
    GL_ARB_texture_storage_multisample
    GL_ARB_vertex_array_object
    GL_KHR_debug
Using driver workarounds:
    no-forward-compatible-core-context
    nv-egl-incorrect-gl11-function-pointers
    no-layout-qualifiers-on-old-glsl
    nv-zero-context-profile-mask
    nv-implementation-color-read-format-dsa-broken
    nv-cubemap-inconsistent-compressed-image-size
    nv-cubemap-broken-full-compressed-image-query
    nv-compressed-block-size-in-bits
creating staticStageObject_
creating staticStageObject_ .. done
I0107 17:59:35.455085 91519 ResourceManager.cpp:962] Importing Basis files as BC7
Trade::AbstractImporter::object3D(): index 0 out of range for 0 entries
jobs/create_google_objects_in_context_job.sh: line 22: 91519 Aborted  (core dumped) 

```
I'm not sure what is wrong in here, and I would really appreciate if you could help me fix the problem.
Thanks a lot.

aclegg3, 2021-01-12:
Hey, @Ainaz99. 👋 

Can you share the asset you are trying to load?

Looks like an asset import issue in the Magnum importer (for .obj?). Maybe related to bad formatting in the progmesh output? 

Also, should you get this working for collision, you should consider setting up a [stage_config.json](https://aihabitat.org/docs/habitat-sim/attributesJSON.html#stage-attributes) with the original Replica asset as the `render_asset` and the decimated version as the `collision_asset` for highest fidelity simulation.

Ainaz99, 2021-01-13:
Hi @aclegg3, thanks for your response.
Apparently, I was doing it the wrong way, I was not using the stage_config file.
So now here's my `mesh.stage_config.json` file.
```
{
        "render_asset": "mesh.ply",
        "collision_asset": "mesh2.ply",
        "up":[0,1,0],
        "front":[0,0,-1],
        "origin":[0,0,0],
        "scale":[2,2,2],
        "gravity":[0,-9.8,0],
        "margin":0.03,
        "friction_coefficient": 0.3,
        "restitution_coefficient": 0.3,
        "units_to_meters":1.0,
        "requires_lighting":true
}
```

And this is the directory containing the original Replica assets and the decimated version (`mesh2.ply`).
```
hotel_0/
|-- glass.sur
|-- mesh.ply
|-- mesh.stage_config.json
|-- mesh2.ply
|-- preseg.bin
|-- preseg.json
|-- semantic.bin
|-- semantic.json
|-- habitat
|   |-- info_preseg_semantic.json
|   |-- info_semantic.json
|   |-- mesh_preseg_semantic.navmesh
|   |-- mesh_preseg_semantic.ply
|   |-- mesh_semantic.navmesh
|   |-- mesh_semantic.ply
|   `-- sorted_faces.bin
`-- textures
    |-- 0-color-ptex.hdr
    |-- 1-color-ptex.hdr
    |-- 2-color-ptex.hdr
    |-- 3-color-ptex.hdr
    |-- 4-color-ptex.hdr
    |-- 5-color-ptex.hdr
    `-- parameters.json

```

And here's the error I get when constructing the simulator:
```
E0113 11:18:09.974095 109893 StageAttributesManager.cpp:39] StageAttributesManager::registerObjectFinalize : Attributes template named/scratch-data/ainaz/replica-google-objects/hotel_0/mesh.stage_config.jsondoes not have a valid render asset handle specified. Aborting.
```
Is there anything else I'm doing wrong in here?

Thanks in advance.

jturner65, 2021-01-13:
Hi @Ainaz99 

What version of Habitat-sim are you using?

Ainaz99, 2021-01-13:
Hi @jturner65,

I'm using version 0.1.6 .

jturner65, 2021-01-13:
Your json config that you shared above is configured for the current master version of sim and is not compatible with your version.  

For the version you are using, look [at this file](https://github.com/facebookresearch/habitat-sim/blob/v0.1.6/data/test_assets/scenes/stage_floor1.stage_config.json) for format.  To specify collision mesh asset use the tag "collision mesh" instead of "collision_asset".

Give that a try and please let me know what you find :).

Ainaz99, 2021-01-13:
Changing the tags to `render mesh` and `collision mesh` worked, thanks.
 But now I'm getting another error while creating the simulator, which I think it's because of the output of progmesh which is set as the collision mesh (mesh2.ply):

```
I0113 18:37:44.755409 113127 ResourceManager.cpp:962] Importing Basis files as BC7
Trade::AbstractImporter::object3D(): index 0 out of range for 0 entries
```

Here's the full error:

```
==== Initialized Sensor Spec: =====
Sensor uuid:  color_sensor
Sensor type:  SensorType.COLOR
Sensor position:  [0. 0. 0.]
Sensor resolution:  [512 512]
===================================
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0113 18:37:39.387195 113127 AssetAttributesManager.cpp:117] Asset attributes (capsule3DSolid : capsule3DSolid_hemiRings_4_cylRings_1_segments_12_halfLen_0.75_useTexCoords_false_useTangents_false) created and registered.
I0113 18:37:39.387508 113127 AssetAttributesManager.cpp:117] Asset attributes (capsule3DWireframe : capsule3DWireframe_hemiRings_8_cylRings_1_segments_16_halfLen_1) created and registered.
I0113 18:37:39.387706 113127 AssetAttributesManager.cpp:117] Asset attributes (coneSolid : coneSolid_segments_12_halfLen_1.25_rings_1_useTexCoords_false_useTangents_false_capEnd_true) created and registered.
I0113 18:37:39.387872 113127 AssetAttributesManager.cpp:117] Asset attributes (coneWireframe : coneWireframe_segments_32_halfLen_1.25) created and registered.
I0113 18:37:39.387984 113127 AssetAttributesManager.cpp:117] Asset attributes (cubeSolid : cubeSolid) created and registered.
I0113 18:37:39.388303 113127 AssetAttributesManager.cpp:117] Asset attributes (cubeWireframe : cubeWireframe) created and registered.
I0113 18:37:39.388518 113127 AssetAttributesManager.cpp:117] Asset attributes (cylinderSolid : cylinderSolid_rings_1_segments_12_halfLen_1_useTexCoords_false_useTangents_false_capEnds_true) created and registered.
I0113 18:37:39.388767 113127 AssetAttributesManager.cpp:117] Asset attributes (cylinderWireframe : cylinderWireframe_rings_1_segments_32_halfLen_1) created and registered.
I0113 18:37:39.388862 113127 AssetAttributesManager.cpp:117] Asset attributes (icosphereSolid : icosphereSolid_subdivs_1) created and registered.
I0113 18:37:39.389199 113127 AssetAttributesManager.cpp:117] Asset attributes (icosphereWireframe : icosphereWireframe_subdivs_1) created and registered.
I0113 18:37:39.389396 113127 AssetAttributesManager.cpp:117] Asset attributes (uvSphereSolid : uvSphereSolid_rings_8_segments_16_useTexCoords_false_useTangents_false) created and registered.
I0113 18:37:39.389482 113127 AssetAttributesManager.cpp:117] Asset attributes (uvSphereWireframe : uvSphereWireframe_rings_16_segments_32) created and registered.
I0113 18:37:39.389569 113127 AssetAttributesManager.cpp:105] AssetAttributesManager::buildCtorFuncPtrMaps : Built default primitive asset templates : 12
I0113 18:37:39.393952 113127 PhysicsAttributesManager.cpp:27] JSON File (/scratch-data/ainaz/habitat-sim/data/default.phys_scene_config.json) Based physics manager attributes created and registered.
I0113 18:37:39.399041 113127 AbstractObjectAttributesManagerBase.h:179] JSON File (/scratch-data/ainaz/replica-google-objects/hotel_0/mesh.stage_config.json) Based Stage attributes created and registered.
W0113 18:37:39.399472 113127 Simulator.cpp:148] Navmesh file not found, checked at 
I0113 18:37:39.399785 113127 SceneGraph.h:93] Created DrawableGroup: 
Renderer: Tesla V100-SXM2-32GB/PCIe/SSE2 by NVIDIA Corporation
OpenGL version: 4.6.0 NVIDIA 440.64
Using optional features:
    GL_ARB_ES2_compatibility
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_invalidate_subdata
    GL_ARB_multi_bind
    GL_ARB_robustness
    GL_ARB_separate_shader_objects
    GL_ARB_texture_filter_anisotropic
    GL_ARB_texture_storage
    GL_ARB_texture_storage_multisample
    GL_ARB_vertex_array_object
    GL_KHR_debug
Using driver workarounds:
    no-forward-compatible-core-context
    nv-egl-incorrect-gl11-function-pointers
    no-layout-qualifiers-on-old-glsl
    nv-zero-context-profile-mask
    nv-implementation-color-read-format-dsa-broken
    nv-cubemap-inconsistent-compressed-image-size
    nv-cubemap-broken-full-compressed-image-query
    nv-compressed-block-size-in-bits
creating staticStageObject_
creating staticStageObject_ .. done
I0113 18:37:39.597506 113127 PTexMeshData.cpp:476] Splitting mesh... 
I0113 18:37:40.356065 113127 PTexMeshData.cpp:383] The number of quads: 1096649, which equals to 2193298 triangles.
I0113 18:37:40.360316 113127 PTexMeshData.cpp:832] Loading mesh 1/6... 
I0113 18:37:40.455554 113127 PTexMeshData.cpp:832] Loading mesh 2/6... 
I0113 18:37:40.475643 113127 PTexMeshData.cpp:832] Loading mesh 3/6... 
I0113 18:37:40.480320 113127 PTexMeshData.cpp:832] Loading mesh 4/6... 
I0113 18:37:40.487354 113127 PTexMeshData.cpp:832] Loading mesh 5/6... 
I0113 18:37:40.558043 113127 PTexMeshData.cpp:832] Loading mesh 6/6... 
I0113 18:37:40.562263 113127 PTexMeshData.cpp:858] Calculating mesh adjacency... 
I0113 18:37:42.207960 113127 PTexMeshData.cpp:905] loading atlas textures: 
I0113 18:37:42.209473 113127 PTexMeshData.cpp:914] Loading atlas 1/6 from /scratch-data/ainaz/replica-google-objects/hotel_0/textures/0-color-ptex.hdr. 
I0113 18:37:42.660079 113127 PTexMeshData.cpp:914] Loading atlas 2/6 from /scratch-data/ainaz/replica-google-objects/hotel_0/textures/1-color-ptex.hdr. 
[... cut ...]
I0113 18:37:43.857050 113127 PTexMeshData.cpp:914] Loading atlas 5/6 from /scratch-data/ainaz/replica-google-objects/hotel_0/textures/4-color-ptex.hdr. 
I0113 18:37:44.384876 113127 PTexMeshData.cpp:914] Loading atlas 6/6 from /scratch-data/ainaz/replica-google-objects/hotel_0/textures/5-color-ptex.hdr. 
I0113 18:37:44.755409 113127 ResourceManager.cpp:962] Importing Basis files as BC7
Trade::AbstractImporter::object3D(): index 0 out of range for 0 entries
jobs/create_rgb_images_replica_google_objects_job.sh: line 22: 113127 Aborted                 (core dumped)
```

closed 2021-01-27 by Ainaz99