facebookresearch/habitat-sim #1017: Cannot install habitat-sim 0.1.2

AssemSadek, 2021-01-11:
# Reproduce
1. download source zip file of 0.1.2 from https://github.com/facebookresearch/habitat-sim/releases/tag/v0.1.2
2. unzip and cd to habitat-sim
3. Install numpy in your python env of choice (e.g., pip install numpy or conda install numpy)
4. Install Habitat-Sim via python setup.py install in your python env of choice (note: python 3 is required)
Use python setup.py install --headless for headless systems (i.e. without an attached display) or if you need multi-gpu support.

# Operating system
* MACOS Big Sur
* Linux Redhat

# Output

running install
running bdist_egg
running egg_info
writing habitat_sim.egg-info/PKG-INFO
writing dependency_links to habitat_sim.egg-info/dependency_links.txt
writing requirements to habitat_sim.egg-info/requires.txt
writing top-level names to habitat_sim.egg-info/top_level.txt
reading manifest file 'habitat_sim.egg-info/SOURCES.txt'
writing manifest file 'habitat_sim.egg-info/SOURCES.txt'
installing library code to build/bdist.macosx-10.9-x86_64/egg
running install_lib
running build_py
running build_ext
fatal: not a git repository (or any of the parent directories): .git
make: *** No targets specified and no makefile found.  Stop.
Traceback (most recent call last):
  File "setup.py", line 325, in <module>
    zip_safe=False,
  File "/Users/asadek/opt/anaconda3/envs/habitat_0_1_2_env/lib/python3.6/site-packages/setuptools/__init__.py", line 153, in setup
    return distutils.core.setup(**attrs)
  File "/Users/asadek/opt/anaconda3/envs/habitat_0_1_2_env/lib/python3.6/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/Users/asadek/opt/anaconda3/envs/habitat_0_1_2_env/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/Users/asadek/opt/anaconda3/envs/habitat_0_1_2_env/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/Users/asadek/opt/anaconda3/envs/habitat_0_1_2_env/lib/python3.6/site-packages/setuptools/command/install.py", line 67, in run
    self.do_egg_install()
  File "/Users/asadek/opt/anaconda3/envs/habitat_0_1_2_env/lib/python3.6/site-packages/setuptools/command/install.py", line 109, in do_egg_install
    self.run_command('bdist_egg')
  File "/Users/asadek/opt/anaconda3/envs/habitat_0_1_2_env/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/Users/asadek/opt/anaconda3/envs/habitat_0_1_2_env/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/Users/asadek/opt/anaconda3/envs/habitat_0_1_2_env/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 167, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/Users/asadek/opt/anaconda3/envs/habitat_0_1_2_env/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 153, in call_command
    self.run_command(cmdname)
  File "/Users/asadek/opt/anaconda3/envs/habitat_0_1_2_env/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/Users/asadek/opt/anaconda3/envs/habitat_0_1_2_env/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/Users/asadek/opt/anaconda3/envs/habitat_0_1_2_env/lib/python3.6/site-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/Users/asadek/opt/anaconda3/envs/habitat_0_1_2_env/lib/python3.6/distutils/command/install_lib.py", line 107, in build
    self.run_command('build_ext')
  File "/Users/asadek/opt/anaconda3/envs/habitat_0_1_2_env/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/Users/asadek/opt/anaconda3/envs/habitat_0_1_2_env/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "setup.py", line 166, in run
    self.build_extension(ext)
  File "setup.py", line 220, in build_extension
    shlex.split("cmake --build {}".format(self.build_temp)) + build_args
  File "/Users/asadek/opt/anaconda3/envs/habitat_0_1_2_env/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', 'build', '--config', 'RelWithDebInfo', '--', '-j']' returned non-zero exit status 2.

dhruvbatra, 2021-01-11:
Out of curiosity, why are you trying to install such an old version?

Skylion007, 2021-01-11:
It needs to be a git repository or else it can't update the submodules.

AssemSadek, 2021-01-12:
> It needs to be a git repository or else it can't update the submodules. You would need to clone the git repo and checkout that tag instead.

I checkout and i got a similar error 

[ 81%] Linking CXX static library libgeo.a
[ 81%] Built target geo
gmake: *** [all] Error 2
Traceback (most recent call last):
  File "setup.py", line 313, in <module>
    setup(
  File "/home/asadek/miniconda3/lib/python3.8/site-packages/setuptools/__init__.py", line 153, in setup
    return distutils.core.setup(**attrs)
  File "/home/asadek/miniconda3/lib/python3.8/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/home/asadek/miniconda3/lib/python3.8/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/home/asadek/miniconda3/lib/python3.8/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/asadek/miniconda3/lib/python3.8/site-packages/setuptools/command/install.py", line 67, in run
    self.do_egg_install()
  File "/home/asadek/miniconda3/lib/python3.8/site-packages/setuptools/command/install.py", line 109, in do_egg_install
    self.run_command('bdist_egg')
  File "/home/asadek/miniconda3/lib/python3.8/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/asadek/miniconda3/lib/python3.8/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/asadek/miniconda3/lib/python3.8/site-packages/setuptools/command/bdist_egg.py", line 167, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/home/asadek/miniconda3/lib/python3.8/site-packages/setuptools/command/bdist_egg.py", line 153, in call_command
    self.run_command(cmdname)
  File "/home/asadek/miniconda3/lib/python3.8/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/asadek/miniconda3/lib/python3.8/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/asadek/miniconda3/lib/python3.8/site-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/home/asadek/miniconda3/lib/python3.8/distutils/command/install_lib.py", line 107, in build
    self.run_command('build_ext')
  File "/home/asadek/miniconda3/lib/python3.8/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/asadek/miniconda3/lib/python3.8/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "setup.py", line 166, in run
    self.build_extension(ext)
  File "setup.py", line 219, in build_extension
    subprocess.check_call(
  File "/home/asadek/miniconda3/lib/python3.8/subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', 'build', '--config', 'RelWithDebInfo', '--', '-j']' returned non-zero exit status 2.

AssemSadek, 2021-01-12:
> Out of curiosity, why are you trying to install such an old version?

@dhruvbatra 

There's a previous work that depends on 0.1.2. I want to rerun it and reproduce the results. 

When I use 0.1.6, I got this error:

from habitat.sims.habitat_simulator import SimulatorActions
ImportError: cannot import name 'SimulatorActions'

So I think there's a lot of revamps that was made between 0.1.2 and 0.1.6 :D

Skylion007, 2021-01-13:
The only information that stack trace gives me is that gmake had an error during compilation and returned exit code 2. The actual error is probably higher up.

closed 2021-05-11 by AssemSadek