facebookresearch/habitat-sim #1019: how to find second or third floor heights

mmahdavian, 2021-01-11:
Hello

I am trying to use replica and gibson datasets in habitat-sim and some of the environments have more than one floor. sim.pathfinder.get_bounds()[0][1] gives me the height of first floor. But I am trying to generate random paths in different floors and save their rgb-image. So I need to know other floor heights too. So I can have it as input to my path generator. Is there any automatic(or manual ) way to find other floors height too?

Thank You

erikwijmans, 2021-01-13:
We don't have any automatic ways to do this.  There isn't a simple path to enter the data manually, but it of course could be done.

srama2512, 2021-01-14:
@mmahdavian  - This might be possible in MP3D since it has annotations for "SemanticLevels". You can access `sim.semantic_scene.levels[i].aabb.center[1]` to get the center of the room bounding box. However, note that a SemanticLevel can span various Y values, and may not correspond to our notion of Y-based floor definitions. You might have to search within this bounding box for finding navigable space.

closed 2021-08-16 by dhruvbatra