facebookresearch/habitat-sim #1020: Adding camera to an agent(loaded via URDF)

sasayesh, 2021-01-12:
## ❓ Questions and Help
Hello,
I am trying to import a robotic arm via URDF. I followed the URDF [tutorial](https://github.com/facebookresearch/habitat-sim/blob/articulated-objects-prototype/examples/tutorials/URDF_robotics_tutorial.py) and I can control it using my custom controller. It would be appreciated to give me a pointer on how I can attach a camera to the robot(similar to [embodied agent](https://aihabitat.org/docs/habitat-sim/rigid-object-tutorial.html#embodied-agents)). Does current implementation accept camera node definition in URDF? 

Thank you!
@msavva
@eundersander
@aclegg3

aclegg3, 2021-01-12:
Currently this is not supported in the articulated object prototype. There is ongoing development on the Habitat Sensor API to support general attachment to arbitrary SceneNodes of choice. Once this is complete you should be able to manually attach your sensors to the SceneNode for a link. We are looking into a URDF sensor API as well, but this is work in progress at this time.

sasayesh, 2021-01-12:
Thanks for the update.
What if I import the robot, not through URDF parser. I am thinking about something like LOCOBOT. Could you please show me the starting point or previous implementations for LOCOBOT?

aclegg3, 2021-01-12:
For a recent discussion on URDF Locobot examples, see issue #974. 

If you want to control the agent on the NavMesh (velocity or position control without wheel dynamics) you can checkout some of our [tutorials](https://aihabitat.org/tutorial/2020/) particularly the "Habitat-Sim Basics for Navigation" and "Habitat-Sim for Interaction" topics.

You can also manually sync the sensors to the transform of a particular URDF link by using the `get_articulated_link_rigid_state` function and setting the transformation of the agent or camera SceneNode manually at each frame. An example of manual camera transformation can be found in the "Advanced Topic : Motion Tracking Camera" example in the "Habitat-Sim Advanced Topics" tutorial topic.

closed 2021-08-16 by dhruvbatra