facebookresearch/habitat-sim #1038: Ray Rllib Integration (Fork/Exec)

smorad, 2021-01-19:
## ❓ Questions and Help
I'm trying to run parallel instances of habitat using ray rllib. Rllib creates a new worker process for each environment, but unfortunately only a single instance of habitat works (logs below). Each instance is allocated its own GPU. I assume this is related to https://github.com/facebookresearch/habitat-sim/issues/664, but the Rllib multiprocessing modules are written in C++ and do not use python's `multiprocessing` module. I believe the `fork` call occurs here: https://github.com/ray-project/ray/blob/86d5000047593439c768d35565ece702b1b62162/src/ray/util/process.cc#L97

I'm willing to put in some effort to fix this if anyone has an idea of how to solve this.

Failure:
```
2021-01-19 22:27:00,827 Initializing dataset ObjectNav-v1
--
  | Starting habitat instance on gpu 3
  | 2021-01-19 22:34:33,684 initializing sim Sim-v0
  | WARNING: Logging before InitGoogleLogging() is written to STDERR
  | I0119 22:34:33.688239 24290 ManagedContainerBase.cpp:19] ManagedContainerBase::convertFilenameToJSON : Filename : default changed to proposed JSON configuration filename : default.scene_dataset_config.json
  | I0119 22:34:33.688387 24290 AssetAttributesManager.cpp:117] Asset attributes (capsule3DSolid : capsule3DSolid_hemiRings_4_cylRings_1_segments_12_halfLen_0.75_useTexCoords_false_useTangents_false) created and registered.
  | I0119 22:34:33.688447 24290 AssetAttributesManager.cpp:117] Asset attributes (capsule3DWireframe : capsule3DWireframe_hemiRings_8_cylRings_1_segments_16_halfLen_1) created and registered.
  | I0119 22:34:33.688495 24290 AssetAttributesManager.cpp:117] Asset attributes (coneSolid : coneSolid_segments_12_halfLen_1.25_rings_1_useTexCoords_false_useTangents_false_capEnd_true) created and registered.
  | I0119 22:34:33.688529 24290 AssetAttributesManager.cpp:117] Asset attributes (coneWireframe : coneWireframe_segments_32_halfLen_1.25) created and registered.
  | I0119 22:34:33.688547 24290 AssetAttributesManager.cpp:117] Asset attributes (cubeSolid : cubeSolid) created and registered.
  | I0119 22:34:33.688565 24290 AssetAttributesManager.cpp:117] Asset attributes (cubeWireframe : cubeWireframe) created and registered.
  | I0119 22:34:33.688647 24290 AssetAttributesManager.cpp:117] Asset attributes (cylinderSolid : cylinderSolid_rings_1_segments_12_halfLen_1_useTexCoords_false_useTangents_false_capEnds_true) created and registered.
  | I0119 22:34:33.688704 24290 AssetAttributesManager.cpp:117] Asset attributes (cylinderWireframe : cylinderWireframe_rings_1_segments_32_halfLen_1) created and registered.
  | I0119 22:34:33.688725 24290 AssetAttributesManager.cpp:117] Asset attributes (icosphereSolid : icosphereSolid_subdivs_1) created and registered.
  | I0119 22:34:33.688747 24290 AssetAttributesManager.cpp:117] Asset attributes (icosphereWireframe : icosphereWireframe_subdivs_1) created and registered.
  | I0119 22:34:33.688778 24290 AssetAttributesManager.cpp:117] Asset attributes (uvSphereSolid : uvSphereSolid_rings_8_segments_16_useTexCoords_false_useTangents_false) created and registered.
  | I0119 22:34:33.688803 24290 AssetAttributesManager.cpp:117] Asset attributes (uvSphereWireframe : uvSphereWireframe_rings_16_segments_32) created and registered.
  | I0119 22:34:33.688817 24290 AssetAttributesManager.cpp:105] AssetAttributesManager::buildCtorFuncPtrMaps : Built default primitive asset templates : 12
  | I0119 22:34:33.689654 24290 SceneDatasetAttributesManager.cpp:23] File (default) not found, so new default dataset attributes created and registered.
  | I0119 22:34:33.689661 24290 MetadataMediator.cpp:47] MetadataMediator::createDataset : Dataset default successfully created.
  | I0119 22:34:33.694871 24290 PhysicsAttributesManager.cpp:27] File (./data/default.physics_config.json) not found, so new default physics manager attributes created and registered.
  | I0119 22:34:33.694888 24290 ManagedContainerBase.cpp:19] ManagedContainerBase::convertFilenameToJSON : Filename : /root/habitat-lab/data/scene_datasets/mp3d/sT4fr6TAbpF/sT4fr6TAbpF.glb changed to proposed JSON configuration filename : /root/habitat-lab/data/scene_datasets/mp3d/sT4fr6TAbpF/sT4fr6TAbpF.stage_config.json
  | I0119 22:34:33.695024 24290 AbstractObjectAttributesManagerBase.h:180] File (/root/habitat-lab/data/scene_datasets/mp3d/sT4fr6TAbpF/sT4fr6TAbpF.glb) exists but is not a recognized config filename extension, so new default Stage attributes created and registered.
  | I0119 22:34:33.695040 24290 Simulator.cpp:156] Loading navmesh from /root/habitat-lab/data/scene_datasets/mp3d/sT4fr6TAbpF/sT4fr6TAbpF.navmesh
  | I0119 22:34:33.695317 24290 Simulator.cpp:158] Loaded.
  | I0119 22:34:33.695339 24290 SceneGraph.h:93] Created DrawableGroup:
  | Platform::WindowlessEglApplication::tryCreateContext(): unable to find EGL device for CUDA device 3
  | WindowlessContext: Unable to create windowless context
```

Success

```
  | Starting habitat instance on gpu 0
-- | --
  | 2021-01-19 22:27:00,825 Initializing dataset ObjectNav-v1
  | 2021-01-19 22:34:34,703 initializing sim Sim-v0
  | WARNING: Logging before InitGoogleLogging() is written to STDERR
  | I0119 22:34:34.707082 24322 ManagedContainerBase.cpp:19] ManagedContainerBase::convertFilenameToJSON : Filename : default changed to proposed JSON configuration filename : default.scene_dataset_config.json
  | I0119 22:34:34.707244 24322 AssetAttributesManager.cpp:117] Asset attributes (capsule3DSolid : capsule3DSolid_hemiRings_4_cylRings_1_segments_12_halfLen_0.75_useTexCoords_false_useTangents_false) created and registered.
  | I0119 22:34:34.707302 24322 AssetAttributesManager.cpp:117] Asset attributes (capsule3DWireframe : capsule3DWireframe_hemiRings_8_cylRings_1_segments_16_halfLen_1) created and registered.
  | I0119 22:34:34.707351 24322 AssetAttributesManager.cpp:117] Asset attributes (coneSolid : coneSolid_segments_12_halfLen_1.25_rings_1_useTexCoords_false_useTangents_false_capEnd_true) created and registered.
  | I0119 22:34:34.707384 24322 AssetAttributesManager.cpp:117] Asset attributes (coneWireframe : coneWireframe_segments_32_halfLen_1.25) created and registered.
  | I0119 22:34:34.707402 24322 AssetAttributesManager.cpp:117] Asset attributes (cubeSolid : cubeSolid) created and registered.
  | I0119 22:34:34.707418 24322 AssetAttributesManager.cpp:117] Asset attributes (cubeWireframe : cubeWireframe) created and registered.
  | I0119 22:34:34.707458 24322 AssetAttributesManager.cpp:117] Asset attributes (cylinderSolid : cylinderSolid_rings_1_segments_12_halfLen_1_useTexCoords_false_useTangents_false_capEnds_true) created and registered.
  | I0119 22:34:34.707497 24322 AssetAttributesManager.cpp:117] Asset attributes (cylinderWireframe : cylinderWireframe_rings_1_segments_32_halfLen_1) created and registered.
  | I0119 22:34:34.707517 24322 AssetAttributesManager.cpp:117] Asset attributes (icosphereSolid : icosphereSolid_subdivs_1) created and registered.
  | I0119 22:34:34.707541 24322 AssetAttributesManager.cpp:117] Asset attributes (icosphereWireframe : icosphereWireframe_subdivs_1) created and registered.
  | I0119 22:34:34.707568 24322 AssetAttributesManager.cpp:117] Asset attributes (uvSphereSolid : uvSphereSolid_rings_8_segments_16_useTexCoords_false_useTangents_false) created and registered.
  | I0119 22:34:34.707593 24322 AssetAttributesManager.cpp:117] Asset attributes (uvSphereWireframe : uvSphereWireframe_rings_16_segments_32) created and registered.
  | I0119 22:34:34.707605 24322 AssetAttributesManager.cpp:105] AssetAttributesManager::buildCtorFuncPtrMaps : Built default primitive asset templates : 12
  | I0119 22:34:34.708432 24322 SceneDatasetAttributesManager.cpp:23] File (default) not found, so new default dataset attributes created and registered.
  | I0119 22:34:34.708438 24322 MetadataMediator.cpp:47] MetadataMediator::createDataset : Dataset default successfully created.
  | I0119 22:34:34.713798 24322 PhysicsAttributesManager.cpp:27] File (./data/default.physics_config.json) not found, so new default physics manager attributes created and registered.
  | I0119 22:34:34.713815 24322 ManagedContainerBase.cpp:19] ManagedContainerBase::convertFilenameToJSON : Filename : /root/habitat-lab/data/scene_datasets/mp3d/sT4fr6TAbpF/sT4fr6TAbpF.glb changed to proposed JSON configuration filename : /root/habitat-lab/data/scene_datasets/mp3d/sT4fr6TAbpF/sT4fr6TAbpF.stage_config.json
[... cut ...]
  | I0119 22:34:38.958553 24322 simulator.py:213] Loaded navmesh /root/habitat-lab/data/scene_datasets/mp3d/sT4fr6TAbpF/sT4fr6TAbpF.navmesh
  | I0119 22:34:38.958972 24322 simulator.py:224] Recomputing navmesh for agent's height 0.88 and radius 0.18.
  | I0119 22:34:38.973908 24322 PathFinder.cpp:382] Building navmesh with 367x615 cells
  | I0119 22:34:39.213542 24322 PathFinder.cpp:650] Created navmesh with 677 vertices 344 polygons
  | I0119 22:34:39.213613 24322 Simulator.cpp:710] reconstruct navmesh successful
  | 2021-01-19 22:34:39,224 Initializing task ObjectNav-v1
```

erikwijmans, 2021-01-19:
This is unfortunately a core limitation of EGL -- after a CUDA context has been initialized in a process, forking and then trying to get an EGL context does not work.  What does work is forking processes before initializing CUDA.  I am not familiar enough with ray to know if that is achievable or not.

One other method is to still use something like `habitat.VectorEnv` with just 1 worker as that will isolate the simulator into a separate processes, but that introduces additional overhead unfortunately.

smorad, 2021-01-25:
@erikwijmans Is this the issue you're referring to? https://doc.magnum.graphics/magnum/classMagnum_1_1Platform_1_1WindowlessEglApplication.html#Platform-WindowlessEglApplication-device-selection

Could you point me to where the EGL and CUDA contexts are initialized in habitat? Would this be in a call to `magnum` somewhere?

erikwijmans, 2021-01-25:
Habitat-sim does not initialize a CUDA context before the EGL context and only initializes a CUDA context when using the GPU accelerated noise models or GPU-GPU transfer of rendering results.  If habitat-sim is not built with cuda, it will never initialize one. 

Yeah, WindowlessEglApplication is what selects the EGL device, this block of code here: https://github.com/mosra/magnum/blob/46f4518eb6ca121dfa3da2884dc5c8ad5417e97b/src/Magnum/Platform/WindowlessEglApplication.cpp#L165-L212

[egl-test-case.zip](https://github.com/facebookresearch/habitat-sim/files/5868492/egl-test-case.zip). That is my current little EGL tester if it is helpful.  Any device with the extension `EGL_NV_device_cuda` should be reported as a cuda device (assuming CUDA_VISIBLE_DEVICES is not set)

smorad, 2021-01-26:
For any others that try this: I was able to get ray working with multiple habitat instances. As Erik says, the issue is CUDA initialization. I was doing the following:
```
class MySim(habitat.RLEnv):
  ...
  cfg.SIMULATOR.HABITAT_SIM_V0.GPU_DEVICE_ID = ray.get_gpu_ids()[0]
  super().__init__(self, cfg)
```
to ensure the ray worker and habitat were colocated on the same GPU. The `ray.get_gpu_ids` call initializes a CUDA context and breaks habitat. Ray automatically sets `CUDA_VISIBLE_DEVICES` to the GPU(s) allocated for a specific worker. This can be used for colocation without initializing a CUDA context.

closed 2021-01-26 by smorad

erikwijmans, 2021-01-27:
Actually, if ray is using CUDA visible devices, then just using GPU 0 for habitat should work.  Habitat listens to CUDA visible devices.