facebookresearch/habitat-sim #1041: Problems in producing the area for Gibson glb files

vincent341, 2021-01-21:
## ❓ Questions and Help
Hello. I would like to compute the total area and navigable area for Gibon scenes. I found the [Habitat-sim Basics for Navigation tutorial ](https://colab.research.google.com/github/facebookresearch/habitat-sim/blob/master/examples/tutorials/colabs/ECCV_2020_Navigation.ipynb) is the one I should follow. 

So I firstly installed habitat-sim by running "conda install habitat-sim headless -c conda-forge -c aihabitat-nightly" on my desktop.

Then I create a tiny version of the tutorial by copying some parts of the tutorial on my desktop as follows;
```

# @title Configure Sim Settings

test_scene = "./data/scene_datasets/mp3d/17DRP5sb8fy/17DRP5sb8fy.glb"
display=False

rgb_sensor = True  # @param {type:"boolean"}
depth_sensor = True  # @param {type:"boolean"}
semantic_sensor = True  # @param {type:"boolean"}

sim_settings = {
    "width": 256,  # Spatial resolution of the observations
    "height": 256,
    "scene": test_scene,  # Scene path
    "default_agent": 0,
    "sensor_height": 1.5,  # Height of sensors in meters
    "color_sensor": rgb_sensor,  # RGB sensor
    "depth_sensor": depth_sensor,  # Depth sensor
    "semantic_sensor": semantic_sensor,  # Semantic sensor
    "seed": 1,  # used in the random navigation
    "enable_physics": False,  # kinematics only
}

def make_cfg(settings):
    sim_cfg = habitat_sim.SimulatorConfiguration()
    sim_cfg.gpu_device_id = 0
    sim_cfg.scene_id = settings["scene"]
    sim_cfg.enable_physics = settings["enable_physics"]

    # Note: all sensors must have the same resolution
    sensors = {
        "color_sensor": {
            "sensor_type": habitat_sim.SensorType.COLOR,
            "resolution": [settings["height"], settings["width"]],
            "position": [0.0, settings["sensor_height"], 0.0],
        },
        "depth_sensor": {
            "sensor_type": habitat_sim.SensorType.DEPTH,
            "resolution": [settings["height"], settings["width"]],
            "position": [0.0, settings["sensor_height"], 0.0],
        },
        "semantic_sensor": {
            "sensor_type": habitat_sim.SensorType.SEMANTIC,
            "resolution": [settings["height"], settings["width"]],
            "position": [0.0, settings["sensor_height"], 0.0],
        },
    }

    sensor_specs = []
    for sensor_uuid, sensor_params in sensors.items():
        if settings[sensor_uuid]:
            sensor_spec = habitat_sim.SensorSpec()
            sensor_spec.uuid = sensor_uuid
            sensor_spec.sensor_type = sensor_params["sensor_type"]
            sensor_spec.resolution = sensor_params["resolution"]
            sensor_spec.position = sensor_params["position"]

            sensor_specs.append(sensor_spec)

    # Here you can specify the amount of displacement in a forward action and the turn angle
    agent_cfg = habitat_sim.agent.AgentConfiguration()
    agent_cfg.sensor_specifications = sensor_specs
    agent_cfg.action_space = {
        "move_forward": habitat_sim.agent.ActionSpec(
            "move_forward", habitat_sim.agent.ActuationSpec(amount=0.25)
        ),
        "turn_left": habitat_sim.agent.ActionSpec(
            "turn_left", habitat_sim.agent.ActuationSpec(amount=30.0)
        ),
        "turn_right": habitat_sim.agent.ActionSpec(
            "turn_right", habitat_sim.agent.ActuationSpec(amount=30.0)
        ),
    }

    return habitat_sim.Configuration(sim_cfg, [agent_cfg])

def print_scene_recur(scene, limit_output=10):
    print(dir(scene))
    print(
        f"House has {len(scene.levels)} levels, {len(scene.regions)} regions and {len(scene.objects)} objects"
    )
    print(f"House center:{scene.aabb.center} dims:{scene.aabb.sizes}")

    count = 0
    for level in scene.levels:
        print(
            f"Level id:{level.id}, center:{level.aabb.center},"
            f" dims:{level.aabb.sizes}"
        )
        for region in level.regions:
            print(
                f"Region id:{region.id}, category:{region.category.name()},"
                f" center:{region.aabb.center}, dims:{region.aabb.sizes}"
            )
            for obj in region.objects:
                print(
                    f"Object id:{obj.id}, category:{obj.category.name()},"
                    f" center:{obj.aabb.center}, dims:{obj.aabb.sizes}"
                )
                count += 1
                if count >= limit_output:
                    return

# Print semantic annotation information (id, category, bounding box details)
# about levels, regions and objects in a hierarchical fashion

cfg = make_cfg(sim_settings)
# Needed to handle out of order cell run in Colab
try:  # Got to make initialization idiot proof
    sim.close()
except NameError:
    pass
sim = habitat_sim.Simulator(cfg)

scene = sim.semantic_scene
print_scene_recur(scene)
```
The code produced corrected result for the test scene. 

But when I replace the test_scene by a scene "Adrian.glb" in Gibson dataset. The outputs  are quite strange, as follows

> House has 0 levels, 0 regions and 0 objects
> House center:[0. 0. 0.] dims:[-inf -inf -inf]

Could anyone give any hint on this problem? Any suggestion would be highly appreciated.

By the way, what's the difference between the result returned by region.aabb.sizes and "sim.pathfinder.get_bounds()" of NavMesh?

erikwijmans, 2021-01-21:
`sim.semantic_scene` access whatever semantic annotations a dataset has.  Only a very small subset of Gibson has semantic annotations (and I forget if that has region annotations or not). `sim.pathfinder.get_bounds()` returns the AABB of the navigable space as computed by the navmesh.

vincent341, 2021-01-22:
Hi @erikwijmans ,

Thanks so much for your help. In fact, I would like to compute the area for environments. I tried two ways to compute the area. 
The first way is by using sim.pathfinder.get_bounds(). I assume _sim.pathfinder.get_bounds()_ returns _[xa,ya,za]_ and _[xb,yb,zb]_, which are the coordinate of diagonal points. I compute the area, denoted by bound_area, by _bound_area = |xb-xa|*|zb-za|_. I assume _x,y,z_ denotes length, height and width respectively (right?).  As far as I understand, bound_area is the total area of an environment.

The second way I tried is by calling _sim.pathfinder.navigable_area_. I believe _sim.pathfinder.navigable_area_ returns the area which is traversable, excluding some objects, such as furnitures.  The area it returns is named _nav_area_. 

If my above understanding is correct, _nav_area_<= _bound_area_. 

I computed _bound_area_ and _nav_area_ for the scene in Gibson v1. The distribution is shown below.
[image]

The figure shows that in some cases, _nav_area_> _bound_area_, which is against my understanding.  In addition, the [gibson website](http://gibsonenv.stanford.edu/models/?id=Adrian) gives an official total area for each scene. My computed _bound_area_ is different from the officially given area by a large margin. 

Would you mind giving me any hint on the aforementioned two inconsistence? Thanks.

vincent341, 2021-01-22:
By the way, it seems that there is a ".json.gz" and a ".glb" file for each scene. What data do they contain respectively?

erikwijmans, 2021-01-22:
> nav_area> bound_area

That is fine.  Your bounded area is approximating the navigable surface as a flat plane.  When the navigable surface is not a flat plane (i.e. a multi-story house), having nav_area > bound_area is expected.

>  In addition, the gibson website gives an official total area for each scene. My computed bound_area is different from the officially given area by a large margin.

I do not know how those numbers were computed so I can't comment on this.

vincent341, 2021-01-24:
> > nav_area> bound_area
> 
> That is fine. Your bounded area is approximating the navigable surface as a flat plane. When the navigable surface is not a flat plane (i.e. a multi-story house), having nav_area > bound_area is expected.
> 
> > In addition, the gibson website gives an official total area for each scene. My computed bound_area is different from the officially given area by a large margin.
> 
> I do not know how Gibson computed those numbers were computed so I can't comment on this.

Thans so much for your quick answer.
After some learning, it seems that there are 3 ways to compute the area. Beside _bound_area_ and _nav_area_,  another way to compute the area is by calling _top_down_map_ = _sim.pathfinder.get_topdown_map()_ . _top_down_map_ corresponds the top down map of the current story. The area can be computed by _top_down_area = width(_top_down_map_)*length(_top_down_map)*meter_per_pixel_.  The relationship among _bound_area_ , _nav_area_ and _top_down_area is

If _story=1_, _bound_area=top_down_area>=nav_area_
If _stroy>1_, _bound_area=top_down_area_. _nav_area_ may be larger or less than _bound_area_.

Is my understanding corrected?

erikwijmans, 2021-01-24:
bound_area=top_down_area (modulo rounding errors) has to be true as the size of the top_down_map image is determined by the pathfinder bounds.  

The relationship between nav_area and top_down_free_area is hard to say anything concert about.  top_down_free_area makes approximates that could both increase area, i.e. a cell being marked as navigable even when < 50% of it is actually navigable, or decrease area, i.e. the environment have slight level changes even when being one floor or a cell being marked as not navigable even when most of it.

To be clear, the only of these that is the actual navigable area is nav_area.  top_down_free_area is a reasonable approximation of the navigable area for a given height Y.  Neither bound_area nor top_down_area are good approximates of anything.

vincent341, 2021-01-27:
Hi @erikwijmans , Thanks very much for your explanation.  It helps me quite a lot. 

One more question is the Gibson dataset used in habitat lab (named _Gibson-h_ here) contains about 88 environments, total 1.5GB as shown in the [link](https://github.com/facebookresearch/habitat-lab#data).  But the official Gibson contains more than 500 environments, may up to 89 GB, as given in the [website](https://github.com/StanfordVL/GibsonEnv/tree/master/gibson/data).  It seems that _Gibson-h_ doesn't match any split of the official Gibson dataset given at the [website](https://github.com/StanfordVL/GibsonEnv/tree/master/gibson/data). 

May I know the relationship between Gibson-h and the official Gibson? Is it possible to use the full set of the official Gibson dataset in habitat simulator without any additional settings?

Similar question for the MP3D dataset.

erikwijmans, 2021-01-27:
For Gibson, we curated it as most of the reconstructions are awful.  https://arxiv.org/abs/1904.01201. You can use full Gibson, although you will need to generate pointnav episodes for those missing scenes if that is your intended use.

closed 2021-08-16 by dhruvbatra