facebookresearch/habitat-sim #1046: Problem with using progmesh utility for replica dataset

Ainaz99, 2021-01-25:
## ❓ Questions and Help
Hi,
I am struggling with this problem for some time now, and I would really appreciate it if you can help me fix it. 
Most of the original replica scenes crash in the middle of the simulation when using bullet physics. So I'm trying to use the decimated version of a replica mesh (produced from the progmesh utility) as the collision mesh of the scene, but it seems to fail when creating the simulator: `sim = habitat_sim.Simulator(cfg)`.
This is how the decimated version of the replica mesh is created (mesh2.ply):
```
./progmesh   /scratch-data/replica/hotel_0/habitat/mesh_semantic.ply  /scratch-data/replica/hotel_0/mesh2.ply  500000
1097438
2193298
2193298
loaded
Start computing costs
Done computing costs
.............................................................................................................
```
Here's the config file I use for the stage with the original replica asset (mesh.ply) as the `render mesh` and the decimated version (mesh2.ply) as the `collision mesh`:
```
{
        "render mesh": "mesh.ply",
        "collision mesh": "mesh2.ply",
        "up":[0,1,0],
        "front":[0,0,-1],
        "origin":[0,0,0],
        "gravity":[0,-9.8,0],
        "margin":0.03,
        "friction_coefficient": 0.3,
        "restitution_coefficient": 0.3,
        "units_to_meters":1.0
}
```

And here's the full error (which seems to fail when loading the `collision mesh`) :
```
==== Initialized Sensor Spec: =====
Sensor uuid:  color_sensor
Sensor type:  SensorType.COLOR
Sensor position:  [0. 0. 0.]
Sensor resolution:  [512 512]
===================================
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0113 18:37:39.387195 113127 AssetAttributesManager.cpp:117] Asset attributes (capsule3DSolid : capsule3DSolid_hemiRings_4_cylRings_1_segments_12_halfLen_0.75_useTexCoords_false_useTangents_false) created and registered.
I0113 18:37:39.387508 113127 AssetAttributesManager.cpp:117] Asset attributes (capsule3DWireframe : capsule3DWireframe_hemiRings_8_cylRings_1_segments_16_halfLen_1) created and registered.
I0113 18:37:39.387706 113127 AssetAttributesManager.cpp:117] Asset attributes (coneSolid : coneSolid_segments_12_halfLen_1.25_rings_1_useTexCoords_false_useTangents_false_capEnd_true) created and registered.
I0113 18:37:39.387872 113127 AssetAttributesManager.cpp:117] Asset attributes (coneWireframe : coneWireframe_segments_32_halfLen_1.25) created and registered.
I0113 18:37:39.387984 113127 AssetAttributesManager.cpp:117] Asset attributes (cubeSolid : cubeSolid) created and registered.
I0113 18:37:39.388303 113127 AssetAttributesManager.cpp:117] Asset attributes (cubeWireframe : cubeWireframe) created and registered.
I0113 18:37:39.388518 113127 AssetAttributesManager.cpp:117] Asset attributes (cylinderSolid : cylinderSolid_rings_1_segments_12_halfLen_1_useTexCoords_false_useTangents_false_capEnds_true) created and registered.
I0113 18:37:39.388767 113127 AssetAttributesManager.cpp:117] Asset attributes (cylinderWireframe : cylinderWireframe_rings_1_segments_32_halfLen_1) created and registered.
I0113 18:37:39.388862 113127 AssetAttributesManager.cpp:117] Asset attributes (icosphereSolid : icosphereSolid_subdivs_1) created and registered.
I0113 18:37:39.389199 113127 AssetAttributesManager.cpp:117] Asset attributes (icosphereWireframe : icosphereWireframe_subdivs_1) created and registered.
I0113 18:37:39.389396 113127 AssetAttributesManager.cpp:117] Asset attributes (uvSphereSolid : uvSphereSolid_rings_8_segments_16_useTexCoords_false_useTangents_false) created and registered.
I0113 18:37:39.389482 113127 AssetAttributesManager.cpp:117] Asset attributes (uvSphereWireframe : uvSphereWireframe_rings_16_segments_32) created and registered.
I0113 18:37:39.389569 113127 AssetAttributesManager.cpp:105] AssetAttributesManager::buildCtorFuncPtrMaps : Built default primitive asset templates : 12
I0113 18:37:39.393952 113127 PhysicsAttributesManager.cpp:27] JSON File (/scratch-data/ainaz/habitat-sim/data/default.phys_scene_config.json) Based physics manager attributes created and registered.
I0113 18:37:39.399041 113127 AbstractObjectAttributesManagerBase.h:179] JSON File (/scratch-data/ainaz/replica-google-objects/hotel_0/mesh.stage_config.json) Based Stage attributes created and registered.
W0113 18:37:39.399472 113127 Simulator.cpp:148] Navmesh file not found, checked at 
I0113 18:37:39.399785 113127 SceneGraph.h:93] Created DrawableGroup: 
Renderer: Tesla V100-SXM2-32GB/PCIe/SSE2 by NVIDIA Corporation
OpenGL version: 4.6.0 NVIDIA 440.64
Using optional features:
    GL_ARB_ES2_compatibility
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_invalidate_subdata
    GL_ARB_multi_bind
    GL_ARB_robustness
    GL_ARB_separate_shader_objects
    GL_ARB_texture_filter_anisotropic
    GL_ARB_texture_storage
    GL_ARB_texture_storage_multisample
    GL_ARB_vertex_array_object
    GL_KHR_debug
Using driver workarounds:
    no-forward-compatible-core-context
    nv-egl-incorrect-gl11-function-pointers
    no-layout-qualifiers-on-old-glsl
    nv-zero-context-profile-mask
    nv-implementation-color-read-format-dsa-broken
    nv-cubemap-inconsistent-compressed-image-size
    nv-cubemap-broken-full-compressed-image-query
    nv-compressed-block-size-in-bits
creating staticStageObject_
creating staticStageObject_ .. done
I0113 18:37:39.597506 113127 PTexMeshData.cpp:476] Splitting mesh... 
I0113 18:37:40.356065 113127 PTexMeshData.cpp:383] The number of quads: 1096649, which equals to 2193298 triangles.
I0113 18:37:40.360316 113127 PTexMeshData.cpp:832] Loading mesh 1/6... 
I0113 18:37:40.455554 113127 PTexMeshData.cpp:832] Loading mesh 2/6... 
I0113 18:37:40.475643 113127 PTexMeshData.cpp:832] Loading mesh 3/6... 
I0113 18:37:40.480320 113127 PTexMeshData.cpp:832] Loading mesh 4/6... 
I0113 18:37:40.487354 113127 PTexMeshData.cpp:832] Loading mesh 5/6... 
I0113 18:37:40.558043 113127 PTexMeshData.cpp:832] Loading mesh 6/6... 
I0113 18:37:40.562263 113127 PTexMeshData.cpp:858] Calculating mesh adjacency... 
I0113 18:37:42.207960 113127 PTexMeshData.cpp:905] loading atlas textures: 
I0113 18:37:42.209473 113127 PTexMeshData.cpp:914] Loading atlas 1/6 from /scratch-data/ainaz/replica-google-objects/hotel_0/textures/0-color-ptex.hdr. 
I0113 18:37:42.660079 113127 PTexMeshData.cpp:914] Loading atlas 2/6 from /scratch-data/ainaz/replica-google-objects/hotel_0/textures/1-color-ptex.hdr. 
I0113 18:37:43.177911 113127 PTexMeshData.cpp:914] Loading atlas 3/6 from /scratch-data/ainaz/replica-google-objects/hotel_0/textures/2-color-ptex.hdr. 
I0113 18:37:43.594976 113127 PTexMeshData.cpp:914] Loading atlas 4/6 from /scratch-data/ainaz/replica-google-objects/hotel_0/textures/3-color-ptex.hdr. 
I0113 18:37:43.857050 113127 PTexMeshData.cpp:914] Loading atlas 5/6 from /scratch-data/ainaz/replica-google-objects/hotel_0/textures/4-color-ptex.hdr. 
I0113 18:37:44.384876 113127 PTexMeshData.cpp:914] Loading atlas 6/6 from /scratch-data/ainaz/replica-google-objects/hotel_0/textures/5-color-ptex.hdr. 
I0113 18:37:44.755409 113127 ResourceManager.cpp:962] Importing Basis files as BC7
Trade::AbstractImporter::object3D(): index 0 out of range for 0 entries
jobs/create_rgb_images_replica_google_objects_job.sh: line 22: 113127 Aborted                 (core dumped)
```
Thanks in advance for your help.

aclegg3, 2021-01-26:
Hey @Ainaz99,

Sorry to hear you are experiencing this issue, I know it must be frustrating. Let's try for a functional solution. Because this issue could be related to a number of systems, it would be easiest for me to reproduce and debug if you can provide a download link to the decimated mesh output "mesh2.ply" which you are trying to load.

I do know that the system attempts to load assets differently based on their format, and the .ply importer is not the most well supported outside of the ptex Replica render asset use case. It could be that you have stumbled on an untested edge case. You may have better luck if you convert the collision asset to .obj or .glb (e.g., import and re-export in Blender). I can try this on my end also once I get your collision asset.

Ainaz99, 2021-01-27:
Hey @aclegg3,

Converting the collision mesh to .glb totally worked! :) Thanks a lot for your help.

aclegg3, 2021-01-27:
Happy to hear it.

closed 2021-01-27 by aclegg3