facebookresearch/habitat-sim #1047: problem in function PoseExtractor

wwtx9, 2021-01-25:
## ❓ Questions and Help
When I run sample example

```
import numpy as np
import habitat_sim.registry as registry

from habitat_sim.utils.data import ImageExtractor, PoseExtractor

@registry.register_pose_extractor(name="random_pose_extractor")
class RandomPoseExtractor(PoseExtractor):
    def extract_poses(self, view, fp):
        height, width = view.shape
        num_random_points = 4
        points = []
        while len(points) < num_random_points:
            # Get the row and column of a random point on the topdown view
            row, col = np.random.randint(0, height), np.random.randint(0, width)

            # Convenient method in the PoseExtractor class to check if a point
            # is navigable
            if self._valid_point(row, col, view):
                points.append((row, col))

        poses = []

        # Now we need to define a "point of interest" which is the point the camera will
        # look at. These two points together define a camera position and angle
        for point in points:
            r, c = point
            point_of_interest = (r - 1, c) # Just look forward
            pose = (point, point_of_interest, fp)
            poses.append(pose)

        return poses

scene_filepath = "data/scene_datasets/habitat-test-scenes/skokloster-castle.glb"
extractor = ImageExtractor(
    scene_filepath,
    pose_extractor_name="random_pose_extractor"
)
```
it shows the errors belows:
```
I0125 16:17:27.055393 7667 simulator.py:181] Loaded navmesh data/scene_datasets/habitat-test-scenes/skokloster-castle.navmesh
Traceback (most recent call last):
  File "/home/xxx/PycharmProjects/pythonProject/control.py", line 197, in <module>
    pose_extractor_name="random_pose_extractor"
  File "/home/xxx/anaconda3/envs/habitat/lib/python3.6/site-packages/habitat_sim-0.1.6-py3.6-linux-x86_64.egg/habitat_sim/utils/data/data_extractor.py", line 117, in __init__
    self.poses = self.pose_extractor.extract_all_poses(labels=self.labels)
  File "/home/xxx/anaconda3/envs/habitat/lib/python3.6/site-packages/habitat_sim-0.1.6-py3.6-linux-x86_64.egg/habitat_sim/utils/data/pose_extractor.py", line 39, in extract_all_poses
    self.extract_poses(labels, view, fp), ref_point
TypeError: extract_poses() takes 3 positional arguments but 4 were given
I0125 16:17:27.141350  7667 PhysicsManager.cpp:33] Deconstructing PhysicsManager
I0125 16:17:27.141366  7667 SemanticScene.h:41] Deconstructing SemanticScene
I0125 16:17:27.141368  7667 SceneManager.h:25] Deconstructing SceneManager
I0125 16:17:27.141369  7667 SceneGraph.h:26] Deconstructing SceneGraph
I0125 16:17:27.141400  7667 Sensor.h:81] Deconstructing Sensor
I0125 16:17:27.141410  7667 Sensor.h:81] Deconstructing Sensor
I0125 16:17:27.141418  7667 Sensor.h:81] Deconstructing Sensor
I0125 16:17:27.143587  7667 Renderer.cpp:34] Deconstructing Renderer
I0125 16:17:27.143596  7667 WindowlessContext.h:17] Deconstructing WindowlessContext
I0125 16:17:27.152432  7667 Simulator.cpp:46] Deconstructing Simulator

Process finished with exit code 1

```

erikwijmans, 2021-01-26:
This seems to be fixed on master, you can pull nightly binaries from conda via `-c aihabitat-nightly`

wwtx9, 2021-01-26:
@erikwijmans Thank you very much. It is helpful.

closed 2021-01-26 by wwtx9