facebookresearch/habitat-sim #1050: a few questions about habitat-sim api

wwtx9, 2021-01-26:
## ❓ Questions and Help
I have a few questions about habitat-sim that I want to clear up.
1. Coordinate
How does habitat-sim API define the world coordinate? I mean where is the world coordinate.

Is agent_state.position return (x,y,z) in world coordinate system?
Is agent_state.rotation return (QW,QX,QY,QZ) is rotation from camera coordinate to world coordinate?
Is rotation of agent_state.sensor_states also from the camera coordinate to world coordinate? 

2. Intrinsic matrix
As we know Z * [u, v, 1]T = K * [X, Y, Z]T, where X, Y, Z are represented in camera coordinate. So I think the value of the instrinsic matrix should be the following: 
```
K = [[fx, 0, cx],
         [0, fy, cy],
         [0, 0, 1]]
fx = width of image/(2* tan(hfov/2))
fy = height of image/(2* tan(vfov/2))
cx = width of image/2
cy = height of image/2
```
This is different with K = np.array([
[1 / np.tan(hfov / 2.), 0., 0., 0.],
[0., 1 / np.tan(hfov / 2.), 0., 0.],
[0., 0., 1, 0],
[0., 0., 0, 1]]) [https://github.com/facebookresearch/habitat-sim/issues/80](url)

3. Depth image extracted by the depth camera.
```
In the examples/tutorials/stereo_agent.py Why It took values less than 10 from the depth image and then divided by 10 to normalize?

What is the value in depth image meaning, is the depth value of each pixel in the camera coordinate? So if I want to know a depth of a pixel, I just read this pixel's value in depth image directly?
```
These things confuse me when I coding.

erikwijmans, 2021-01-27:
1.  All of those things are in world coordinates.   There isn't anything that we expose that isn't in the global coordinate frame AFAIK. Note that (0, 0, 0) doesn't have a physical meaning tho (it isn't like the origin is always at the door to the house or something).
2. It's a matter of perspective.  Graphics and modern CV (deep learning) commonly uses u, v in [-1, 1]^2  instead of [0, width) x [0, height) as it makes it easier to write resolution agnostic code. 
3. Depth is in meters (assuming the units of the dataset is meters).  That division by 10 is just a hack to get things into [0, 255] for visualization.

wwtx9, 2021-01-27:
@erikwijmans I really appreciate your reply. It is very helpful. Thanks again!!

closed 2021-01-27 by wwtx9

dhruvbatra, 2021-01-27:
@wwtx9 -- glad this was resolved. If you have the time, consider contributing a PR with this information embedded at places where you expected to find them (new users typically have the freshest set of eyes and can identify where documentation is lacking).