facebookresearch/habitat-sim #1053: Simulator remove_object method isn't deallocating the physObjectId

NaveenVSrinivasan, 2021-01-27:
## 🐛 Bug

When I call remove_object on a given object_id, the object is removed from the scene but the id is not appropriately deallocated and scrubbed. Thus, in downstream methods the simulator stops at the assertIDValidity method in PhysicsManager.h because it catches that the count of an object_id in existingObjects_ is 0.

## System Info

ENVIRONMENT INFO:
Platform: Linux-5.4.0-1025-aws-x86_64-with-debian-buster-sid
Machine: x86_64
Processor: x86_64
Libc version: glibc 2.9
Mac version:
Python version: 3.6.12
Architecture: 64bit
Win version:
System OS: Linux
Release: 5.4.0-1025-aws
Version: #25~18.04.1-Ubuntu SMP Fri Sep 11 12:03:04 UTC 2020
Operational System: linux
GCC version: b'gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0\n'
CMAKE version: b'cmake version 3.19.3\n'
NVIDIA-SMI: b'Wed Jan 27 16:42:13 2021       \n+-----------------------------------------------------------------------------+\n| NVIDIA-SMI 450.51.05    Driver Version: 450.51.05    CUDA Version: 11.0     |\n|-------------------------------+----------------------+----------------------+\n| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |\n| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |\n|                               |                      |               MIG M. |\n|===============================+======================+======================|\n|   0  Tesla K80           On   | 00000000:00:17.0 Off |                    0 |\n| N/A   28C    P0    53W / 149W |   3545MiB / 11441MiB |      0%      Default |\n|                               |                      |                  N/A |\n+-------------------------------+----------------------+----------------------+\n|   1  Tesla K80           On   | 00000000:00:18.0 Off |                    0 |\n| N/A   20C    P8    29W / 149W |      3MiB / 11441MiB |      0%      Default |\n|                               |                      |                  N/A |\n+-------------------------------+----------------------+----------------------+\n|   2  Tesla K80           On   | 00000000:00:19.0 Off |                    0 |\n| N/A   23C    P8    26W / 149W |      3MiB / 11441MiB |      0%      Default |\n|                               |                      |                  N/A |\n+-------------------------------+----------------------+----------------------+\n|   3  Tesla K80           On   | 00000000:00:1A.0 Off |                    0 |\n| N/A   21C    P8    29W / 149W |      3MiB / 11441MiB |      0%      Default |\n|                               |                      |                  N/A |\n+-------------------------------+----------------------+----------------------+\n|   4  Tesla K80           On   | 00000000:00:1B.0 Off |                    0 |\n| N/A   23C    P8    27W / 149W |      3MiB / 11441MiB |      0%      Default |\n|                               |                      |                  N/A |\n+-------------------------------+----------------------+----------------------+\n|   5  Tesla K80           On   | 00000000:00:1C.0 Off |                    0 |\n| N/A   20C    P8    29W / 149W |      3MiB / 11441MiB |      0%      Default |\n|                               |                      |                  N/A |\n+-------------------------------+----------------------+----------------------+\n|   6  Tesla K80           On   | 00000000:00:1D.0 Off |                    0 |\n| N/A   24C    P8    26W / 149W |      3MiB / 11441MiB |      0%      Default |\n|                               |                      |                  N/A |\n+-------------------------------+----------------------+----------------------+\n|   7  Tesla K80           On   | 00000000:00:1E.0 Off |                    0 |\n| N/A   22C    P8    30W / 149W |      3MiB / 11441MiB |      0%      Default |\n|                               |                      |                  N/A |\n+-------------------------------+----------------------+----------------------+\n'

Conda packages versions:
(kondo) srinivasann@ip-172-31-59-134:~/projects/habitat-sim/habitat_sim/utils$ conda list
# packages in environment at /home/srinivasann/miniconda3/envs/kondo:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       1_gnu    conda-forge
absl-py                   0.11.0                   pypi_0    pypi
astor                     0.8.1                    pypi_0    pypi
attrs                     20.3.0             pyhd3deb0d_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.1                      py_0    conda-forge
bullet                    2.88            py36hb3f55d8_1001    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.17.1               h36c2ea0_0    conda-forge
ca-certificates           2020.12.5            ha878542_0    conda-forge
cached-property           1.5.2                    pypi_0    pypi
cachetools                4.2.0                    pypi_0    pypi
certifi                   2020.12.5        py36h5fab9bb_1    conda-forge
chardet                   4.0.0                    pypi_0    pypi
cmake                     3.19.3               h4547794_0    conda-forge
cycler                    0.10.0                     py_2    conda-forge
dataclasses               0.8                      pypi_0    pypi
decorator                 4.4.2                    pypi_0    pypi
expat                     2.2.9                he1b5a44_2    conda-forge
ffmpeg                    4.3.1                hca11adc_2    conda-forge
freetype                  2.10.4               h0708190_1    conda-forge
gast                      0.4.0                    pypi_0    pypi
geos                      3.8.1                he1b5a44_0    conda-forge
gitdb                     4.0.5                      py_0    conda-forge
gitpython                 3.1.12             pyhd8ed1ab_0    conda-forge
gmp                       6.2.1                h58526e2_0    conda-forge
gnutls                    3.6.13               h85f3911_1    conda-forge
google-auth               1.24.0                   pypi_0    pypi
google-auth-oauthlib      0.4.2                    pypi_0    pypi
grpcio                    1.34.0                   pypi_0    pypi
gym                       0.10.9                   pypi_0    pypi
h5py                      3.1.0                    pypi_0    pypi
habitat                   0.1.6                     dev_0    <develop>
habitat-sim               0.1.6                    pypi_0    pypi
headless                  1.0                           0    aihabitat
icu                       67.1                 he1b5a44_0    conda-forge
idna                      2.10                     pypi_0    pypi
ifcfg                     0.21                     pypi_0    pypi
imageio                   2.9.0                      py_0    conda-forge
imageio-ffmpeg            0.4.3              pyhd8ed1ab_0    conda-forge
[... 103 lines cut ...]
withbullet                1.0                           0    aihabitat
x264                      1!161.3030           h7f98852_0    conda-forge
xorg-fixesproto           5.0               h14c3975_1002    conda-forge
xorg-inputproto           2.3.2             h7f98852_1002    conda-forge
xorg-kbproto              1.0.7             h7f98852_1002    conda-forge
xorg-libx11               1.6.12               h516909a_0    conda-forge
xorg-libxau               1.0.9                h7f98852_0    conda-forge
xorg-libxcursor           1.2.0                h516909a_0    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xorg-libxext              1.3.4                h516909a_0    conda-forge
xorg-libxfixes            5.0.3             h516909a_1004    conda-forge
xorg-libxi                1.7.10               h516909a_0    conda-forge
xorg-libxinerama          1.1.4             hf484d3e_1000    conda-forge
xorg-libxrandr            1.5.2                h516909a_1    conda-forge
xorg-libxrender           0.9.10            h516909a_1002    conda-forge
xorg-randrproto           1.5.0             h516909a_1001    conda-forge
xorg-renderproto          0.11.1            h14c3975_1002    conda-forge
[... cut ...] rge
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xz                        5.2.5                h516909a_1    conda-forge
yacs                      0.1.8                    pypi_0    pypi
zeromq                    4.3.3                h58526e2_3    conda-forge
zipp                      3.4.0                    pypi_0    pypi
zlib                      1.2.11            h516909a_1010    conda-forge
zstd                      1.4.8                ha95c52a_1    conda-forge

jturner65, 2021-01-27:
Hi

Can you share a minimal python script that demonstrates the issue you are having?

NaveenVSrinivasan, 2021-01-28:
Resolved! The issue was on my end. The error was caused when an objectId was erroneously cached after an object was removed and was passed to a sim call of getInitializationTemplate().

closed 2021-01-28 by NaveenVSrinivasan