facebookresearch/habitat-sim #1058: I found a contradiction in habitat coordinate

wwtx9, 2021-01-28:
## ❓ Questions and Help
I found a contradiction in habitat coordinate.

When set the left camera's rotation = quaternion(1, 0, 0,0 ) which means left camera's coordinate is same as world coordinate.
Therefore positive z axis is also the direction in which the camera faces forward.

As the agent moves forward, the value of z in sensor's position decreases. When the agent moves forward,  the value of z in sensor's position should increase. This is pretty werid.
I have highlight the value of z of sensor position in each action.
```
No.0 no action
current agent's sensor state: {'left_sensor': SixDOFPose(position=array([-1.8926959,  1.6108389, **19.255245** ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'right_sensor': SixDOFPose(position=array([-1.6926959,  1.6108389, 19.255245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'left_sensor_depth': SixDOFPose(position=array([-1.8926959,  1.6108389, 19.255245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'right_sensor_depth': SixDOFPose(position=array([-1.6926959,  1.6108389, 19.255245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0))} 
No. 0 is finished!
last sensor state: {'left_sensor': SixDOFPose(position=array([-1.8926959,  1.6108389, 19.255245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'right_sensor': SixDOFPose(position=array([-1.6926959,  1.6108389, 19.255245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'left_sensor_depth': SixDOFPose(position=array([-1.8926959,  1.6108389, 19.255245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'right_sensor_depth': SixDOFPose(position=array([-1.6926959,  1.6108389, 19.255245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0))} 
No.1 to select action
action: FORWARD
current agent's sensor state: {'left_sensor': SixDOFPose(position=array([-1.8926959,  1.6196585, **19.005245** ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'right_sensor': SixDOFPose(position=array([-1.6926959,  1.6196585, 19.005245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'left_sensor_depth': SixDOFPose(position=array([-1.8926959,  1.6196585, 19.005245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'right_sensor_depth': SixDOFPose(position=array([-1.6926959,  1.6196585, 19.005245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0))} 
No. 1 is finished!
last sensor state: {'left_sensor': SixDOFPose(position=array([-1.8926959,  1.6108389, 19.255245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'right_sensor': SixDOFPose(position=array([-1.6926959,  1.6108389, 19.255245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'left_sensor_depth': SixDOFPose(position=array([-1.8926959,  1.6108389, 19.255245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'right_sensor_depth': SixDOFPose(position=array([-1.6926959,  1.6108389, 19.255245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0))} 
No.2 to select action
action: FORWARD
current agent's sensor state: {'left_sensor': SixDOFPose(position=array([-1.8926959,  1.628478 , **18.755245** ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'right_sensor': SixDOFPose(position=array([-1.6926959,  1.628478 , 18.755245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'left_sensor_depth': SixDOFPose(position=array([-1.8926959,  1.628478 , 18.755245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'right_sensor_depth': SixDOFPose(position=array([-1.6926959,  1.628478 , 18.755245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0))} 
No. 2 is finished!
last sensor state: {'left_sensor': SixDOFPose(position=array([-1.8926959,  1.6196585, 19.005245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'right_sensor': SixDOFPose(position=array([-1.6926959,  1.6196585, 19.005245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'left_sensor_depth': SixDOFPose(position=array([-1.8926959,  1.6196585, 19.005245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'right_sensor_depth': SixDOFPose(position=array([-1.6926959,  1.6196585, 19.005245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0))} 
No.3 to select action
action: FORWARD
current agent's sensor state: {'left_sensor': SixDOFPose(position=array([-1.8926959,  1.6372976, **18.505245** ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'right_sensor': SixDOFPose(position=array([-1.6926959,  1.6372976, 18.505245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'left_sensor_depth': SixDOFPose(position=array([-1.8926959,  1.6372976, 18.505245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'right_sensor_depth': SixDOFPose(position=array([-1.6926959,  1.6372976, 18.505245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0))} 
No. 3 is finished!
last sensor state: {'left_sensor': SixDOFPose(position=array([-1.8926959,  1.628478 , 18.755245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'right_sensor': SixDOFPose(position=array([-1.6926959,  1.628478 , 18.755245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'left_sensor_depth': SixDOFPose(position=array([-1.8926959,  1.628478 , 18.755245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'right_sensor_depth': SixDOFPose(position=array([-1.6926959,  1.628478 , 18.755245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0))} 
No.4 to select action
action: FORWARD
current agent's sensor state: {'left_sensor': SixDOFPose(position=array([-1.8926959,  1.6461172, **18.255245** ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'right_sensor': SixDOFPose(position=array([-1.6926959,  1.6461172, 18.255245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'left_sensor_depth': SixDOFPose(position=array([-1.8926959,  1.6461172, 18.255245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'right_sensor_depth': SixDOFPose(position=array([-1.6926959,  1.6461172, 18.255245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0))} 
No. 4 is finished!
last sensor state: {'left_sensor': SixDOFPose(position=array([-1.8926959,  1.6372976, 18.505245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'right_sensor': SixDOFPose(position=array([-1.6926959,  1.6372976, 18.505245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'left_sensor_depth': SixDOFPose(position=array([-1.8926959,  1.6372976, 18.505245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'right_sensor_depth': SixDOFPose(position=array([-1.6926959,  1.6372976, 18.505245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0))} 
No.5 to select action
action: FORWARD
current agent's sensor state: {'left_sensor': SixDOFPose(position=array([-1.8926959,  1.6549368, **18.005245** ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'right_sensor': SixDOFPose(position=array([-1.6926959,  1.6549368, 18.005245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'left_sensor_depth': SixDOFPose(position=array([-1.8926959,  1.6549368, 18.005245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'right_sensor_depth': SixDOFPose(position=array([-1.6926959,  1.6549368, 18.005245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0))} 
No. 5 is finished!
last sensor state: {'left_sensor': SixDOFPose(position=array([-1.8926959,  1.6461172, 18.255245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'right_sensor': SixDOFPose(position=array([-1.6926959,  1.6461172, 18.255245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'left_sensor_depth': SixDOFPose(position=array([-1.8926959,  1.6461172, 18.255245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'right_sensor_depth': SixDOFPose(position=array([-1.6926959,  1.6461172, 18.255245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0))} 
No.6 to select action
action: FORWARD
current agent's sensor state: {'left_sensor': SixDOFPose(position=array([-1.8926959,  1.6637564, **17.755245** ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'right_sensor': SixDOFPose(position=array([-1.6926959,  1.6637564, 17.755245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'left_sensor_depth': SixDOFPose(position=array([-1.8926959,  1.6637564, 17.755245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0)), 'right_sensor_depth': SixDOFPose(position=array([-1.6926959,  1.6637564, 17.755245 ], dtype=float32), rotation=quaternion(1, 0, 0, 0))} 
No. 6 is finished!
```

erikwijmans, 2021-01-28:
Habitat follows the conventional graphics coordinate system, so, sans rotation, the agent looks down -Z and forward moves along the -Z axis.

closed 2021-01-28 by erikwijmans

wwtx9, 2021-01-28:
@erikwijmans thanks, so when I multiple a roation  [[1, 0, 0],
					                                              [0,-1, 0],
					                                            [0, 0, -1]], whcih means rotation from camera coordinate to sensor coordinate ( is same as world coordinate without rotation), it can get correct value of sensor's position in world coordinate.

erikwijmans, 2021-01-28:
No, the sensor is already in world coordinates.  The world coordinate system follows the conventional graphics one.

wwtx9, 2021-01-29:
@erikwijmans Thanks again for your reply. Here I want to clear. My goal is to estimate position of point (target) in left camera in world coordinate. What I do is: 1) calculate the the position of target in left camera coordinate by intrinsic matrix. Because camera towards positive z axis in camera coordinate. (target seen by left camera, it means target is must in front of left camera, and it's depth is postive in left camera coordiante) 2)Calculate the position of target in world coordiante by rotation information from Habitat Api. As you can see here, left camera forward moves along the -Z axis in world coordinate, but forward move along the +Z axis in its own coordinate (left camera coordiante). So there is a rotation from left camera coordniate to world coordiante. Am I right?

lI know the sensor

erikwijmans, 2021-01-29:
No, the camera looks along -Z in it's frame.  https://aihabitat.org/docs/habitat-lab/view-transform-warp.html may be helpful.  That is using the habitat-lab api to get observations, but the math is identical.

erikwijmans, 2021-01-29:
That is just a convention.  That is the convention OpenCV follows and yes there is rotation between the convention that OpenCV follows and habitat follows.  There has been a lot of discussion about this in other issues, i.e.: https://github.com/facebookresearch/habitat-lab/issues/499

wwtx9, 2021-01-29:
I got it. Really appreciate it.