facebookresearch/habitat-sim #1073: Geodesic distance returning inf

NaveenVSrinivasan, 2021-02-02:
## 🐛 Bug

Calls to the Habitat Simulator's geodesic distance function are returning inf, similar to #405.

## To Reproduce

Steps to reproduce the behavior:

Here is the details of the Gibson scene, object locations, and which geodesic distance calls returned inf.

Gibson scene: Monson
Object 0 Location: [0.0017314087599515915, 0.24978113174438477, -1.9509133100509644]
Object 1 Location: [-0.9319772720336914, 0.35768193006515503, 2.1800076961517334]
Object 2 Location: [0.44389230012893677, 0.35768193006515503, 1.6933047771453857]
Object 3 Location: [-1.316479206085205, 0.25015464425086975, 0.06888800859451294]
Object 4 Location: [-2.756587505340576, 0.25015464425086975, 0.9424381852149963]
Object 5 Location: [-1.4617235660552979, 0.35768193006515503, -0.438821405172348]
Object 6 Location: [-2.88535737991333, 0.35768193006515503, 0.7335346937179565]
Object 7 Location: [-1.487607717514038, 0.3480518162250519, 0.5459910035133362]
Object 8 Location: [-0.4699043333530426, 0.3480518162250519, 0.5629610419273376]
Object 9 Location: [-3.105865240097046, 0.24978113174438477, -0.7899506092071533]
Object 10 Location: [0.0017314087599515915, 0.24978113174438477, -1.9509133100509644]

The geodesic distance calls that involve Object 1 all return inf.

erikwijmans, 2021-02-03:
Please provide pairs of points and agent configuration.  I tried the Object 1 point from above, but couldn't find any points that return inf distance over the 10k I tried for the default agent configuration.

NaveenVSrinivasan, 2021-02-03:
AGENT_0:
    SENSORS: ['RGB_SENSOR', 'DEPTH_SENSOR']
    HEIGHT: 0.88
    RADIUS: 0.18
  HABITAT_SIM_V0:
    GPU_DEVICE_ID: 0
    ALLOW_SLIDING: False
    ENABLE_PHYSICS: True
  SEMANTIC_SENSOR:
    WIDTH: 640
    HEIGHT: 480
    HFOV: 79
    POSITION: [0, 0.88, 0]
  RGB_SENSOR:
    WIDTH: 640
    HEIGHT: 480
    HFOV: 79
    POSITION: [0, 0.88, 0]
    ORIENTATION: [-0.52359877559, 0, 0]
  DEPTH_SENSOR:
    WIDTH: 640
    HEIGHT: 480
    HFOV: 79
    MIN_DEPTH: 0.5
    MAX_DEPTH: 5.0
    POSITION: [0, 0.88, 0]
    ORIENTATION: [-0.52359877559, 0, 0]
    
    By pairs of points do you want coordinates and rotations?

erikwijmans, 2021-02-03:
Coordinates.  I assume that Object 1 Location is either the start or end location for the geodesic distance query, so I need the corresponding start/end.  Rotation doesn't matter for geodesic distance.

Note that any two points are not guaranteed to have a finite distance.  There can be multiple different connected components in a given mesh as reconstruction artifacts, narrow doorways, or added objects can make it so the agent can't reach somewhere.

What dataset it this?  I am not aware of any inf distance issues for the datasets we have released (provided they are used with the proper agent config).

NaveenVSrinivasan, 2021-02-03:
This is the Monson scene from the Gibson dataset.

If you use Object 1 location as the start location for the geodesic distance query, and any of the other object locations as the end location, the return value is inf. 

When I generate my episode by adding objects to the Monson scene, I recompute the navmesh with each object addition and assert that all objects are existing in navegable points.

erikwijmans, 2021-02-03:
If you are adding objects, it is easy to end up causing gaps in navigability.  Query the distance to the center of an added object will also often end up not working as the simulator has to do some guessing because the center of the added object is, by definition, not navigable.  

One way to workaround this is to maintain a second instance of the PathFinder class that has an instance of the navmesh computed before adding objects and use that for distance queries.

closed 2021-04-15 by aclegg3